Koenkk/zigbee2mqtt #10: Support Request

12nick12, 2018-04-27:
SengLED 5000K zigbee bulb
Phillip Hue Color bulb

Koenkk, 2018-04-27:
In case you have these devices support can probably be added very easily. 
Lets start with the hue bulb. 
Can you try to factory reset the bulb? After the factory reset it should automatically pair.

12nick12, 2018-04-27:
I haven't tried yet, I just seen it was on the supported devices list. Any chance you'd happen to know the sengled should just work also?

12nick12, 2018-04-27:
Any chance this software would support a Linear HUSBZB-1?

Koenkk, 2018-04-27:
- Sengled will probably work, but you need to try pairing it with zigbee2mqtt. When paired zigbee2mqtt will print out messages needed to add support.
- Linear HUSBZB-1 -> No, zigbee2mqtt uses https://github.com/zigbeer/zigbee-shepherd which does not support Linear HUSBZB-1

Koenkk, 2018-05-01:
I've created a guide on how to support new devices: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

Koenkk, 2018-05-03:
Let me know if you are still interested to bring support to this device, otherwise I will close this issue.

closed 2018-05-04 by Koenkk