Koenkk/zigbee2mqtt #15: Group support

WoodsterDK, 2018-05-03:
Hi there,

I'm using groups for my IKEA Trådfri bulbs - will that be supported/work with this also?

tb-killa, 2018-05-03:
I Think the Original Group Function comes from the TRÅDFRI app or the TRÅDFRI gateway themselves.
https://www.ikea.com/gb/en/customer-service/smart-lighting-support/faq-smart-lighting/groups-moods/

I Think that this Group Feature should be builded on the mqtt Side and not inside this "Bridge".
You could build some Sort of Logic in Node-Red with MQTT Output to multiple Devices.

Koenkk, 2018-05-03:
I agree with @tb-killa that groups is the responsibility of your home automation software (E.G. home assistant/openHAB).

WoodsterDK, 2018-05-03:
Well it depends, the groups are easy to do in  HA but then the bulbs get turned on by one

If in a zigbee group wouldn't they act in sync?

Well, perhaps the bulbs are to be left at the IKEA gateway, end then the zigbee2mqtt can take care of the Xiaomi components.

palitu, 2018-05-04:
I would expect that the group in HASS will turn them on so quickly it would
appear in sync.

ciotlosm, 2018-05-10:
In my opinion this component should keep it's intended purpose of stateless gateway. The logic and grouping should sit in the "brain" which is usually the "hub". Either homeassistant/hassio or openhab or some other.

Koenkk, 2018-05-10:
Will not be implemented for now. Groups should be created in your home automation software.

subzero79, 2018-08-09:
Just to comment on this, the sequential turn on/off via home assistant either by group or using the light group platform involves sending the payloads to the mqtt broker, zigbee2mqtt will read one by one, making it not instant. 2 seconds until the last light is off, here is the subscription 

```
Aug 09 11:41:48 zigbee2mqtt/kitchen_1_8ab4/set {"state": "OFF"}
Aug 09 11:41:48 zigbee2mqtt/kitchen_1_8ab4 {"state":"OFF","color_temp":370,"brightness":248}
Aug 09 11:41:48 zigbee2mqtt/kitchen_1_8ab4 {"state":"OFF","color_temp":370,"brightness":248}
Aug 09 11:41:49 zigbee2mqtt/kitchen_2_b44c/set {"state": "OFF"}
Aug 09 11:41:49 zigbee2mqtt/kitchen_2_b44c {"state":"OFF","color_temp":370,"brightness":251}
Aug 09 11:41:49 zigbee2mqtt/kitchen_3_a3a5/set {"state": "OFF"}
Aug 09 11:41:49 zigbee2mqtt/kitchen_2_b44c {"state":"OFF","color_temp":370,"brightness":251}
Aug 09 11:41:49 zigbee2mqtt/kitchen_3_a3a5 {"state":"OFF","color_temp":370,"brightness":250}
Aug 09 11:41:50 zigbee2mqtt/kitchen_3_a3a5 {"state":"OFF","color_temp":370,"brightness":250}
Aug 09 11:41:50 zigbee2mqtt/kitchen_4_b8a5/set {"state": "OFF"}
Aug 09 11:41:50 zigbee2mqtt/kitchen_4_b8a5 {"state":"OFF","color_temp":350}
Aug 09 11:41:50 zigbee2mqtt/kitchen_4_b8a5 {"state":"OFF","color_temp":350}
```
```
Aug 09 11:41:48 ha node[16636]: 2018-8-9 11:41:48 INFO Zigbee publish to '0x8418260000018ab4', genOnOff - off - {} - null
Aug 09 11:41:48 ha node[16636]: 2018-8-9 11:41:48 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_1_8ab4', payload: '{"state":"OFF","color_temp":370,"brightness":248}'
Aug 09 11:41:48 ha node[16636]: 2018-8-9 11:41:48 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_1_8ab4', payload: '{"state":"OFF","color_temp":370,"brightness":248}'
Aug 09 11:41:49 ha node[16636]: 2018-8-9 11:41:49 INFO Zigbee publish to '0x841826000001b44c', genOnOff - off - {} - null
Aug 09 11:41:49 ha node[16636]: 2018-8-9 11:41:49 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_2_b44c', payload: '{"state":"OFF","color_temp":370,"brightness":251}'
Aug 09 11:41:49 ha node[16636]: 2018-8-9 11:41:49 INFO Zigbee publish to '0x841826000001a3a5', genOnOff - off - {} - null
Aug 09 11:41:49 ha node[16636]: 2018-8-9 11:41:49 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_2_b44c', payload: '{"state":"OFF","color_temp":370,"brightness":251}'
Aug 09 11:41:49 ha node[16636]: 2018-8-9 11:41:49 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_3_a3a5', payload: '{"state":"OFF","color_temp":370,"brightness":250}'
Aug 09 11:41:50 ha node[16636]: 2018-8-9 11:41:50 INFO MQTT publish, topic: 'zigbee2mqtt/kitchen_3_a3a5', payload: '{"state":"OFF","color_temp":370,"brightness":250}'
Aug 09 11:41:50 ha node[16636]: 2018-8-9 11:41:50 INFO Zigbee publish to '0x841826000001b8a5', genOnOff - off - {} - null
```
the esp milight project  uses groups of bulbs to control them and on/off, brightness and color change is instant for a group of bulbs

IMO this should be implemented in zigbee2mqtt, the bottleneck is mqtt. So the automation sw should be sending one payload.

subzero79, 2018-08-28:
Going through the shepherd library docs, seems like is supported. The function is bind/unbind

https://github.com/zigbeer/zigbee-shepherd/wiki#API_bind

So basically grab one endpoint and associate other EP(s) with them. This could be a topic for example

zigbee2mqtt/fan_lounge_a950/bind "fan_masterbedroom_825b"

Koenkk, 2018-08-28:
Does that create a group? e.g. after binding, when turning on `fan_lounge_a950` does `fan_masterbedroom_825b` also turn on?

ryanbeaton, 2018-08-28:
@subzero79 that looks good and there is also an unbind but how to manage ongoing? Would zigbee2mqtt have to keep a database?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

subzero79, 2018-08-28:
@koenkk from what I read in zigbee docs, yes when an ep is turned on, it should also turn the other binded ep. I was trying yesterday to hardcode a bind to test it in zigbee2mqtt but no luck yet. 

@ryanbeaton no, as I understand the db is created, updated and maintained by z.shepherd. Zigbee2mqtt doesn’t touch it.

subzero79, 2018-08-28:
Is not very clear in zigbee shepherd how to create a group. Because it shows binding two ep, then the other example binding an ep to a group. I would think the group id is created after you first bind two ep. then subsequently you can bind more ep’s to that group Id

subzero79, 2018-08-29:
sorry i am referring to endpoints as zigbee shepherd mentions, but as i understand only routers should be binded (grouped)? Don't know what would be the purpose of binding endpoints.

This is what i was reading when i was understanding wireshark captures when i found about groups in zigbee

https://www.edn.com/Pdf/ViewPdf?contentItemId=4204960

Maybe we should ask in zigbeer for more clarification

tb-killa, 2018-09-10:
I´v do some research and found some interessting stuff:

First of all: on ``cc-znp`` we have some sort of implementation for Group support:
https://github.com/zigbeer/cc-znp/wiki/6.-Z-Stack-MT-API-Reference-Tables#zdo-indication
``
ZDO_EXT_REMOVE_GROUP
ZDO_EXT_REMOVE_ALL_GROUP
ZDO_EXT_FIND_ALL_GROUPS_ENDPOINT
ZDO_EXT_FIND_GROUP
ZDO_EXT_ADD_GROUP
ZDO_EXT_COUNT_ALL_GROUPS
``
so maybe we could build some debug code for first test.

after that i do some sort of research and search on ``zigbee-shepherd`` source code for stuff like ``extAddGroup`` and found them in the file ``lib\components\zdo_helper.js``.
But sadly 
..
extFindGroup:       null,
    extAddGroup:        null,
    extCountAllGroups:  null,
`` 
so its not complete implemented on zigbee shepherd themselve.

As i could see now ... for fully implementation we need help and a lot of work!

Maybe @Koenkk could work on this too if moving is finished ;) ?

Koenkk, 2018-09-11:
@splitice have you already made some investigations for your application?

splitice, 2018-09-13:
We have no interest at this time in developing ZS's group support. We just don't need it for our applications. 

That being said I'm pretty sure at-least the ZS af layer (af.js) has group support. I'm not sure about adding and removing though.

ryanbeaton, 2018-11-28:
Some group work happening in here
https://github.com/Koenkk/zigbee2mqtt/issues/102#issuecomment-442379881

Koenkk, 2018-12-21:
Sneak peak: https://www.youtube.com/watch?v=3zI8mg0opzY

Koenkk, 2018-12-26:
@qosmio @subzero79 soon I will make an issue how to test the groups feature which explains how to use it and how it works. Note that the current MQTT topics are not final (still want to refactor some stuff)

Koenkk, 2018-12-27:
Group support has been implemented, feedback or questions about group support can be asked here: #764

closed 2018-12-27 by Koenkk