Koenkk/zigbee2mqtt #17: SJCGQ11LM (Xiaomi Water Leak Sensor) doesn't report immediately

Koenkk, 2018-05-04:
But does with the native gateway: https://www.youtube.com/watch?time_continue=149&v=8yGm2vH8HYQ

Koenkk, 2018-05-04:
@starkun it reports immediately after you drop it into the water? What configuration do you have?

tube0013, 2018-05-04:
@Koenkk I'm the who posted this in the HA forums.  I just tried running cat /dev/ttyACM0 and immediately get feed back when I drop it into water, also get feedback as soon as I dry it off.

kirovilya, 2018-05-08:
You try to detect leak event with attReport message. But statusChange event is more good for this.
Look at https://github.com/kirovilya/ioBroker.zigbee/blob/master/main.js#L523

ciotlosm, 2018-05-08:
@kirovilya do you have a description between the differen attrReport, devChange, statusChange? Would be great when adding more devices.

tb-killa, 2018-05-08:
@ciotlosm Look here: https://github.com/zigbeer/zigbee-shepherd/wiki#event-ind
- 'ind'

devIncoming
devLeaving
devChange
devStatus
attReport

kirovilya, 2018-05-08:
@ciotlosm Unfortunately not know. These were obtained by experience and in google.
Different devices uses different message type.

ciotlosm, 2018-05-08:
Thanks @tb-killa

kirovilya, 2018-05-08:
You also need modification of zigbee-shepherd library with this events support.
Like commit in my fork https://github.com/kirovilya/zigbee-shepherd/commit/77eee0836bb8866f28e843e89d060376cd9debca from not applied PR https://github.com/zigbeer/zigbee-shepherd/pull/6

Koenkk, 2018-05-08:
Thank you very much @kirovilya, works perfect now!

closed 2018-05-08 by Koenkk