Koenkk/zigbee2mqtt #20: Schematics and open questions

jwillmer, 2018-05-06:
The following image is linked in the readme: [image]

I guess a more detailed schematic looks like this?

```
[Generic MQTT gateway] <-> [Linux with zigbee2mqtt software] <-> [CC2531 dongle] <-> [Zigbee device 1-N]
```

- I have the dongle as Zigbee connector 
- This software interacts with the dongle and runs on Linux
- The software needs a MQTT gateway address to send/receive data

My open questions are:

1) Does it need to be the CC2531 dongle or can the software work with any dongle?
2) Why does the CC2531 need a custom firmware? Is this firmware part of the project / what does it change?
3) Is my assumption correct that I can use a Raspberry Pi Zero together with this software and the CC2531 dongle to get a Zigbee to MQTT bridge?
4) Can I have the setup described in 3 multiple times to cover my whole house? - Is there a cheaper/less power consumption way?

Koenkk, 2018-05-06:
1. zigbee2mqtt uses zigbee-shepherd which only supports the CC2530 and CC2531.
2. To use the CC2531 as a coordinator instead of sniffer.
3. Yes
4. Zigbee is a mesh network: http://www.rfwireless-world.com/images/zigbee-mesh-network.jpg
5. see 4

jwillmer, 2018-05-06:
Ah O.K. of course it supports mesh. The router part is interesting. Does any of the supported hardware have this functionality? Is there an easy way to know which devices work as router?

Koenkk, 2018-05-06:
I haven't found out how to check this, but I assume that most of the devices **without** battery act as a router. (E.G. Tradfri bulbs)

closed 2018-05-06 by jwillmer