Koenkk/zigbee2mqtt #21: TRADFRI bulb E27 opal 1000lm

aivs, 2018-05-06:
2018-5-6 18:51:11 WARN Device with modelID 'TRADFRI bulb E27 opal 1000lm' is not supported.
2018-5-6 18:51:11 WARN Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device

Please add to lib/devices.js

```
    'TRADFRI bulb E27 opal 1000lm': {
        model: 'LED1623G12',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 1000 lumen, dimmable',
        supports: 'on/off, brightness',
        homeassistant: [homeassistant.light_brightness_colortemp]
    }
```

2018-5-6 18:52:13 INFO 0x000b57fffe3298aa (0x000b57fffe3298aa): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable

Koenkk, 2018-05-06:
Done, can you check if it works?

aivs, 2018-05-06:
I'm newbie in mqtt, how I can check? How to send on/off and dim command ?

tb-killa, 2018-05-06:
You could Look at the Wiki :
https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure

Koenkk, 2018-05-06:
Yes, you can copy the example from the wiki, Philips Hue Go, and remove the color and color_temperature.

Koenkk, 2018-05-10:
@aivs and @dabont, What is the model number of your TRADFRI bulb? It is listed on the side of the bulb and starts with *LED*.

dabont, 2018-05-10:
LED1623G12

dabont, 2018-05-10:
Editing the devices.js by inserting a W in the name does not solve the issue. Hassio cannot operate the lamp and everytime I start Zigbee2Mqtt it pairs the lamp again.

aivs, 2018-05-10:
Now I’m using original zigbee-shepherd and ikea bulb works well. I wrote the HTTP interface to work with my z-wave automation system 
https://github.com/Z-Wave-Me/home-automation
and now I can dim the bulb from mobile app and other z-wave/zigbee switches.

Koenkk, 2018-05-11:
@dabont how did you add the bulb the hassio?

dabont, 2018-05-11:
@Koenkk It was automatically paired by Hassio using MQTT. For the sensors it works great, but the lamp does show up but when I try to operate it, it fails.

`2018-5-11 07:07:27 INFO ikeaTradfriTest (0x000b57fffe8a91b1): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white`

`2018-5-11 07:07:28 INFO MQTT publish, topic: 'homeassistant/light/0x000b57fffe8a91b1/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/ikeaTradfriTest/set","state_topic":"zigbee2mqtt/ikeaTradfriTest","availability_topic":"zigbee2mqtt/bridge/state","name":"ikeaTradfriTest"}'
`
`2018-5-11 07:25:10 ERROR Cannot handle 'zigbee2mqtt/0x000b57fffe8a91b1/set' because deviceID of '0x000b57fffe8a91b1' cannot be found`

Typing this message I see that there is a descrepancy between `zigbee2mqtt/ikeaTradfriTest/set` and  `zigbee2mqtt/0x000b57fffe8a91b1/set`

Futhermore, the Ikea bulb shows up multiple times in the database

```
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0001fa3902","nwkAddr":41017,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather"$
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00012434f4","nwkAddr":7958,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_sw$
{"id":6,"type":"Router","ieeeAddr":"0x000b57fffe8a91b1","nwkAddr":52947,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","mod$
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0001de5bc3","nwkAddr":13044,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_m$
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0009eb09c4","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1525984637,"endpoint$
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0001f5073c","nwkAddr":32894,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather"$
{"$$indexCreated":{"fieldName":"id","unique":true,"sparse":false}}
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0009eb09c4","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6],"status":"online","joinTime":1526022446,"endpoint$
{"id":2,"type":"EndDevice","ieeeAddr":"0x00158d0001fa3902","nwkAddr":41017,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather"$
{"id":5,"type":"EndDevice","ieeeAddr":"0x00158d00012434f4","nwkAddr":7958,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_sw$
{"id":6,"type":"Router","ieeeAddr":"0x000b57fffe8a91b1","nwkAddr":52947,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","mod$
{"id":4,"type":"EndDevice","ieeeAddr":"0x00158d0001de5bc3","nwkAddr":13044,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_m$
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0001f5073c","nwkAddr":32894,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather"$
{"id":6,"type":"Router","ieeeAddr":"0x000b57fffe8a91b1","nwkAddr":52947,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","mod$
{"id":6,"type":"Router","ieeeAddr":"0x000b57fffe8a91b1","nwkAddr":52947,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","mod$
```
any ideas?

Koenkk, 2018-05-11:
@dabont  After you changed a friendly name in the `configuration.yaml` you need to restart home assistant so that the ID will be updated. Will update the wiki with this.

dabont, 2018-05-11:
Thanks, it also took a reboot of zigbee-shepherd

Koenkk, 2018-05-11:
So with the `W` everything works? In that case it seems that the LED1623G12 has 2 possible modelids, maybe this could be due to different firmware versions.

Please let me know if everything works, then I will also add support for the modelID with the `W`.

dabont, 2018-05-12:
Yes, it works like a charm. On/off & dimmable. Can be closed if you ask me.

Koenkk, 2018-05-12:
Supported in master now!

closed 2018-05-12 by Koenkk