Koenkk/zigbee2mqtt #25: Provide Wiki Page for "Debugging Informations / How to"

tb-killa, 2018-05-08:
The Wiki should contain a page with Informations about "how to debug".

btw. We should refer to the zigbee shepherd wiki: https://github.com/zigbeer/zigbee-shepherd/wiki#8-debug-messages

Koenkk, 2018-05-08:
Thanks! I've setup an initial wiki page: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug. Is any information missing?

ciotlosm, 2018-05-10:
This was very useful. It seems there is a problem with linking the cube once you have it linked to something else (like a different stick). I had to go into debug mode (thanks for the wiki) and found out when I had to keep pressing link to make it connect again.

I had to keep clicking once I got a bunch of:
```
zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +18ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +7s
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +10s
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +2s
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +588ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +18s
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +862ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +38ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +912ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +525ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +0ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +506ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +534ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +1s
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +531ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +1ms
  zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0 +5s
```

closed 2018-05-11 by Koenkk

wsw70, 2019-02-22:
@Koenkk I think the page vanished (and the Wiki is not available from the main page)

Koenkk, 2019-02-22:
@wsw70 http://www.zigbee2mqtt.io/#how-tos http://www.zigbee2mqtt.io/how_tos/how_to_debug.html