Koenkk/zigbee2mqtt #26: Increase device limit beyond 48 devices.

Koenkk, 2018-05-08:
This is currently a limitation of the firmware for the CC2531.

Some resources:
* https://e2e.ti.com/support/wireless_connectivity/zigbee_6lowpan_802-15-4_mac/f/158/p/509541/1848708

ciotlosm, 2018-05-09:
Is there a way to get the a link to the firmware source code with details on how to build it as well?

Koenkk, 2018-05-09:
@oskarn97 would it be possible to share any details on how you compiled this firmware?

oskarn97, 2018-05-09:
It is not possible to increase the device limit beyond 48 devices since to memory limitation on the Chip. I compiled the Z Stack Home with IAR Embedded Workbench with some flags required for encryption and maximum device limit.

Koenkk, 2018-05-09:
Would it be possible to reduce the size of the firmware so there is more memory left for devices? http://processors.wiki.ti.com/index.php/Optimizing_Flash_and_RAM_Usage_of_Z-Stack_for_CC2530

Koenkk, 2018-05-09:
I doubt if `NWK_MAX_DEVICE_LIST` defines the maximum number of devices in the network. In this [official Texas Instruments document](http://www.compel.ru/wordpress/wp-content/uploads/2014/02/swra427b.pdf) chapter 4 Z-Stack Knobs they are talking about a 400+ nodes network while only having set `NWK_MAX_DEVICE_LIST` to `10`. I have the feeling that when adding more routers to your network you can add more nodes to your network.

Maybe 48 devices is not the limit at all....

oskarn97, 2018-05-09:
Yes, the device limit refers to the number of devices that can be paired to a single coordinator. You can use more of course.

ioudeveldhuis, 2018-05-17:
@oskarn97 Could you still tell us what you changed / done to compile the firmware and how to generate the .hex file ?

oskarn97, 2018-05-21:
I've just increased the device limit and enabled secure link join. Apart from that I did not modify the default TI Z Stack Home.
[image]

Koenkk, 2018-05-26:
By reducing the `MAXMEMHEAP` in `OnBoard.h` to e.g. 1000 I was able to compile with `NWK_MAX_DEVICE_LIST    90`.

@osarkn97 how did you enable secure link join?

oskarn97, 2018-05-26:
add the "TC_LINKKEY_JOIN" compile on the preprocessor option

ptvoinfo, 2018-05-27:
@Koenkk Is it safe to reduce MAXMEMHEAP?

Koenkk, 2018-05-27:
@ptvoinfo we need to do some tests with it.

Koenkk, 2018-05-27:
I think it's not possible to reduce the heap size, I'm getting the same error as #65 in this case.

Koenkk, 2018-05-28:
I've created a [wiki page](https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size) which explains how to add more than 48 devices to your network.

closed 2018-05-28 by Koenkk

Koenkk, 2018-12-01:
@makuser maybe; but that setup is much more expensive.

For now I don't see any problem when using the CC2531/CC2530 with the current firmwares. It's stable even for larger networks.

vpomax, 2019-01-25:
I have 32 xiaomi zigbee devices...
I need to use 1 coordinator e 2 router CC2531 to manage all 32 devices?

Koenkk, 2019-01-26:
@vpomax could be a bit tricky, I would recommend adding another CC2531 router to be sure.

cmorlok, 2019-01-26:
With the alternative firmware from
https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/alternatives/max_devices/
it should be possible to use 44 devices with one CC2531 without router.

Koenkk, 2019-01-26:
@cmorlok yes, but note that that firmware **could** have some downsides. It has less memory available which could lead to freezing and less responsive devices.

cmorlok, 2019-01-26:
Is the limit for cc2530 the same?

Koenkk, 2019-01-26:
Yes