Koenkk/zigbee2mqtt #28: Installation issues

dabont, 2018-05-09:
Hi, great to see renewed interest in this project. The old project I could install quite easily. However, I'm running into issue with this version. See below dump.

npm version = 6.0.0
Node version = 8.11.1

Could you please help?

`pi@raspberrypi:~/zigbee2mqtt $ npm start

> zigbee2mqtt@0.1.0 start /home/pi/zigbee2mqtt
> node index.js

/home/pi/zigbee2mqtt/node_modules/serialport/node_modules/bindings/bindings.js:91
  throw err
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/zigbee2mqtt/node_modules/serialport/build/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/build/Debug/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/build/Release/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/out/Debug/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/Debug/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/out/Release/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/Release/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/build/default/serialport.node
 → /home/pi/zigbee2mqtt/node_modules/serialport/compiled/8.11.1/linux/arm/serialport.node
    at bindings (/home/pi/zigbee2mqtt/node_modules/serialport/node_modules/bindings/bindings.js:88:9)
    at Object.<anonymous> (/home/pi/zigbee2mqtt/node_modules/serialport/lib/bindings.js:3:35)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/pi/zigbee2mqtt/node_modules/serialport/lib/serialport.js:12:25)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-05-09T09_55_35_050Z-debug.log
`

Koenkk, 2018-05-09:
Can you try:

```bash
rm -rf node_modules
npm install
npm start
```

dabont, 2018-05-09:
Thanks Koen. Reinstalling did the trick, is there any reason for it?

Koenkk, 2018-05-09:
Everytime `npm-shrinkwrap.json` is updated this needs to be done.

closed 2018-05-09 by Koenkk

mstiem, 2019-07-21:
Hi guys,

I actually have the same issue again on my pi zero w. I did all as discribed in your manpage but always get the error with the actual master:

`pi@raspberrypi:/opt/zigbee2mqtt $ npm --version
6.10.1
pi@raspberrypi:/opt/zigbee2mqtt $ node --version
v10.16.0
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.5.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Debug/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/out/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/Release/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/build/default/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/compiled/10.16.0/linux/arm/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/release/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/debug/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/addon-build/default/install-root/bindings.node
 → /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/binding/node-v64-linux-arm/bindings.node
    at bindings (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/linux.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:776:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:787:10)
    at Module.load (internal/modules/cjs/loader.js:653:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:593:12)
    at Function.Module._load (internal/modules/cjs/loader.js:585:3)
    at Module.require (internal/modules/cjs/loader.js:690:17)
    at require (internal/modules/cjs/helpers.js:25:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/bindings/lib/index.js:14:22)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.5.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.5.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-07-21T08_34_11_168Z-debug.log`

Any ideas? The solution you described in post 2 did not work for me so far.

jesperldk, 2019-11-25:
uh, I got exactly the same problem. Did try the solution from post 2, but it did not work :-(

jesperldk, 2019-11-25:
Didn't realize this issue is closed, opened a new one: #2412