Koenkk/zigbee2mqtt #29: Error when paired device is not known

dabont, 2018-05-09:
Really impressed with the progress from the original work of Andrew. I found a little error.

I previously paired devices to my dongle (with the Andrew version of the software). With the new fork when the program receives a message from an previuosly paired device it crashes, perhaps you can easily ctach this error.

```
2018-5-9 17:55:40 INFO MQTT publish, topic: 'zigbee2mqtt/sensorWoonkamer', payload: '{"temperature":26.67,"humidity":49.44,"pressure":1002}'
spinlock: false []
/home/pi/zigbee2mqtt/node_modules/q/q.js:155
                throw e;
                ^

Error: Unhandled "error" event. (Cannot get the Node Descriptor of the Device: 0x00158d00012434f4 (Error: Timed out after 10000 ms))
    at ZShepherd.emit (events.js:186:19)
    at /home/pi/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:609:28
    at _rejected (/home/pi/zigbee2mqtt/node_modules/q/q.js:864:24)
    at /home/pi/zigbee2mqtt/node_modules/q/q.js:890:30
    at Promise.when (/home/pi/zigbee2mqtt/node_modules/q/q.js:1142:31)
    at Promise.promise.promiseDispatch (/home/pi/zigbee2mqtt/node_modules/q/q.js:808:41)
    at /home/pi/zigbee2mqtt/node_modules/q/q.js:624:44
    at runSingle (/home/pi/zigbee2mqtt/node_modules/q/q.js:137:13)
    at flush (/home/pi/zigbee2mqtt/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-05-09T17_56_21_421Z-debug.log

```

Koenkk, 2018-05-09:
Did you also copy the database.db?

kirovilya, 2018-05-10:
Hi. I seen this problem: "Error: Timed out after 10000 ms"... 
For devices: RTCGQ01LM, MCCGQ01LM, WXKG01LM (may be more) need press setup key (small button on sensor) for 5 second at start until led blinking and than every 2-3 seconds on pairing process to revoke sleeping.
(sorry for my English)

dabont, 2018-05-10:
Hi, no did not copy the database. Lost all information after a clean install on my Raspberry pi.

@kirovilya Thanks

ciotlosm, 2018-05-10:
Still this is a bug on pairing on new devices where you get a timeout if you don't do what @kirovilya suggested (pressing link buttons). I think this needs either a fix or a mention in the documentation for linking devices to the stick.

Koenkk, 2018-05-10:
Yes, this needs to be fixed in zigbee-shepherd.

dabont, 2018-05-10:
Since the topic is on pairing and I dont want to start another one.

I see from the suported list that the 1000lm E27 is supported however, I get a not supported error.

```
2018-5-10 18:31:09 INFO New device joined the network!
2018-5-10 18:31:09 WARN Device with modelID 'TRADFRI bulb E27 W opal 1000lm' is not supported.
2018-5-10 18:31:09 WARN Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device
2018-5-10 18:31:09 WARN Device with modelID 'TRADFRI bulb E27 W opal 1000lm' is not supported.
2018-5-10 18:31:09 WARN Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device
^C2018-5-10 18:31:22 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'offline'
```
I'm sure that I did not update it correctly.

I have performed a `npm update` should that be enough?

Thanks guys

Koenkk, 2018-05-10:
You need to do a `git pull`

dabont, 2018-05-10:
Thanks Koen, I have a lot to learn.

The `git pull` result in a stashing thing which I need to find out for myself. However i found the follwing, it looks like a `W` is missing.

`2018-5-10 18:49:39 WARN Device with modelID 'TRADFRI bulb E27 W opal 1000lm' is not supported.
`
Since it shows in the devices.js as

` 'TRADFRI bulb E27 opal 1000lm': {`

Is there an issue in the code? When I try to insert the W the program crashes

Koenkk, 2018-05-10:
@dabont  this seems to be related to #21, let's continue over there.

Koenkk, 2018-05-12:
Only thing left here is the `Cannot get the Node Descriptor of the Device` bug. Let's continue that over here #41

closed 2018-05-12 by Koenkk

mcsSolutions, 2018-11-14:
The pairing solution for multiple devices looks to be similar to kirovilya post of pressing the link button during the process.  I receive the same Timed out after 10000 ms error with Iris 3326-L motion sensor.  This unit does not have link button.  Pairing is done by remove/replace the battery.  It keeps trying every 20 seconds.  Any suggestions on how to overcome the error for this device?