Koenkk/zigbee2mqtt #32: [Bug] Library appends to payload previous messages instead of sending just what is received.

ciotlosm, 2018-05-10:
Messages get merged on subsequent calls from same device. This is more of a problem for example on Xiaomi Cube where you can have rotate with two parameters (action and direction) vs normal actions where you have only one. 

```
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"temperature":26.86}'
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"temperature":26.86,"humidity":36.63}'
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"temperature":26.86,"humidity":36.63,"pressure":970}'
```

Koenkk, 2018-05-11:
This is done on purpose because home assistant requires all sensor values when it receives a message on a topic. 

In the case of:
```
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"temperature":26.86}'
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"humidity":36.63}'
2018-5-7 22:57:20 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001fa5c9d', payload: '{"pressure":970}'
```

The temperature and humidity sensor value would be null.

ciotlosm, 2018-05-11:
You have different sensors defined in home assistant for temperature/ humidity and air pressure. You can define them with separate messages. I think the only advantage would have been battery. 

For now this is not that big issue but will be for cube, once you have two parameters (angle number on rotate).

Koenkk, 2018-05-11:
Separate messages don't solve the problem, this can only be solved by using separate topics. (e.g. `zigbee2mqtt/0x00158d0001fa5c9d/temperature`, this however would be inconsistent with https://www.home-assistant.io/components/light.mqtt_json/ which is used for all lights. (this requires one topic per light).

Therefore I would prefer to set an attribute on the converter to not cache this message. (e.g. `cache: false`)

Koenkk, 2018-05-11:
The disableCache option has been added which disabled caching for certain converters. So in the future we can safely add the cube angles for the MFKZQ01LM

closed 2018-05-11 by Koenkk