Koenkk/zigbee2mqtt #34: Extract shepherdSettings in configuration.yaml

ciotlosm, 2018-05-10:
Please help extract `shepherdSettings` in `configuration.yaml`. 

So far we need for the hassio addon to persist data for the following to change:
```
dbPath: `${__dirname}/../data/database.db`
```

I assume the same is true for the other setting if you have multiple CC2531:
```
net: {panId: 0x1a62}
```

Koenkk, 2018-05-10:
Yes, will do this

Koenkk, 2018-05-11:
I'm wondering why there should be a configuration option for the `dbPath`. Can't this be done by mounting the `/app/data` volume to the persistent hassio folder?

tb-killa, 2018-05-11:
@Koenkk i think it should be the users choice to point to the DB file.
I vote up for the dbpath inside the configuration file.

danielwelch, 2018-05-11:
@Koenkk The current approach we're taking in the add-on doesn't mount `configuration.yaml` into a persistent Hass.io volume, but instead writes the options (from the Hass.io UI) to `configuration.yaml` in-place within the container on startup. Hass.io allows the add-on to specify certain volumes (such as `/share`) that can be mounted within an add-on container, so if we could specify the DB location, we could save to one of these persistent volumes. 

The problem with the current approach is that the location is hard-coded, and essentially "within" a bunch of things that we would not want to persist between builds/versions. In addition to making it possible to persist the data, breaking out the `dbPath` as a config option would make things less rigid in this sense.

dgomes, 2018-05-11:
Also the log file should remain in a separate directory.

I think /data is overly used for static stuff

Koenkk, 2018-05-11:
@danielwelch this sounds like a good approach, however how would you handle the devices being automatically added to `configuration.yaml`? In order for a device to work it has to be there.

danielwelch, 2018-05-11:
Sounds like `configuration.yaml` needs to be persisted too for this to work properly since device data is persisted there (not just the DB), right?

If that's the case, I think the easiest thing would be to allow the user to specify a path for this file as well via the Hass.io add-on, just like we're requesting for the db. This would let us store `configuration.yaml` in `/share`, and the user could edit it directly.

I understand these are big requests, sorry!

Koenkk, 2018-05-11:
Yes that is the case, wouldn't is be easier to symlink /app/data to /share?

danielwelch, 2018-05-11:
I am far from a Docker expert, but my understanding is that symlinking into and out of docker containers leads to problems and is generally not a good idea.

Since the major motivation for this change is the add-on, I'd be happy to work on a solution and make a PR. I'll focus on making this backward-compatible for current users.

Koenkk, 2018-05-12:
Both `dbPath` and `panId` are configurable now, can this be closed?

danielwelch, 2018-05-12:
I'd say so, thanks!

closed 2018-05-12 by Koenkk