Koenkk/zigbee2mqtt #35: Implement remove device from database

ciotlosm, 2018-05-10:
In case you have a defective device and want to disconnect it from your gateway, you need to run .remove for zigbee-shepard (https://github.com/zigbeer/zigbee-shepherd/wiki#removeieeeaddr-cfg-callback).

Ideal this would be an MQTT command that you can run.

palitu, 2018-05-12:
yeah i agree.

tb-killa, 2018-05-16:
@Koenkk Do you actually work on this ?

I think we should do: 
```
var dev = shepherd.find($device, 1).getDevice();
shepherd._devbox, 'remove', dev._getId()
shepherd._devbox, 'sync', dev._getId()
```

What type of mqtt topic whould we do ?

Koenkk, 2018-05-16:
I haven't worked on this yet. 

The MQTT topic should be `zigbee2mqtt/bridge/config/remove` with the device `friendly_name` as the payload.

Functionality should be implemented here: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L153

tb-killa, 2018-06-04:
@Koenkk : I work on this Issue but need some help for the remove function themselve:
Look at my Comments on: https://github.com/tb-killa/zigbee2mqtt
Maybe there are some Sort of Flawed thinking.

Koenkk, 2018-06-04:
@tb-killa looks good, where can I find the comments?

ciotlosm, 2018-06-04:
@tb-killa please open a PR with your changes against this repo. Use WIP in title to know that this wont' be merged quite yet. We can then help easier with review

kirovilya, 2018-06-04:
For my app was made forceRemove in controller:
```
    forceRemove(deviceID, callback) {
        const device = this.shepherd._findDevByAddr(deviceID);
        // force
        return this.shepherd._unregisterDev(device, (err, result) => {
            return callback(err, result);
        });
    }
```

tb-killa, 2018-06-04:
Okay I think it makes sense for us.
I will rework my "removedevice" function and will test again if i got the time.

kirovilya, 2018-06-04:
@tb-killa xiaomi sensor "device" do not removed by classic method... but "router" do. if classic remove not success - force remove.

ciotlosm, 2018-06-06:
@kirovilya why did you need a callback on force remove?

kirovilya, 2018-06-06:
@ciotlosm For waiting remove from shepherd and then delete other app objects.

ciotlosm, 2018-06-06:
Can you review my PR (I updated a bit the one from @tb-killa ) and let me know which other app objects to remove?

kirovilya, 2018-06-06:
@ciotlosm ok. I remove objects in my app: states, history...

ciotlosm, 2018-06-06:
@tb-killa  no worries. Sometimes I'm busy and happy when people assist with mine 👍 
@kirovilya states and history are not part of zigbee2mqtt right?

kirovilya, 2018-06-06:
@ciotlosm yes, not zigbee2mqtt part

ciotlosm, 2018-06-06:
@kirovilya can you review code here: #99  and let me know if you are happy with implementation?

kirovilya, 2018-06-06:
@ciotlosm sorry, not today and not tomorrow. birthday :)

ciotlosm, 2018-06-06:
Oh, no worries. Happy birthday. I'll have a drink for that!

kirovilya, 2018-06-06:
@ciotlosm thank you! :)

closed 2018-06-10 by Koenkk

pcb1962, 2019-02-02:
Did this ever get implemented? I have devices which I don't use appear on my network map, I assume this is because they have entries in database.db, I would like to remove these. I guess it's not ok to just delete the relevant lines from database.db?

pcb1962, 2019-02-02:
Thank you, apologies for not finding the relevant mqtt topic, I'm going to study that page now.

mawi13, 2019-03-23:
Link is broken... new one:
https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

dmak, 2019-07-14:
Does it perform any validation of the message? I just posted `rubbish` via Home Assistant interface
[image]
and I see the following in the log:
```
zigbee2mqtt:debug 7/14/2019, 1:51:34 PM Received MQTT message on 'zigbee2mqtt/bridge/config/remove' with data 'rubbish'
zigbee2mqtt:info 7/14/2019, 1:51:34 PM Successfully removed rubbish
zigbee2mqtt:info 7/14/2019, 1:51:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"rubbish"}'
```
It reports that it was successfully removed :)