Koenkk/zigbee2mqtt #40: Belkin Wemo Bulb

tube0013, 2018-05-12:
I got this working by adding the following to device.js

```   // Belkin
    'MZ100': {
        model: 'MZ100',
        vendor: 'Wemo',
        description: 'Belkin WeMo Smart LED Bulb 800 lumen, dimmable',
        supports: 'on/off, brightness',
        homeassistant: [homeassistant.light_brightness]
    },
```

It shows in HomeAssistant and I'm able to control it, but I still get these messages:

```
Zigbee publish to '0x94103ef6bf428aad', genOnOff - on - {} - 1
Zigbee publish to '0x94103ef6bf428aad', genLevelCtrl - moveToLevel - {"level":6,"transtime":0} - 1
MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON"}'
No converter available for 'MZ100' with cid 'genLevelCtrl' and type 'devChange'
Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues with this message.
Zigbee publish to '0x94103ef6bf428aad', genOnOff - off - {} - 1
MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"OFF"}'
No converter available for 'MZ100' with cid 'genOnOff' and type 'devChange'
Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues with this message.
```

Koenkk, 2018-05-12:
In order for a device to be supported you also need to add converters to `lib/converters/zigbee2mqtt.js`, see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices#3-adding-converters-for-your-device-to-libconverterszigbee2mqttjs

Koenkk, 2018-05-12:
I've added initial support for this bulb in the dev branch: https://github.com/Koenkk/zigbee2mqtt/tree/dev. Can you test if everything works? 

Also can you describe how you factory resetted the bulb (so I can update the wiki with this)?

tube0013, 2018-05-15:
I pulled in the changes in dev and all seems good.  - Sample output (note running zigbee2mqtt via systemd so it runs at startup of a headless pi) as turning light on and off and adjusting brightness.

```
May 15 17:55:45 zPi zigbee2mqtt[22912]: 2018-5-15 17:55:45 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON","brightness":255}'
May 15 17:55:48 zPi zigbee2mqtt[22912]: 2018-5-15 17:55:48 #033[32mINFO#033[39m Zigbee publish to '0x94103ef6bf428aad', genOnOff - on - {} - 1
May 15 17:55:48 zPi zigbee2mqtt[22912]: 2018-5-15 17:55:48 #033[32mINFO#033[39m Zigbee publish to '0x94103ef6bf428aad', genLevelCtrl - moveToLevel - {"level":144,"transtime":0} - 1
May 15 17:55:49 zPi zigbee2mqtt[22912]: 2018-5-15 17:55:49 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON","brightness":255}'
May 15 17:55:50 zPi zigbee2mqtt[22912]: 2018-5-15 17:55:50 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON","brightness":144}'
May 15 17:56:14 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:14 #033[32mINFO#033[39m Zigbee publish to '0x94103ef6bf428aad', genOnOff - on - {} - 1
May 15 17:56:14 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:14 #033[32mINFO#033[39m Zigbee publish to '0x94103ef6bf428aad', genLevelCtrl - moveToLevel - {"level":255,"transtime":0} - 1
May 15 17:56:14 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:14 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON","brightness":144}'
May 15 17:56:16 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:16 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"ON","brightness":255}'
May 15 17:56:35 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:35 #033[32mINFO#033[39m Zigbee publish to '0x94103ef6bf428aad', genOnOff - off - {} - 1
May 15 17:56:36 zPi zigbee2mqtt[22912]: 2018-5-15 17:56:36 #033[32mINFO#033[39m MQTT publish, topic: 'zigbee2mqtt/0x94103ef6bf428aad', payload: '{"state":"OFF","brightness":255}'
```

Koenkk, 2018-05-16:
Supported now

closed 2018-05-16 by Koenkk