Koenkk/zigbee2mqtt #41: Unhandled Error - Cannot get the Node Descriptor

palitu, 2018-05-12:
I am trying to pair a Xiaomi round switch with a newly create bridge, and have this error when i try to pair it.  

Unfortunately it crashes the bridge :(

```
/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:155
                throw e;
                ^

Error: Unhandled "error" event. (Cannot get the Node Descriptor of the Device: 0x00158d0001256963 (Error: Timed out after 10000 ms))
    at ZShepherd.emit (events.js:186:19)
    at /home/leigh/apps/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:609:28
    at _rejected (/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:864:24)
    at /home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:890:30
    at Promise.when (/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:1142:31)
    at Promise.promise.promiseDispatch (/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:808:41)
    at /home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:624:44
    at runSingle (/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:137:13)
    at flush (/home/leigh/apps/zigbee2mqtt/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/leigh/.npm/_logs/2018-05-12T03_44_42_166Z-debug.log

```

Koenkk, 2018-05-12:
This is a bug in zigbee-shepherd. For these MiJia devices to pair you need to keep pressing the reset button as mentioned here: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#xiaomi

palitu, 2018-05-14:
Yeah, took a heap of tries to get my motion sensor paired, but got there in the end.

ciotlosm, 2018-05-30:
Anyone with enough knowledge to patch zigbee-shepherd? I mean, to open a very specific merge request for this specific error

ciotlosm, 2018-06-06:
This is no longer an issue. The bug that crashed the process is fixed. The error still appears on timeout but that is different thing.

closed 2018-06-07 by Koenkk