Koenkk/zigbee2mqtt #45: I have a proposal for cooperation with you

kirovilya, 2018-05-14:
Hi, Koen!

We are both developing the subject of Zigbee-devices (Xiaomi and other), so I propose to combine our experience.
You do the mqtt-version, and I'm the adapter for the ioBroker. But both these decisions use NodeJS and Zigbee-Shepherd library. 
Therefore, I propose to create a common library of device and parsers for further use in your and my applications. This library should depend only on the zigbee-shepherd, handle the zigbee events, and generate processed events.

I am accustomed to your approach to the architecture of parsers (unlike my single file) - I suggest taking this as a basis.

I think this is an excuse not to do double work and not to copy the code from the library to the library.

What can you say about this?

p.s. Sorry for my English

kirovilya, 2018-05-14:
Perhaps this idea is better.

devices.js - i see homeassistant dependence (can cut)
zigbee2mqtt.js - is good

For architecture, i want to see multiple generated events on one zigbee-message.
For example like one genMultistateInput message for magic cube must generate events:
rotate - rotation flag,
rotate_dir - rotation direction,
rotate_side - rotation cube side (if we can detect :)  ),
rotate_angel - rotation angel
And we can handle any of this event as we need.

DurandA, 2018-05-14:
I also have a [related project](https://github.com/DurandA/xiaomi-zigbee-adapter) and would be very interested in a common library.

This project is in early phase as it only supports a few devices yet. I am also looking for contributors to support more devices.

Koenkk, 2018-05-14:
I prefer the flow of having: `zigbee-shepherd message -> converter -> friendly message`.

The `friendly message` could then have all the attributes needed (e.g. `rotate_dir` and  `rotate_side`).

kirovilya, 2018-05-14:
I thought more and suggest that there should be a dependence on common `zigbee-shepherd` and `zcl-packet`. Because we need device management commands: switching outlets, requesting states, light control and others.

> I prefer the flow of having: zigbee-shepherd message -> converter -> friendly message.

it's ok.
and also back control: `friendly message -> converter -> zigbee command`

kirovilya, 2018-05-15:
Something like this
[image]

kirovilya, 2018-05-17:
I see progress.

Some files ready to share in common lib:
* devices.js - is good
* converters/* - good too (I suggest renaming converters files, without mentioning "mqtt". for example, replace on "app" or simpler "tozb.js" and "fromzb.js")
* but zigbee.js have some strange (for me) dependense on utils. Or you think not nessesery to share it?
* what about controller.js? At present it combine mqtt, zigbee and converters functional together.

But I want to see "mqtt", "logging", "settings" functional externally.

What you think?

Koenkk, 2018-05-17:
I'm only planning to move the converters/* and devices.js to a seperate library because I feel it's not nice/maintainable to create another library on top of zigbee-shepherd. This will keep the library very thin which allows our projects for maximum flexibility.

kirovilya, 2018-05-17:
Ok, I agree. Let's start split and use and upgrade :)

kirovilya, 2018-05-18:
Hello again...
What you think about moving bindings from parsers.js to devices.js?
Named parsers will binding in device-object. And controller will send to parser it device-object (for getting any parameters, like voltageRange in example). And also for commands.

Example:
```
devices = {
    'lumi.sensor_switch': {
        model: 'WXKG01LM',
        vendor: 'Xiaomi',
        description: 'MiJia wireless switch',
        supports: 'single, double, triple, quadruple, many and long click',
        voltageRange: battery3V,
        parsers: ['batteryReport', 'clicks'],
        commands: [],
    },
}
```
```
parsers = {
    batteryReport: {
        cid: 'genBasic',
        type: 'attReport',
        convert: (dev, msg) => {
            let voltage = null;
            if (msg.data.data['65281']) {
                voltage = msg.data.data['65281']['1'];
            } else if (msg.data.data['65282']) {
                voltage = msg.data.data['65282']['1'].elmVal;
            }
            if (voltage) {
                return {battery: toPercentage(voltage, dev.voltageRange.min, dev.voltageRange.max)};
            }
        },
    },
    clicks: {
        cid: 'genOnOff',
        type: 'attReport',
        disableCache: true,
        convert: (dev, msg, publish) => {
            const deviceID = msg.endpoints[0].device.ieeeAddr;
            const state = msg.data.data['onOff'];

            // 0 = click down, 1 = click up, else = multiple clicks
            if (state === 0) {
                store[deviceID] = setTimeout(() => {
                    publish({click: 'long'});
                    store[deviceID] = null;
                }, 300); // After 300 milliseconds of not releasing we assume long click.
            } else if (state === 1) {
                if (store[deviceID]) {
                    clearTimeout(store[deviceID]);
                    store[deviceID] = null;
                    publish({click: 'single'});
                }
            } else {
                const clicks = msg.data.data['32768'];
                const payload = clickLookup[clicks] ? clickLookup[clicks] : 'many';
                publish({click: payload});
            }
        },
    },
}
```

It may be useful for creating new devices in one place (or in files) without parsers modification.

Koenkk, 2018-05-18:
Good idea!

Koenkk, 2018-05-23:
Repo is up: https://github.com/Koenkk/zigbee-shepherd-converters

To-do for myself:
- [ ] Refactor zigbee2mqtt to use zigbee-shepherd-converters
- [ ] Update zigbee2mqtt wiki on how to support new devices
- [ ] Add travis/eslint configuration to zigbee-shepherd-converters repo

kirovilya, 2018-05-23:
Ok, thank you. I, was busy too :)
I try to use this lib in my project.

Need some work with device endpoints: in devices it is {'l1': 2, 'l2': 3}, but in converters {2: 'l1',3: 'l2'}
Can I try to refactor it to one standart and put it in devices?

Koenkk, 2018-05-23:
Ok, good idea!

Koenkk, 2018-05-24:
Everything is ready :smile:

closed 2018-05-24 by Koenkk