Koenkk/zigbee2mqtt #46: Add Ikea OTA Update Functionality

tb-killa, 2018-05-14:
It would be great if we could also update our Ikea Products.
I am currently experimenting with the update file(s).

To download this files, you only need to parse a simple json file:
http://fw.ota.homesmart.ikea.net/feed/version_info.json
name and fw_binary_url could be simple extract.

We now need a way to transfer the data.

cbu99, 2018-06-24:
Probably deConz thread can give some insight?
[https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96](url)

kirovilya, 2018-06-27:
@splitice start to add this feature to his forks https://github.com/HalleyAssist/zcl-packet/commit/07bd96a8a9fdca28c1924c2a9e81ac052d6ac379

splitice, 2018-06-27:
FYI we are successfully OTA updating sensors in-house currently.

The actual OTA server isn't open source currently as it's tied into our main Zive app (closed source) however you are very welcome to use our zcl-packet and zcl-id forks for genOta definitions.

kirovilya, 2018-06-27:
@splitice thank you for sharing the information!

tb-killa, 2018-08-06:
@splitice 
Would you be willing to provide us with a sample code from your ota server?

In principle, we do not have a start to finalize a possible solution afterwards.

splitice, 2018-08-06:
@tb-killa I have a meeting with the IP owner on Monday. I'll see what the mood is.  Currently got alot on our plate so not sure when we would have time to separate it out though. 

There is a known bug in our OTA implementation that I have to go hunting for at some time, so perhaps that will be the excuse needed.

tb-killa, 2018-08-06:
I say thanks for all your efforts :)

I'm looking forward to your feedback.

tb-killa, 2018-08-15:
@splitice any news on this?
It would be So great if we could get some starting / init snippet ;)

closed 2018-10-22 by tb-killa

abmantis, 2019-01-08:
@tb-killa why have you closed this issue? It is a very interesting feature to have in the future.

NewFolk, 2019-03-10:
Hello, is there any news on OTA? I think it is very interesting feature that the main gateway should have.

andreasbrett, 2019-03-24:
+1 especially now that IKEA is partnering up with Sonos and Xiaomi for new and/or combined smart home products