Koenkk/zigbee2mqtt #48: Possibility to capture NodeJs crash messages to a log file

ptvoinfo, 2018-05-16:
It will be very useful to capture crash messages and all console output to a log file. It will help to diagnose problems with new functionality on headless systems or if the scripts works in the service mode.

Koenkk, 2018-05-16:
Zigbee2mqtt log is already written to `data/log.txt`. 

For the exception, I prefer the accepted solution from https://stackoverflow.com/questions/10122245/capture-node-js-crash-reason (your process manager is responsible of this).

ptvoinfo, 2018-05-16:
In this case I will get 2+ log files. In case of a problem I must open both files and sync these files by time. If debug output is enabled this process can be hard.

ciotlosm, 2018-05-16:
You can run tail on multiple files with "-f": tail -f /var/log/auth.log /var/log/daemon.log

Koenkk, 2018-05-16:
In that case you would do: `npm start 1> log.out 2> log.out`

ciotlosm, 2018-05-28:
@danielwelch Maybe add this setup for https://github.com/danielwelch/hassio-zigbee2mqtt/blob/master/zigbee2mqtt/run.sh#L16 for easier debugging. It does makes sense to keep one log with timestamps when debugging.

ciotlosm, 2018-06-02:
@ptvoinfo can this issue be closed?

ptvoinfo, 2018-06-03:
@ciotlosm yes, please. I'm using the method:

> npm start 1> log.out 2> log.out

closed 2018-06-03 by ptvoinfo