Koenkk/zigbee2mqtt #51: Trying to pair an Ikea Tradfri dimmer causes crash

james-fry, 2018-05-17:
I have tried to pair an Ikea dimmer like [this](https://www.ikea.com/gb/en/products/lighting/smart-lighting/tr%C3%A5dfri-wireless-dimmer-white-art-00347831/).

To be honest I was not expecting it to work as I understand that supporting this device is complex due to the way it works.

Nonetheless, here is the log from the crash:

`2018-5-17 19:39:29 INFO Zigbee: allowing new devices to join.
spinlock: false []
spinlock: true []
spinlock: true [ { func: [Function: func], ieeeAddr: '0x000b57fffe13a3ef' } ]
already in joinqueue
/app/node_modules/q/q.js:155
                throw e;
                ^

Error: Unhandled "error" event. (Cannot get the Node Descriptor of the Device: 0x000b57fffe13a3ef (Error: Timed out after 10000 ms))
    at ZShepherd.emit (events.js:186:19)
    at /app/node_modules/zigbee-shepherd/lib/components/controller.js:609:28
    at _rejected (/app/node_modules/q/q.js:864:24)
    at /app/node_modules/q/q.js:890:30
    at Promise.when (/app/node_modules/q/q.js:1142:31)
    at Promise.promise.promiseDispatch (/app/node_modules/q/q.js:808:41)
    at /app/node_modules/q/q.js:624:44
    at runSingle (/app/node_modules/q/q.js:137:13)
    at flush (/app/node_modules/q/q.js:125:13)
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-05-17T18_39_57_529Z-debug.log
`

Koenkk, 2018-05-17:
Ive got the same error when I didn't reset my Hue Go. Can you try with the following procedure: https://www.google.nl/url?sa=t&source=web&rct=j&url=https://m.youtube.com/watch%3Fv%3Dd5o3-729w7s&ved=0ahUKEwjRwcHYx43bAhWMLFAKHTIjDU4QyCkIKTAA&usg=AOvVaw0u4KV8O-FXTp0xHTfQZu92

james-fry, 2018-05-17:
That's what I did to get the log above :)
Tried again and success (as much as I expected):
`
spinlock: false []
spinlock: true []
spinlock: true [ { func: [Function: func], ieeeAddr: '0x000b57fffe13a3ef' } ]
already in joinqueue
2018-5-17 21:32:50 INFO Connecting with device, please wait...
2018-5-17 21:32:56 INFO Connecting with device, please wait...
2018-5-17 21:33:03 INFO Connecting with device, please wait...
2018-5-17 21:33:10 INFO Connecting with device, please wait...
2018-5-17 21:33:17 INFO Connecting with device, please wait...
2018-5-17 21:33:24 INFO Connecting with device, please wait...
2018-5-17 21:33:26 INFO Connecting with device, please wait...
2018-5-17 21:33:26 INFO Connecting with device, please wait...
2018-5-17 21:33:27 INFO Connecting with device, please wait...
2018-5-17 21:33:27 INFO Connecting with device, please wait...
2018-5-17 21:33:27 INFO New device joined the network!
2018-5-17 21:33:27 INFO New device with address 0x000b57fffe13a3ef connected!
2018-5-17 21:33:27 WARN Device with modelID 'TRADFRI wireless dimmer' is not supported.
2018-5-17 21:33:27 WARN Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device
2018-5-17 21:33:27 WARN Device with modelID 'TRADFRI wireless dimmer' is not supported.
2018-5-17 21:33:27 WARN Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device
device already in network
`

OK so this is the 1st step...

Koenkk, 2018-05-17:
Yes, now you can follow: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices

closed 2018-05-17 by james-fry

Koenkk, 2018-05-18:
Did you get it working?

james-fry, 2018-05-18:
Not yet - I'm working with @danielwelch's [hassio addon](https://github.com/danielwelch/hassio-zigbee2mqtt) (docker) but that doesn't give me the ability to change code- only the config.
I need to either build the hassio addon locally and use my fork of zigbee2mqtt for development, or daniels hassio addon needs to be changed to allow copy of the device support file (devices.js, converters etc) from mapped volume on start of the addon.