Koenkk/zigbee2mqtt #52: Extend range of CC2531 signal

james-fry, 2018-05-18:
The CC2531 does not seem to have as good range as my Xiaomi hub.
Using same type of Xiaomi sensor (new generation reed switch sensor - MCCGQ11LM) I found that the CC2531 is failing to receive updates, where the Xiaomi does.
(Neither network has active/powered devices to act as repeater)

Is there any way to improve range?

1) I have the dongle on a USB extension to move it away from the metal of my VM server, but I think I can use a longer extension to get further away from the metal...

2) I can add powered device (e.g.) bulb as a repeater. Problem with this is that bulbs do get hard switched off sometimes. I dont have any zigbee sockets.

3) There are CC2531 devices with a RF front end and external antenna with supposedly higher range.
e.g. [this one](https://www.ebay.co.uk/itm/CC2531-RFX2401C-PA-LNA-RF-Front-End-2-4GHz-IEEE-802-15-4-ZigBee-USB-dongle/141941450431?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2055119.m1438.l2649)
But these cost more + less easy to program (need to solder header) + not sure if firmware is compatible with the RFX2401C PA chip

ptvoinfo, 2018-05-19:
You may add a custom router with the external antenna (based on 2nd board from the left)
http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
I'm using 2 routers and 1 coordinator in my rather large home (about 500 m2).

kirovilya, 2018-05-19:
@ptvoinfo It is good news! Thank you. Can you make and share cc2531 firmware for router?

ptvoinfo, 2018-05-19:
@kirovilya 
I can try. But why CC2531 is better than CC2530? CC2530 is cheaper and has the external antenna.

kirovilya, 2018-05-19:
@ptvoinfo yes, but cc2531 is simplier for other users :) "plug and use"

ptvoinfo, 2018-05-19:
@kirovilya this is not simple as you think :)
1. you need to flash a new firmware.
2. you need to place it somewhere with a 5V power source and usually this is a hidden place. So IMHO, CC2530 is better. 
Therefore I use CC2531 as a sniffer only.

james-fry, 2018-05-19:
Plug into one of these?
USB Mains Charger 1Amp/1000mAh Rapid Speed Universal for Apple and Samsung Devices for Samsung S7 / S8 / Galaxy / Tab / HTC / Nexus / Xperia / Hudl / One Plus 2 /3 / Apple iPhone 5 / 5c / 5s / 6 / 6 Plus /6s / 6s Plus / 7 / 7 Plus - iPad 1 / 2 - Mobile Phone / iPod / MP3 & 4 Player Adaptor Socket - By Guilty Gadgets https://www.amazon.co.uk/dp/B01N25761P/ref=cm_sw_r_cp_api_i_KxdaBbMNEQA2E

james-fry, 2018-05-19:
Hi @ptvoinfo 
I read your blog post linked above - thx for sharing.
I notice that for the router firmware you mention that you "Pre-configured manufacturer code for Xiaomi devices."
Does this mean that the router f/w needs to be customised for the devices that will be in the network? Can it be made generic?

ptvoinfo, 2018-05-20:
Hi @james-fry 
A generic router must be configured first using Z-Tool (set network ID, default channel, security key). From my point of view this is not an easy process.

james-fry, 2018-05-20:
Understood. With my limited understanding of zigbee, I had assumed that we'd need to pair the router with the coordinator in the same was as an end device - e.g. perhaps using one of the two ptm switches on the cc2531 sniffer to initiate pairing?. This would then set the ID/channel/key in same way as a device like the xiaomi smart sockets (which are both end device and router).
BTW I totally get that writing f/w for this is a big job, but maybe there is capability in this forum to get it working?

james-fry, 2018-05-21:
Hi @kirovilya - I noticed that your testing includes two boards with RF frontends (cc2530+cc2591, cc2530+RFX2401).
Are you aware of whether the ["48 device" coordinator firmware ](https://github.com/Koenkk/zigbee2mqtt/tree/master/firmware) for this project is compatible with those frontends? As I understand it the CC253x SoC has to program/control the frontend chip via GPIOs, so the f/w needs to include support. Both 2591 and RFX2401 have different requirements.
@oskarn97 - If Im correct you were the source for the f/w. Would you be able to confirm this? 

Maybe we can start a list of compatible boards over and above the recommended CC2531 sniffer USB dongle?

oskarn97, 2018-05-21:
I've just increased the device limit and enabled secure link join. Apart from that I did not modify the default TI Z Stack Home.

james-fry, 2018-05-21:
Thanks for the swift replies, @kirovilya @oskarn97 

So as a summary we have support for:
- CC2531 with this firmware: https://github.com/Koenkk/zigbee2mqtt/tree/master/firmware
- CC2530 with this firmware: https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex
- CC2530 + CC2591 with this firmware: https://github.com/kirovilya/files/blob/master/CC2530ZNP-Test_for_CC2591.hex

Not supported
- CC2531 with any RF front end (CC2591 or RFX2401)
- CC2530 with RFX2401

Is this correct?

kirovilya, 2018-05-21:
@james-fry not correct.
https://github.com/kirovilya/files/blob/master/CC2530ZNP-Test_for_CC2591.hex compatible with CC2591 and RFX2401.
I not found CC2531 with any RF.

james-fry, 2018-05-21:
@kirovilya Thanks. Updated the summary with this confirmation.

dgomes, 2018-05-21:
We need to make this information sticky in the Wiki :)

kirovilya, 2018-05-21:
@dgomes I also maintain wiki in Russian. it will be possible to give a link to it as a localized resource :)

james-fry, 2018-05-21:
I'm happy to update wiki and make a PR later today.

ptvoinfo, 2018-05-21:
@kirovilya 
> CC2530 with this firmware: https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex

What communication settings does this firmware use? What pins must be connected? I've tried 115200,8N1 pins (P02, P03, GND, 3.3V) and it does not work for me. My own firmware works.

kirovilya, 2018-05-21:
@ptvoinfo P20 to GND

ptvoinfo, 2018-05-21:
@kirovilya P20 to GND: tried too.

kirovilya, 2018-05-21:
@ptvoinfo Hmm. My connections to UART cable:
RX - P03
TX - P02
3.3v - VCC
GND - GND

and on board wire P20 to GND. Port speed 115200, but **rtscts** is **false**

What error did you receive?

ptvoinfo, 2018-05-21:
@kirovilya 
I'm using UART TTL to USB converter. Wiring is the same. zigbee2mqtt cannot initialize CC2530. Data timeout.

james-fry, 2018-05-21:
@Koenkk - I didn't realise that wiki is not part of repo (GH noob here), so I cant edit your wiki in my fork and submit PR :(
Are you able to give me permission to update the wiki? Alternative I can message you with the wiki markup?

kirovilya, 2018-05-21:
@ptvoinfo Very strange... Try to reboot :)
I flashed and used cc2530 many times...

kirovilya, 2018-05-21:
@ptvoinfo may be you have CC2530 + CC2591 board ?

kirovilya, 2018-05-21:
@ptvoinfo in case cc2530+RF i need connect P06 and P04 to GND too (do not ask me "why" :) ). may be it help in your case.

Koenkk, 2018-05-21:
@james-fry you should be able to edit the wiki now.

james-fry, 2018-05-21:
Great. Will update shortly.

james-fry, 2018-05-21:
Updated wiki. 
New page for supported CC sniffer devices:
https://github.com/Koenkk/zigbee2mqtt/wiki/Supported-sniffer-devices

james-fry, 2018-05-21:
This guy on smartthings forum made a nice CC2350 based solution with custom pcb:
https://community.smartthings.com/t/another-zigbee-repeater-solution-with-temperature-humidity-pressure-and-light-sensors/123815

He sees the usefulness of usb for power source :)

kiucheng, 2018-05-21:
The guy over smartthings does seem to sharing his firmware. I found this: setting a slightly more expensive Xbee S2C as router https://alselectro.wordpress.com/2017/01/23/zigbee-xbee-s2c-how-to-configure-as-coordinator-router-end-device/

kirovilya, 2018-05-22:
@kiucheng As I know, Xbee S2C can't work with xiaomi devices. May be I wrong...

ryanbeaton, 2018-05-22:
HI @kirovilya , do you have a link for the cc2530+RFX2401 ("ZIGBEE RF to USB dongle)?

https://www.aliexpress.com/item/1pcs-lot-CC2530-CC2591-USB-ZigBee-dongle-board-free-shipping/1813876248.html
Is the only one I could find but it is unavailable.

kirovilya, 2018-05-22:
@ryanbeaton I got here (two times) https://www.aliexpress.com/item/RF-TO-USB-CC2530-CC2591-RF-switch-USB-transparent-serial-data-transmission-equipment/1996354384.html
Description says "CC2591", but in fact it has GB2530-S chip 
[image]
wich have CC2530+RFX2401 http://www.gban.cn/en/product_show.asp?action=p295&id=47

james-fry, 2018-05-22:
Haha I was just about to post that same link. I speculated that you were the person who left the reviews :)

The GB2530-S chip seems to have a RF(x)2401 front end:
http://www.gban.cn/en/product_show.asp?action=p295&id=47

kirovilya, 2018-05-22:
@james-fry @ryanbeaton Yes, 5 pin header use for ccdebugger (look at description on Aliexpress). Cable in complect exists.

ryanbeaton, 2018-05-22:
thanks, have ordered
wasn't to sure about description but this helped
http://www.gban.cn/en/product_show.asp?action=p297&id=43

james-fry, 2018-05-22:
I also ordered. Hope the antenna and additional sensitivity from the Rf frontend will resolve my range issues

kirovilya, 2018-05-22:
@ryanbeaton @james-fry remember me when received devices :) because:
[image]

james-fry, 2018-05-22:
Better get my fine tip soldering iron out... what’s this for?

kirovilya, 2018-05-22:
Yes

ryanbeaton, 2018-05-22:
@kirovilya Oh! I forgot about those comments of yours. Sorry but can you clarify, I just solder up like that and then flash. Do I also leave it like that to operate also. Thanak

kirovilya, 2018-05-23:
@ryanbeaton Yoy may solder after flash and leave it.
And other one thing i said: "sensor can not send long range signal"...
So, Coordinator will have "long range", but sensors - no.

ptvoinfo, 2018-05-23:
I would share a solution for my problem with CC2530 and the ZNP firmware (https://github.com/kirovilya/files/blob/master/CC2530ZNP-Pro-Secure_LinkKeyJoin.hex).

**Problem:** **zigbee2mqtt** does not want to connect to CC2530 with correct wiring.

> RX - P03
> TX - P02
> 3.3v - VCC
> GND - GND
> P20 to GND

**Background:** The ZNP firmware for CC2530 has the serial boot loader code. This boot loader starts and verifies an checksum of the application code. Then it waits for **1 minute** for any commands from UART. 0xEF - skip the boot loader code and go to the normal mode and 0x10 - stay in the boot loader mode **forever**. 

**Solution:** Send 0xEF to a UART port. I'm not sure, but it is possible we can send this command to zigbee2mqtt when a timeout appears with first initialization attempt.

kirovilya, 2018-05-23:
@ptvoinfo It is very-very strange... 

I look at code zigbee2mqtt, how it is start zigbee-shepherd:

```
const shepherdSettings = {
    net: {
        panId: settings.get().advanced.pan_id,
        channelList: [settings.get().advanced.hasOwnProperty('channel') ? settings.get().advanced.channel : 11],
    },
    dbPath: data.joinPath('database.db'),
};
...
shepherd = new ZShepherd(settings.get().serial.port, shepherdSettings);
```

There is no port parameters. When i start, i set port speed and rtscts:

```
shepherd = new ZShepherd(port, {
        net: {panId: 0x1a62},
        sp: { baudrate: 115200, rtscts: false },
        dbPath: dbDir+'/shepherd.db'
});
```

Try to append this parameter on start.

ptvoinfo, 2018-05-23:
This does not matter. The same problem appears without zigbee2mqtt. I cannot connect to CC2530 using ZTool too for 1 minute without sending this magic byte (0xEF).

kirovilya, 2018-05-23:
@ptvoinfo what the board are you use? I use this https://www.aliexpress.com/item/CC2530-Zigbee-UART-Wireless-Core-Board-Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html without any problem.

ptvoinfo, 2018-05-23:
@kirovilya Yes, the same board from the same seller. I can only say "it happens" :)

kirovilya, 2018-05-23:
@ptvoinfo thank you for experiens. but i don't know what to do...

james-fry, 2018-05-23:
@kirovilya 
Regarding you comment:
```And other one thing i said: "sensor can not send long range signal"...
So, Coordinator will have "long range", but sensors - no.```

My understanding is that the RF frontend has a low noise amp, and hence the sensitivity of the receiver is increased. So the end device is no "louder" but the coordinator's ability to "hear" is increased.
That plus additional gain from a real antenna should, I hope, give additional range for sensors.

kirovilya, 2018-05-23:
@james-fry Yes, all true. it is good that you understand this.

james-fry, 2018-05-23:
@kirovilya @ryanbeaton FWIW I found the manuals for the USB stick and the SoC here:
http://www.gban.cn/yk_Picture/hdp/20140523091382828282.pdf
http://www.gban.cn/yk_Picture/hdp/2014052216470315315.pdf

ptvoinfo, 2018-05-25:
I've prepared a router firmware for CC2531: 
http://ptvo.info/cc2531-based-router-firmware-136/
Your comments are welcome!

kirovilya, 2018-05-25:
@ptvoinfo good work! will try later.

james-fry, 2018-05-25:
Nice job @ptvoinfo. I have started to update the wiki with links for your 2530 and 2531 router firmwares.
Right now Im at work and your site is blocked, so will complete the updates later.
When my new device arrives I will reflash my 2531 as a router and test.

realroywalker, 2018-05-26:
Can anyone clarify what equipment is needed to use https://www.aliexpress.com/item/RF-TO-USB-CC2530-CC2591-RF-switch-USB-transparent-serial-data-transmission-equipment/1996354384.html

I see the modifications to connect a few pins above, but does it require ccdebugger AND the downloader cable as described for the other stick in the getting started guide?
If so, can anyone confirm if its a specific downloader cable?

Lastly I'm correct in thinking this is currently the easiest setup to go with for best signal?

ptvoinfo, 2018-05-27:
@realroywalker The CC debugger and download cable is required in any case. In some cable you need a proprietary cable, in some case you need to solder wires to the board/dongle.

realroywalker, 2018-05-27:
@ptvoinfo thank you. I've been searching aliexpress but cannot find any downloaded cable for the cc2530, only cc2531 seems available.

ptvoinfo, 2018-05-27:
@realroywalker 
CC2530 boards: http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
I've used a cable for CC2531 with the board from the first image. 
The second board has been temporarily connected using a dupont cable (female - female). Pinout: http://ptvo.info/wp-content/uploads/2018/05/smartrf04eb-pinout.jpg

Koenkk, 2018-05-27:
Would it be possible to pair end devices to routers? In this way we can increase the network size. #26

[218 more replies]

closed 2018-06-15 by james-fry