Koenkk/zigbee2mqtt #55: Xiaomi Motion Sensor Not sending occupancy:false

palitu, 2018-05-20:
I have a motion sensor setup, however it rarely sends a occupancy:false.

it is a `Motion_Front_Yard (0x00158d00010e49cc): RTCGQ01LM - Xiaomi MiJia human body movement sensor`

[image]

as you can see, it was true all over night, when we were all asleep.  Any help would be appreciated. 

```
devices:
  '0x00158d00010e49cc':
    friendly_name: 'Motion_Front_Yard'
    occupancy_timeout: 30
    retain: false
```

Koenkk, 2018-05-20:
Did zigbee2mqtt crash during this period?

palitu, 2018-05-20:
Yeah most likely. I'm Getting a lot of index out of bounds crashes.

I don't know how often it crashes, but I would say. Alot!

palitu, 2018-05-20:
roger!

thanks

palitu, 2018-05-28:
@Koenkk, i have the motion sensor set to 10s and it still doesn't send a `occupancy : false` message 

is there something that i should be doing to get it to send it?  it updates the `occupancy : true` every 5s, but nothing else.

Koenkk, 2018-05-28:
As long as it detects occupancy it doesn't sent a `occupancy: false`. After 10s of no occupancy (in your case) it should send `occupancy: false`.

palitu, 2018-05-29:
Ah - i recon this has to do with me using an old docker container.  I'll
update when i have the details.