Koenkk/zigbee2mqtt #56: LED1623G12 E26 version not supported

oakbrad, 2018-05-20:
Thank you for this project, just getting started with your Docker container.

I have the LED1623G12 model bulb, but in E26 plug.

```
0 19:30:00 INFO New device with address 0x000b57fffe2f1df7 connected!
0 19:30:00 WARN Device with modelID 'TRADFRI bulb E26 opal 1000lm' is not supported.
```

Seems this would be a simple addition, but I am not sure how to edit the device files with the Docker container. Can you advise how best to do this with Docker? Can the needed files be written to a volume so that I can edit them?

Koenkk, 2018-05-20:
Isn't the correct model id LED16**22**G12? Got it from Technical information part of https://m.ikea.com/ca/en/catalog/products/art/00339435/.

oakbrad, 2018-05-20:
LED1623G12 is what is printed on my bulb - my model supports different color temperatures same as what's listed in the supported devices. What you linked I think is only the basic "warm white" version.

I think just different plug type for different markets? I am in US where E26 is most common

Koenkk, 2018-05-20:
The LED1623G12 only supports on/off and brightness (no color temperature). As far as the models I've seen LED15xxxxx = color temperature, LED16xxxxx = no color temperature.

oakbrad, 2018-05-20:
Sorry you are correct, no color temp, got my bulbs mixed up.

It does appear to be the same as the 1622 linked, the color temp, watts, and lumens match. Mine definitely has 162**3** printed on it though and is E26. If I remember right this was part of a kit with a dimmer puck.

Koenkk, 2018-05-20:
Ok, no problem! I will add support for it tomorrow.

Koenkk, 2018-05-21:
Supported now.

closed 2018-05-21 by Koenkk