Koenkk/zigbee2mqtt #58: Running in Docker

Odianosen25, 2018-05-22:
Hello @Koenkk ,

Thanks for all your great work.

I am having some issues running the code in docker, My setup is using Libreelec (Kodi) with the docker addon. Had to go via this route, as I am having an issue with range and instead of trying to get a new Pi, I may as well maximise the Pis with Kodi in the other rooms; then just buy a new sniffer. 

I get the following error when I try to run the code
```
#docker -H 0.0.0.0:2375 run -it -v $(pwd)/.config/share/data:/app/data --device=/dev/ttyAMA0 koenkk/zigbee2mqtt:arm32v6
Using '/app/data' as data directory

> zigbee2mqtt@0.1.0 start /app
> node index.js

2018-5-22 05:43:06 INFO Starting zigbee-shepherd
2018-5-22 05:43:09 ERROR Error while starting zigbee-shepherd!
2018-5-22 05:43:09 ERROR Failed to start
^C2018-5-22 05:43:19 ERROR Not connected to MQTT server!
2018-5-22 05:43:19 ERROR Cannot send message: topic: 'zigbee2mqtt/bridge/state', payload: 'offline
2018-5-22 05:43:19 INFO zigbee-shepherd stopped
```  

It seems not to be connecting to my mqtt broker, which I have setup in the `configuration.yaml` file. I am sure that mqtt configuration is right, as I have used it in another setup, which is not based on using the docker.

Please I need help in pointing me to the logs so I can see what exactly the issue is. The logs in the `/data` directory says the same above, so doesn't help. I need to know if its an authentication issue (which I doubt), or network or something.

Though looking at the solution based on using the CC2530 based Zigbee coordinator and router, but still will be nice to just get a solution on this one.

Thanks and kind regards

ciotlosm, 2018-05-22:
Have a look at: https://github.com/danielwelch/hassio-zigbee2mqtt

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Odianosen25, 2018-05-22:
@ciotlosm,

Thanks for the response.

But I am not using hassio, but docker running on my Kodi instance. The link you sent me, asks that I set `debug` to True to get my debug messages. but that is not working in my instance.

Thanks

Koenkk, 2018-05-22:
While the stick is plugged in (and the green led is on) can you press the button on the stick that is closest to the USB? The green led should now turn off. Now try starting again.

Odianosen25, 2018-05-22:
Still same thing unfortunately

Koenkk, 2018-05-22:
Start with `DEBUG=* npm start`

Odianosen25, 2018-05-22:
@Koenkk ,

Sorry I don't get. You mean I should run something like this?

`docker -H 0.0.0.0:2375 run -it -v $(pwd)/.config/share/data:/app/data --device=/dev/ttyAMA0 DEBUG=* npm start koenkk/zigbee2mqtt:arm32v6`

Or where do I put `DEBUG=* npm start` in?

Koenkk, 2018-05-22:
Sorry, this is not possible with docker yet. Whats the output of `ls /dev | grep tty`

Odianosen25, 2018-05-22:
This is what I got
```
tty
tty0
tty1
tty10
tty11
tty12
tty13
tty14
tty15
tty16
tty17
tty18
tty19
tty2
tty20
tty21
tty22
tty23
tty24
tty25
tty26
tty27
tty28
tty29
tty3
tty30
tty31
tty32
tty33
tty34
tty35
tty36
tty37
tty38
tty39
tty4
tty40
tty41
tty42
tty43
tty44
tty45
tty46
tty47
tty48
tty49
tty5
tty50
tty51
tty52
tty53
tty54
tty55
tty56
tty57
tty58
tty59
tty6
tty60
tty61
tty62
tty63
tty7
tty8
tty9
ttyAMA0
```
A lot right?

DurandA, 2018-05-22:
Does Docker have the authorization to access ttyAMA0 without sudo?

Odianosen25, 2018-05-22:
@DurandA,

Well in LibreELEC, I noticed all is running as root.

@Koenkk,

Ok thanks for clarifying that. I was using it on another Pi, before moving it to this one; so I am quite certain it is fine. As I said above, it is possible I don't have the proper driver and unfortunately LibreELEC is a bit naughty at allowing someone install anything.

Will seek out a driver for it, and let you know when I have the `/dev/ttyACM0` showing up. Not home now, so can't physically check play with it now till evening.

Thanks and regards

Odianosen25, 2018-05-22:
Ok @koenkk, I have reconnected it, and I can now see the sniffer. But now it still fails, and doesn’t hook as before.

So any more pointers please?

Thanks

ciotlosm, 2018-05-22:
You could still start with custom npm start for debug just like the hassio addon does. Just do a small wrap on top and change the entrypoint to something custom to be able to do it. 

Can you please repost the logs at startup? Maybe something is different from your original post.

Odianosen25, 2018-05-22:
Hello @ciotlosm,

As requested
```
LibreELEC:~ # docker -H 0.0.0.0:2375 run -it -v $(pwd)/.config/share/data:/app/d
ata --device=/dev/ttyACM0 koenkk/zigbee2mqtt:arm32v6
Using '/app/data' as data directory

> zigbee2mqtt@0.1.0 start /app
> node index.js

2018-5-22 17:45:51 INFO Starting zigbee-shepherd
2018-5-22 17:45:51 ERROR Error while starting zigbee-shepherd!
2018-5-22 17:45:51 ERROR Failed to start
```
I am not so good in docker, so don't understand what you meant by
```
You could still start with custom npm start for debug just like the hassio addon does. Just do a small wrap on top and change the entrypoint to something custom to be able to do it.
```

Could you help with a sample?

ciotlosm, 2018-05-22:
@Odianosen25 if it's something useful for others please post the solution as well. Might be others searching and your solution could put them on the right track.

ciotlosm, 2018-05-22:
There are no dumb solutions. Every solution helps someone looking in the wrong place. Glad you got it sorted.