Koenkk/zigbee2mqtt #61: Request a report of status of device

kirovilya, 2018-05-24:
I can't find how request current status of device...
For example, a temperature sensor sends a report when the values changed significantly, but i want it regularly (hourly, minutely). Like "heartbeat" with Xiaomi gateway.

dgomes, 2018-05-24:
I implemented that for a power plug:

https://github.com/Koenkk/zigbee2mqtt/commit/567bdca27647f434d4fe00120b1a4567b36a811e

kirovilya, 2018-05-24:
@dgomes I see... But it not work for xiaomi devices :(

Koenkk, 2018-05-24:
I also noticed that this indeed doesn't work with Xiaomi devices.

dgomes, 2018-05-24:
bummer :( 

Did not try with my xiaomi devices...

ciotlosm, 2018-05-24:
This would have been awesome. Even reducing the value for "changed significantly" to be under 0.5 degrees would be an improvement for temperature sensors from xiaomi.

ptvoinfo, 2018-05-25:
In most cases these read report request will fail because battery powered devices are in the deep sleep mode and does not process incoming requests.

kirovilya, 2018-05-25:
https://community.smartthings.com/t/xiaomi-zigbee-door-window-sensor-motion-sensor-smart-button-device-type-beta/31948/105 
Here I found some code for "//battery reporting and heartbeat" and "Refreshing Battery" (in lastes strings of code block), but can't repeat in NodeJS

tb-killa, 2018-05-25:
@kirovilya : Do you test with the "publish" function ?
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/zigbee.js#L135

We should check the needing Data.

kirovilya, 2018-05-25:
@tb-killa not yet. I tested "report" function

tb-killa, 2018-05-27:
Doesnt we could build an api to readout the last state of the device specific state from the database file?
As long as the device Doesnt send ne state to the zigbee shepherd themselve we could query this.

Another one is the needing work to trigger the endpoint themselve like @kirovilya mentioned.

ciotlosm, 2018-05-27:
I disagree with adding another layer of state. Add unrequired complexity

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

tb-killa, 2018-05-30:
@kirovilya : i check your pointed groovy file and the marked section : first i search for the commands and found this great Information: https://community.smartthings.com/t/zigbee-commands-rattr-wattr-cmd/1294/8
We could now do some more Tests.
If i got some time i will check Different Things, nice now too that my Devices arrived Today.

lolorc, 2018-06-06:
I own several osram plugs, would the report work with them ? (i'm using zigate to control them at the moment) it would be useful to be able to poll them for changes, and also to automatically get a status change when the physical button is pressed

Koenkk, 2018-06-06:
@lolorc the OSRAM plug is fully supported, a status change event is send then the physical button is pressed, see https://github.com/Koenkk/zigbee2mqtt/issues/62

lolorc, 2018-08-27:
instead of opening another github issue, would it also make sense to be able to trigger a status update of a device from a mqtt command ?

Koenkk, 2018-08-30:
@lolorc yes, that is definitely some  functionality I want to add!

trimethia, 2018-12-19:
Any updates? Would be great to be able to ask for a status report of my tradfri bulb.

Koenkk, 2018-12-19:
@trimethia I think you want #518

closed 2019-02-24 by stale[bot]