Koenkk/zigbee2mqtt #62: Osram smart plug

jay-jey, 2018-05-24:
Hello I tried to add a Osram smart plug. I tried to folow the description here
https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
, but I couldn't find the  node_modules/zigbee-shepherd-converters/devices.js I have only one folder with node_modules/zigbee-shepherd but whitout a device.js

Koenkk, 2018-05-24:
This is a recent change, please update zigbee2mqtt

```bash
git pull
rm -rf node_modules
npm install
```

Koenkk, 2018-05-27:
What is the model ID of this device? In the log it will show as

```
Device with modelID '**MODELID**' is not supported.
```

If you share it I can add support for it.

jay-jey, 2018-05-27:
Thanks for the support, when I get the bridge running again I will post the data.

nobodyman1, 2018-05-27:
Do you mean the OSRAM SMART+?
This one produces following output:
`WARN Device with modelID 'Plug 01' is not supported.`

nobodyman1, 2018-05-28:
It's AB3257001NJ.
But how do I add it?
The log shows
`INFO New device joined the network!`
`INFO New device with address ... connected!`
If I press the button nothing happens.

Koenkk, 2018-05-28:
@nobodyman1 thanks, the plug is now partially supported. 

Can you update zigbee2mqtt to the latest version, turn the plug on/off (with the button on it) and post the log here?

nobodyman1, 2018-05-28:
```
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:24 INFO Connecting with device, please wait...
2018-5-28 18:56:25 INFO Connecting with device, please wait...
2018-5-28 18:56:25 INFO Connecting with device, please wait...
2018-5-28 18:56:25 INFO Connecting with device, please wait...
2018-5-28 18:56:25 INFO New device joined the network!
2018-5-28 18:56:25 WARN Device with modelID 'Plug 01' is not supported.
2018-5-28 18:56:25 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-5-28 18:56:25 WARN Device with modelID 'Plug 01' is not supported.
2018-5-28 18:56:25 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

If I press the button still nothing happens.

Koenkk, 2018-05-28:
Can you start with `DEBUG=* npm start`?

nobodyman1, 2018-05-28:
Yes, but still nothing happens... No output.

Koenkk, 2018-05-28:
Strange, does controlling (on/off) of the switch work (via e.g. home assistant)?

Koenkk, 2018-05-28:
Can you test with zigbee2mqtt? 

Publish `{"state": "ON"}` to `zigbee2mqtt/DEVICE_ID`.

nobodyman1, 2018-05-28:
Still nothing. I also tried a second one.

Koenkk, 2018-06-04:
Fully supported now, pressing on the buttons gives feedback and controlling works. 

Feel free to reopen in case of any issues.

closed 2018-06-04 by Koenkk

tavoc, 2018-11-23:
Hi there, 
i have 2 AB3257001NJ, but they behave strange.
They connect to zigbee2mqtt <>fhem, and if I press the button I see the state switches from off to on.
But i can not change the state from fhem (webcmd on:off).

I get the following error:

`2018-11-23 13:05:14 - info: Zigbee publish to '0x84182600000f0aad', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
2018-11-23 13:05:21 - error: Zigbee publish to '0x84182600000f0aad', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 233. MAC no ack.
`

In FHEM I get the following list:
`Internals:
   DEF        AB3257001NJ 0x84182600000f0aad
   FRIENDLYNAME 
   IODev      MQTT
   MODEL      AB3257001NJ
   NAME       0x84182600000f0aad
   NOTIFYDEV  AB3257001NJ 0x84182600000f0aad
   NR         54
   SID        0x84182600000f0aad
   STATE      OFF
   TYPE       XiaomiMQTTDevice
   Helper:
     DBLOG:
       state:
         logdb:
           TIME       1542974746.17735
           VALUE      off
       transmission-state:
         logdb:
           TIME       1542954915.25413
           VALUE      subscription acknowledged
   READINGS:
     2018-11-23 06:04:54   linkquality     105
     2018-11-23 06:04:54   state           OFF
     2018-11-23 07:35:15   transmission-state subscription acknowledged
   message_ids:
   subscribe:
     zigbee2mqtt/0x84182600000f0aad
     xiaomi/0x84182600000f0aad/#
   subscribeExpr:
     ^zigbee2mqtt\/0x84182600000f0aad$
     ^xiaomi\/0x84182600000f0aad.*$
   subscribeQos:
     xiaomi/0x84182600000f0aad/# 0
     zigbee2mqtt/0x84182600000f0aad 0
Attributes:
   IODev      MQTT
   room       XiaomiMQTTDevice
   webCmd     on:off`

I am using the masterbranch from yesterday.
COnfiguration.yaml:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM1
devices:
  '0x00158d000255fb50':
    friendly_name: '0x00158d000255fb50'
    retain: false
  '0x00158d0002701c64':
    friendly_name: '0x00158d0002701c64'
    retain: false
  '0x00158d0002701e32':
    friendly_name: '0x00158d0002701e32'
    retain: false
  '0x00158d0002391d58':
    friendly_name: '0x00158d0002391d58'
    retain: false
  '0x84182600000f0aad':
    friendly_name: '0x84182600000f0aad'
    retain: false
`
state.json
`{
    "0x84182600000f0aad": {
        "state": "OFF",
        "linkquality": 105
    },
    "0x00158d000255fb50": {
        "battery": 42,
        "voltage": 2985,
        "linkquality": 52
    }
}
`

database entry
`{"id":7,"type":"Router","ieeeAddr":"0x84182600000f0aad","nwkAddr":63229,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","epList":[3],"status":"online","joinTime":1542949450,"endpoints":{"3":{"profId":49246,"epId":3,"devId":16,"inClusterList":[0,3,4,5,6,2820,4096,64527],"outClusterList":[25],"clusters":{"64527":{"dir":{"value":1},"attrs":{"0":0}},"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":1,"appVersion":12,"stackVersion":2,"hwVersion":1,"manufacturerName":"OSRAM","modelId":"Plug 01","dateCode":"20140331DEOS****","powerSource":1,"deviceEnabled":1,"swBuildId":"V1.04.12"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":0,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genOta":{"dir":{"value":2},"attrs":{}},"haElectricalMeasurement":{"dir":{"value":1},"attrs":{"measurementType":8,"activePower":28000}},"lightLink":{"dir":{"value":1},"attrs":{}}}}},"_id":"1w7MdK4VBa4BrPs0"}
`

Koenkk, 2018-11-23:
```status code: 233. MAC no ack.``` means that the zigbee network cannot reach the device. Does it work when having the device close the the CC2531?

vuiperez, 2018-11-25:
hello Koenkk, 
first of all many thnaks for sharing your job!

i paired 2 osram smart plug to extend the range of the zigbee, but they don't pair has router only has normal switch.
here the logs
[image]

[image]

can you help please?

best regards

Koenkk, 2018-11-25:
Why do you think these are not paired as a router?

tavoc, 2018-11-28:
> 
> 
> `status code: 233. MAC no ack.` means that the zigbee network cannot reach the device. Does it work when having the device close the the CC2531?

Thanks, I did not recognize that the Osrams have a shorter range than my other devices. Put them nearer and they are working.
Thank you

Alestrix, 2019-02-23:
> Why do you think these are not paired as a router?

Sorry for hijacking this issue.
I too cannot use the plug as router. In the network graph it always shows as "offline", but controlling the plug works without issues. Putting an end device close does not trigger its traffic to go through the router (also not after hours and also not if reception would be much better).
See also issue #372