Koenkk/zigbee2mqtt #63: Xiaomi Aqara double key wireless wall switch (WXKG02LM)

ciB89, 2018-05-25:
The sniffer seems not to find this device / pair with it. Others also have the same problem, as reported in the HA forum.

Koenkk, 2018-05-25:
Pairing is a little bit more difficult for this device, keep both buttons pressed until the 2 lights on the bottom start blinking. After some time it should pair.

Odianosen25, 2018-05-25:
@ciB89,

Well as @Koenkk said, it can be a bit difficult and I just completed pairing all mine last night. The best way I figured out was to follow the following steps:
- Ensure the switch is close to it. I already had mine mounted so had to remove them all and place them within like 50-100cm of the sniffer (well maybe case I had mine somewhere at the corner).
- Press the left one down and hold it down
- Wait until both lights starts flashing
- Keep holding it down and don't release until both stops flashing and the left LED flashes once
- Wait and take a chill and let the stuff do its magic

There was the temptation to keep pressing the button to "keep it alive", especially after the nightmare of pairing my Mija sensors which required it. but don't do this, as I think it made matters worse. Just leave it once you see the message pop up `..._lock[]` (sorry can't remember exactly what it says)

Hope it helps. Cheers

Odianosen25, 2018-05-25:
@Koenkk,

That said, please can you add within the MQTT messages, status update of the pairing process? For example if one was to activate the add devices over mqtt, it will be nice to be able to get the status of the pairing process over mqtt.

Basically the same messages sent over the console as follows:
```
2018-05-22T18:00:47.197Z - info: Zigbee: allowing new devices to join.
2018-05-22T18:23:39.197Z - info: ..._lock[] (or whatever it is as its not in the logs) 
2018-05-22T18:23:40.204Z - info: Connecting with device, please wait...
2018-05-22T18:23:43.409Z - info: Connecting with device, please wait...
2018-05-22T18:23:46.658Z - info: Connecting with device, please wait...
2018-05-22T18:23:49.882Z - info: Connecting with device, please wait...
2018-05-22T18:23:53.117Z - info: Connecting with device, please wait...
2018-05-22T18:23:56.353Z - info: Connecting with device, please wait...
2018-05-22T18:23:59.612Z - info: Connecting with device, please wait...
2018-05-22T18:24:02.838Z - info: Connecting with device, please wait...
2018-05-22T18:24:06.073Z - info: Connecting with device, please wait...
2018-05-22T18:24:09.310Z - info: Connecting with device, please wait...
2018-05-22T18:24:12.558Z - info: Connecting with device, please wait...
2018-05-22T18:24:15.775Z - info: Connecting with device, please wait...
2018-05-22T18:24:19.011Z - info: Connecting with device, please wait...
2018-05-22T18:24:22.233Z - info: Connecting with device, please wait...
2018-05-22T18:24:25.492Z - info: Connecting with device, please wait...
2018-05-22T18:24:25.506Z - info: New device joined the network!
2018-05-22T18:24:25.516Z - info: New device with address 0x00158d000xxxxxxx connected!
```

I will then pick them over mqtt. So that I can start and get status all via my GUI or Voice system.

Thanks

ciB89, 2018-05-25:
I'm running the Hassio ([version](https://github.com/danielwelch/hassio-zigbee2mqtt), but I set debug to true. But except for my other zigbee device (so far got only one other) it shows me nothing when I try to pair..

Koenkk, 2018-05-25:
Ok, I will make a video how to pair once I'm back from holiday.

ciB89, 2018-05-25:
Ok, thanks :). I don't feel like I'm making a mistake though, but who knows.. let's hope for the best (and that my switch isn't just broken).

Koenkk, 2018-05-25:
One thing you also can try is changing the battery, I had one Xiaomi device which I couldnt pair, noticed that the blue pair light was quite dark, replacing the battery fixed the problem.

ciB89, 2018-05-25:
Battery seems fine, lights are quite bright

Koenkk, 2018-05-25:
Than it's very strange that you see nothing after a factory reset of the device when running with `DEBUG=* npm start`

ciB89, 2018-05-25:
This is what I tried: [video](https://photos.app.goo.gl/3L8dTgwgqRHybq6C2)

Koenkk, 2018-05-25:
If I remember correctly, in my case both lights flashed. Can you try pressing both buttons?

ciB89, 2018-05-25:
Already tried that. Both LEDs turn on and stay on after the flashing, but still nothing on the debugger.

ciB89, 2018-05-25:
I'll try recplacing the battery.

ciB89, 2018-05-27:
Replacing the battery did the trick. But now it crashes everytime:
`2018-5-27 22:23:40 INFO Connected to MQTT server
2018-5-27 22:23:40 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-5-27 22:23:41 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0002486c1c/action/config', payload: '{"icon":"mdi:cube","value_template":"{{ value_json.action }}","json_attributes":["battery"],"state_topic":"zigbee2mqtt/cube","availability_topic":"zigbee2mqtt/bridge/state","name":"cube"}'
/app/lib/homeassistant.js:238
    const friendlyName = settings.getDevice(deviceID).friendly_name;
                                                     ^
TypeError: Cannot read property 'friendly_name' of undefined
    at discover (/app/lib/homeassistant.js:238:54)
    at Object.discover (/app/lib/homeassistant.js:267:42)
    at zigbee.getAllClients.forEach (/app/lib/controller.js:59:35)
    at Array.forEach (<anonymous>)
    at Controller.handleStarted (/app/lib/controller.js:57:41)
    at mqtt.connect (/app/lib/controller.js:48:85)
    at MqttClient.client.on (/app/lib/mqtt.js:26:13)
    at emitOne (events.js:121:20)
    at MqttClient.emit (events.js:211:7)
    at MqttClient._handleConnack (/app/node_modules/mqtt/lib/client.js:891:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-05-27T20_23_41_064Z-debug.log`

What's happening?

ciB89, 2018-05-27:
Ok, fixed it. Apparently I screwed up the pairing process and in the configuartion.yaml was the device still missing, so I added it manually, now it works.

ciB89, 2018-05-28:
Still need help with this. After adding the button in the configuration.yaml (I got the addresse from the database.db), it shows that it joins in the logs etc, the switch also thinks its paired, I receive "commands" when the buttons are pressed, but there is no MQTT publish happening - the sensor in HA is not updated.
Depending on the buttons pressed, the logs only output:

`Mon, 28 May 2018 10:01:53 GMT cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 13780, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 60, securityuse: 0, timestamp: 7392275, transseqnumber: 0, len: 11, data: <Buffer 18 21 0a 00 00 10 01 00 00 10 00> }
Mon, 28 May 2018 10:01:53 GMT zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
Mon, 28 May 2018 10:01:53 GMT cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 13780, srcendpoint: 2, dstendpoint: 1, wasbroadcast: 0, linkquality: 55, securityuse: 0, timestamp: 7393851, transseqnumber: 0, len: 11, data: <Buffer 18 22 0a 00 00 10 01 00 00 10 00> }
Mon, 28 May 2018 10:01:53 GMT zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
Mon, 28 May 2018 10:01:54 GMT cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 6, srcaddr: 13780, srcendpoint: 3, dstendpoint: 1, wasbroadcast: 0, linkquality: 55, securityuse: 0, timestamp: 7395353, transseqnumber: 0, len: 11, data: <Buffer 18 23 0a 00 00 10 01 00 00 10 00> }
Mon, 28 May 2018 10:01:54 GMT zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0`

Is there a way to depair devices again so that I can repair it and see if that fixes it? Simply trying to start the paring process from the switch outputs ".. is already joined", deleting the switch from the database.db and config only results in the same error as above again.

Koenkk, 2018-05-28:
Can you provide the whole log when running with `DEBUG=* npm start` after pressing some buttons?

ciB89, 2018-05-28:
When I manually added the switch to the configuration.yaml then this is all it does when pressing the buttons on the switch, it just keeps repeating itself depending on what I press (srcendpoint: 1 is left button, 2 is right, 3 is both). If I don't add it manually to the config the zigbee2mqtt crashes before I can press anything.

Koenkk, 2018-05-28:
It is strange that you manually have to add the configuration.yaml, can you post the log of this crash?

ciB89, 2018-05-28:
I already posted it above. Here is it again:

```
2018-5-27 22:23:40 INFO Connected to MQTT server
2018-5-27 22:23:40 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-5-27 22:23:41 INFO MQTT publish, topic: 'homeassistant/sensor/0x00158d0002486c1c/action/config', payload: '{"icon":"mdi:cube","value_template":"{{ value_json.action }}","json_attributes":["battery"],"state_topic":"zigbee2mqtt/cube","availability_topic":"zigbee2mqtt/bridge/state","name":"cube"}'
/app/lib/homeassistant.js:238
    const friendlyName = settings.getDevice(deviceID).friendly_name;
                                                     ^
TypeError: Cannot read property 'friendly_name' of undefined
    at discover (/app/lib/homeassistant.js:238:54)
    at Object.discover (/app/lib/homeassistant.js:267:42)
    at zigbee.getAllClients.forEach (/app/lib/controller.js:59:35)
    at Array.forEach (<anonymous>)
    at Controller.handleStarted (/app/lib/controller.js:57:41)
    at mqtt.connect (/app/lib/controller.js:48:85)
    at MqttClient.client.on (/app/lib/mqtt.js:26:13)
    at emitOne (events.js:121:20)
    at MqttClient.emit (events.js:211:7)
    at MqttClient._handleConnack (/app/node_modules/mqtt/lib/client.js:891:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-05-27T20_23_41_064Z-debug.log
```

Koenkk, 2018-05-28:
Sorry, the commit above fixes the crash. It happens when the device is paired but no message is send yet. Can you update zigbee2mqtt?

ciB89, 2018-05-28:
Unfortunately im running Hassio, so I guess I have to wait until fixes are pushed to the other repo - right?

Koenkk, 2018-05-28:
For now yes.

Offtopic: we are configuring a way to do this automatically in: https://github.com/danielwelch/hassio-zigbee2mqtt/issues/19.

ciotlosm, 2018-06-15:
@ciB89 was the problem fixed? By now the fixes should have arrived in hassio addon. If it works now, can you please close the issue?

ciB89, 2018-06-15:
Yeah, it works. Changing the battery did the trick. Apparently the battery was too low to pair successfully (although this was not obvious, thanks for the tip @Koenkk

closed 2018-06-15 by ciB89

Koenkk, 2018-06-15:
Updated the wiki: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices#xiaomi, thanks @ciB89

sknsean, 2018-12-03:
Hi
I'm getting "  zigbee2mqtt:error 2018-12-3 20:03:58 Cannot get the Node Descriptor of the Device: 0x00158d00026ba1da (Error: Timed out after 10000 ms)" when trying to pair this device. I'm trying what is described here in the thread

Koenkk, 2018-12-05:
@sknsean please:

- stop zigbee2mqtt
- remove all entries of `0x00158d00026ba1da ` from `database.db`
- start zigbee2mqtt
- try to pair again

giggo, 2018-12-06:
I can confirm that changing batteries did wonders and pairing worked.

jenicek001, 2020-06-13:
I had serious problems with pairing with WXKG02LM. I tried:
- to pair with original Xiaomi GW via Mi Home App - no success
- to pair with CC2531 - no success
- use USB extender cable to bring CC2531 to 5cm distance - no success
- replace original battery w 3,001V by new one with 3,31V - no success
- dismount from wall (damaged wall - original stisker is very strong :( ) - no success
- again remove battery and let battery removed for a while
- reinsert the battery
- hold (left) button - after 10s of holding the button till both LEDs started blinking - now finally the left one kept active (before both LEDs flashed 5-times and nothing happened) - now the CC2531 paired flawlessly
- in this case no matter how far from CC2531 I was during new pairing

AntoineTESTE, 2020-12-22:
Hello friends.

For those of you who have had trouble pairing the switch.
Here's a trick that might help you.
During the battery change operation, after removing it, I pressed the little red button (on one of the switches) under the battery, a blue light flashed, and probably discharged the circuit. I then put the battery back in and it worked

polsup2, 2021-05-02:
> Hello friends.
>
> For those of you who have had trouble pairing the switch.
> Here's a trick that might help you.
> During the battery change operation, after removing it, I pressed the little red button [...]

This advice worked for me! I almost lost hope after trying all previous checklists with batteries, distance etc)

After discharging the circuit by pressing the button it pairs immediately.

Hardware: aqara 2 button switch wxkg07lm. Coordinator rf-star 2652.  
Software: z2m 1.18.1-1, hassio on rasp pi 3

airens, 2021-08-28:
> Hello friends.
>
> For those of you who have had trouble pairing the switch.
> Here's a trick that might help you.
> During the battery change operation, after removing it, I pressed the little red button [...]

Thanks A LOT, I've been trying to pair it for an hour until I tryed your trick!

zakamancode, 2021-09-19:
> Hello friends.
>
> For those of you who have had trouble pairing the switch.
> Here's a trick that might help you.
> During the battery change operation, after removing it, I pressed the little red button [...]

That was the missing step for pairing,thank you!

ShafiqUrRehman, 2024-05-26:
Reading this thread in 2024 :) 
Following works for me

0. Step Zero (Very Important): Remove the switch from any hub or Zigbee2MQTT (z2m). I force-removed the switch from z2m.
1. Remove the battery for 30 seconds.
2. Single-press the scene button next to the battery compartment. The LED may blink for a second. For me, sometimes the LED blinks and other times it doesn't.
3. Put the battery back in.
4. Press and hold the left button. The led should turn on and should stay on. Do not release it until step 7 below.
5. After 10 seconds, both LEDs will blink 3-4 times, and then the left LED will blink a few times.
6. Keep holding the button. After another 10 seconds, both LEDs will blink 3-4 times again.
7. Now release the button.
8. The left led will blink few times.
9. In the next 5 - 10 seconds, you should see it pairing.