Koenkk/zigbee2mqtt #65: Error when connected to Broker

Odianosen25, 2018-05-25:
Hello @Koenkk ,

Sorry of late I get the following error spat out once it connected to my broker

```
2018-5-25 18:56:40 INFO Connecting to MQTT server at mqtt://192.168.0.30
2018-5-25 18:56:40 INFO zigbee-shepherd ready
2018-5-25 18:56:40 INFO Connected to MQTT server
2018-5-25 18:56:40 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-5-25 18:56:40 WARN `permit_join` set to  `true` in configuration.yaml.
2018-5-25 18:56:40 WARN Allowing new devices to join.
2018-5-25 18:56:40 WARN Set `permit_join` to `false` once you joined all devices.
2018-5-25 18:56:40 INFO Zigbee: allowing new devices to join.
2018-5-25 18:56:41 INFO
        {"message":"request unsuccess: 16","stack":"Error: request unsuccess: 16\n    at /app/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:257:13)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr (/app/node_modules/cc-znp/lib/ccznp.js:357:14)\n    at /app/node_modules/cc-znp/lib/ccznp.js:333:22\n    at _combinedTickCallback (internal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/next_tick.js:180:9)"}
```
I have 11 devices so far, and it seems I can't connect any more.

Also how does one remove a device to repair? I have a device which isn't sending any messages. Ehen i try to repair, i get the message that it already exist.

Thanks and regards

ciotlosm, 2018-05-25:
@Odianosen25 using localhost won't work. If you are using hassio addon you should be using  mqtt://homeassistant as that DNS is available and should work (this should be fixed by https://github.com/danielwelch/hassio-zigbee2mqtt/pull/17) 

There is already an issue for automatic removal from database here: https://github.com/Koenkk/zigbee2mqtt/issues/35 but I think there should be a way to do this manually by editing files inside your `/share/zigbee2mqtt` folder.

Odianosen25, 2018-05-25:
@ciotlosm, Thanks for the response.

I am not using the Hassio but just docker, but I get your point.

Will try the remove, maybe it will solve my problem but I know not yet.

Regards

ciotlosm, 2018-05-25:
@Odianosen25 even without hassio when using docker, unless you start the container with `--net=host` you have a different network. If you start your `homeassistant` container using a static name using `--name` like `myhome` you should be able to use `mqtt://myhome` in the other containers.

Odianosen25, 2018-05-25:
@ciotlosm But outside the network thing, what do you think the error I am getting is all about?

Regards

ciotlosm, 2018-05-25:
@Odianosen25 not that familiar with the code but I'll try and have a look. In the meantime for everyone else reference, here is the error expanded:

```
 Error: request unsuccess: 16
    at /app/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21
    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:257:13)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at CcZnp._mtIncomingDataHdlr (/app/node_modules/cc-znp/lib/ccznp.js:357:14)
    at /app/node_modules/cc-znp/lib/ccznp.js:333:22
    at _combinedTickCallback (internal/process/next_tick.js:131:7)
    at process._tickCallback (internal/process/next_tick.js:180:9)"}
```

Odianosen25, 2018-05-26:
Hello @ciotlosm and @Koenkk,

I have taken the time to repair all my devices, so I could possible track down what is going on with the errors. All I did was to delete my database, and as I pair each device that doesn't give me an error, I copy the files and keep somewhere.

I noticed the following
- This error was generated when pairing some of my devices. This cuts across different ones like the WXKG02LM, WSDCGQ11LM, RTCGQ11LM, MCCGQ11LM. I got just 11 devices and it affected 5 of them
- If the error was generated, it is only noticeable if one was to stop and start the bridge again
- Pairing was still possible, but if one was to restart the bridge, after about 2 devices that generated that error has been added, pairing is not possible; so will have to revert back to a working database. The system still works though

I have paired everything and working fine. Just that I fear that as I buy more devices, might have to start the process again if it can't accept new devices due to this error.

Let me know if you want more info.

Regards

Koenkk, 2018-05-26:
So everything is OK now?

Odianosen25, 2018-05-26:
Well for now yes, until I try to add something else,m then I will be very sure. Thanks

Koenkk, 2018-05-26:
Ok, feel free to reopen.

Odianosen25, 2018-05-26:
@Koenkk  sorry can you re-open?

Just figured out I didn't pair one of my devices, and as I noted above, as I am now getting the following errors
```
{"message":"request unsuccess: 16","stack":"Error: request unsuccess: 16\n    at /app/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21\n    at CcZnp.<anonymous> (/app/node_modules/cc-znp/lib/ccznp.js:257:13)\n    at Object.onceWrapper (events.js:315:30)\n    at emitOne (events.js:116:13)\n    at CcZnp.emit (events.js:211:7)\n    at CcZnp._mtIncomingDataHdlr (/app/node_modules/cc-znp/lib/ccznp.js:357:14)\n    at /app/node_modules/cc-znp/lib/ccznp.js:333:22\n    at _combinedTickCallback (internal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/next_tick.js:180:9)"}
```
I can no longer add any device.

Please any help?

Koenkk, 2018-05-26:
Is a specific device causing this issue?

Koenkk, 2018-05-27:
This error seems to happen when the CC2531 runs out of heap. 

@Odianosen25 I've compiled a firmware with an increased heap size, down side is that it only allows 40 devices to be connected. (instead of 48). https://github.com/Koenkk/zigbee2mqtt-firmware/blob/78c3d89d28db07554046c6905f53050a0638244e/coordinator/Z-Stack%20Home%201.2.2a.44539/Projects/zstack/ZNP/CC253x/dev/CC2531ZNP-Prod.hex

Odianosen25, 2018-05-27:
@Koenkk,

Ok thanks. Will test it and let you know what's up. Well I have no issues with the 40 devices, as I am always willing to increase the number of CC2531. But anyways will get back.

Odianosen25, 2018-05-28:
@Koenkk,

Many thanks for the update, and that solved my issues so far. I was able to add the new devices, and it no longer spat out all those errors.

Though u said its now limited to just 40 devices, but I have noticed you mentioned in the forum adding routers like electrically powered devices increases the number. So maybe I will go that route when the time comes.

I will keep monitoring it, and report if there are any new issues. Closing this for now.

ciotlosm, 2018-05-28:
Would be nice to get updated wiki docs with this setup and limitations around it

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Odianosen25, 2018-05-28:
@ciotlosm supported

Odianosen25, 2018-05-28:
Yes it sure did thanks

Koenkk, 2018-05-28:
Thanks! This is now the default firmware.

Koenkk, 2018-06-01:
@akileos can you try with the following firmware: https://raw.githubusercontent.com/Koenkk/zigbee2mqtt-firmware/master/coordinator/Z-Stack%20Home%201.2.2a.44539/Projects/zstack/ZNP/CC253x/dev/CC2531ZNP-Prod-test.hex ?

akileos, 2018-06-01:
Will do and report back !

Koenkk, 2018-06-01:
Moved firmware to: https://github.com/Koenkk/zigbee2mqtt-firmware/tree/master/firmware/test/coordinator/CC2531

akileos, 2018-06-02:
Error disappeared with new firmware, seem to work as it was before. 
Don't have more devices at hand to find the new limit.
Thanks.

Odianosen25, 2018-06-02:
What’s the new limit @koenkk?

Koenkk, 2018-06-02:
35 devices

Koenkk, 2018-06-02:
Ok, this firmware is now the default one.

Koenkk, 2018-06-02:
I've updated: https://github.com/Koenkk/zigbee2mqtt/wiki/ZigBee-network#network-size with the new limit.

closed 2018-06-02 by Koenkk

ciotlosm, 2018-06-02:
@Odianosen25 please open specific issues for enhancements to avoid bloating closed issues.

Odianosen25, 2018-06-02:
Oh ok boss @ciotlosm

jay-jey, 2018-06-12:
Hello, I've now getting the same arrow as Odianosen25 but I use the new firmware. Did I missed something in this conversation, when I understood it correct the  issue should be solved with the new firmware, or? 
My bridge worked for some time and I could also add an controll some devices but since yesterday I get: 

`2018-06-12T04:53:28.800Z - info: Starting zigbee-shepherd
2018-06-12T04:53:30.189Z - info: zigbee-shepherd started
2018-06-12T04:53:30.219Z - info: Currently 6 devices are joined:
2018-06-12T04:53:30.223Z - info: 0x00158d000209002a (0x00158d000209002a): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-06-12T04:53:30.227Z - info: 0x84182600000d08ce (0x84182600000d08ce): AB3257001NJ - OSRAM Smart+ plug (Router)
2018-06-12T04:53:30.249Z - info: 0x000b57fffeaf4080 (0x000b57fffeaf4080): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
2018-06-12T04:53:30.251Z - info: 0x00158d000227b1dd (0x00158d000227b1dd): WSDCGQ01LM - Xiaomi MiJia temperature & humidity sensor  (EndDevice)
2018-06-12T04:53:30.254Z - info: 0x000b57fffe93146d (0x000b57fffe93146d): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
2018-06-12T04:53:30.256Z - info: 0x000b57fffe88f5d1 (0x000b57fffe88f5d1): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
2018-06-12T04:53:30.258Z - warn: `permit_join` set to  `true` in configuration.yaml.
2018-06-12T04:53:30.260Z - warn: Allowing new devices to join.
2018-06-12T04:53:30.262Z - warn: Set `permit_join` to `false` once you joined all devices.
2018-06-12T04:53:30.264Z - info: Zigbee: allowing new devices to join.
2018-06-12T04:53:30.272Z - info: Connecting to MQTT server at mqtt://localhost
2018-06-12T04:53:30.891Z - info: zigbee-shepherd ready
2018-06-12T04:53:30.929Z - info: Connected to MQTT server
2018-06-12T04:53:30.932Z - info: MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-06-12T04:53:30.962Z - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000209002a/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d000209002a","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d000209002a_temperature","unique_id":"0x00158d000209002a_temperature_zigbee2mqtt"}'
2018-06-12T04:53:30.965Z - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000209002a/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d000209002a","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d000209002a_humidity","unique_id":"0x00158d000209002a_humidity_zigbee2mqtt"}'
2018-06-12T04:53:30.968Z - info: MQTT publish, topic: 'homeassistant/switch/0x84182600000d08ce/switch/config', payload: '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x84182600000d08ce/set","state_topic":"zigbee2mqtt/0x84182600000d08ce","availability_topic":"zigbee2mqtt/bridge/state","name":"0x84182600000d08ce","unique_id":"0x84182600000d08ce_switch_zigbee2mqtt"}'
2018-06-12T04:53:30.971Z - info: MQTT publish, topic: 'homeassistant/light/0x000b57fffeaf4080/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0x000b57fffeaf4080/set","state_topic":"zigbee2mqtt/0x000b57fffeaf4080","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000b57fffeaf4080","unique_id":"0x000b57fffeaf4080_light_zigbee2mqtt"}'
2018-06-12T04:53:30.974Z - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000227b1dd/temperature/config', payload: '{"unit_of_measurement":"°C","icon":"mdi:temperature-celsius","value_template":"{{ value_json.temperature }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d000227b1dd","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d000227b1dd_temperature","unique_id":"0x00158d000227b1dd_temperature_zigbee2mqtt"}'
2018-06-12T04:53:30.977Z - info: MQTT publish, topic: 'homeassistant/sensor/0x00158d000227b1dd/humidity/config', payload: '{"unit_of_measurement":"%","icon":"mdi:water-percent","value_template":"{{ value_json.humidity }}","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/0x00158d000227b1dd","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d000227b1dd_humidity","unique_id":"0x00158d000227b1dd_humidity_zigbee2mqtt"}'
2018-06-12T04:53:30.979Z - info: MQTT publish, topic: 'homeassistant/light/0x000b57fffe93146d/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0x000b57fffe93146d/set","state_topic":"zigbee2mqtt/0x000b57fffe93146d","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000b57fffe93146d","unique_id":"0x000b57fffe93146d_light_zigbee2mqtt"}'
2018-06-12T04:53:30.982Z - info: MQTT publish, topic: 'homeassistant/light/0x000b57fffe88f5d1/light/config', payload: '{"brightness":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0x000b57fffe88f5d1/set","state_topic":"zigbee2mqtt/0x000b57fffe88f5d1","availability_topic":"zigbee2mqtt/bridge/state","name":"0x000b57fffe88f5d1","unique_id":"0x000b57fffe88f5d1_light_zigbee2mqtt"}'
2018-06-12T04:53:31.174Z - info:  Error: request unsuccess: 17
[... cut ...]
2018-06-12T04:53:50.403Z - info: Zigbee publish to '0x000b57fffe88f5d1', genOnOff - on - {} - null
2018-06-12T04:53:51.341Z - info: Zigbee publish to '0x000b57fffeaf4080', genOnOff - on - {} - null
2018-06-12T04:53:52.282Z - info: Zigbee publish to '0x84182600000d08ce', genOnOff - on - {} - null
2018-06-12T04:54:20.281Z - info: Succesfully configured 0x84182600000d08ce
2018-06-12T04:54:28.536Z - info: Zigbee publish to '0x84182600000d08ce', genOnOff - on - {} - null
2018-06-12T04:54:31.669Z - info: Zigbee publish to '0x84182600000d08ce', genOnOff - off - {} - null
`

ciotlosm, 2018-06-12:
@jay-jey please use hastebin for such large log files. 
@Koenkk maybe allow access for me to help with cleanup issues (tagging, updating code like the one above)

https://hastebin.com/anisesumog.scala

@jay-jey the error is not the same. You have connection to your stick. The problem is here:
```
2018-06-12T04:53:31.174Z - info: Error: request unsuccess: 17 
at /opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/zdo.js:46:21 
at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:257:13) 
at Object.onceWrapper (events.js:315:30) 
at emitOne (events.js:116:13) 
at CcZnp.emit (events.js:211:7) 
at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:357:14) 
at /opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:333:22 
at _combinedTickCallback (internal/process/next_tick.js:131:7) 
at process._tickCallback (internal/process/next_tick.js:180:9)
```

jay-jey, 2018-06-12:
okay, thanks for the link to hastebin, I tried to find something to put in the code a little bit better but I couldn't find anything. 
So if this is not the same problem, should I open a new issue?

ciotlosm, 2018-06-12:
@jay-jey this is different issue, please open a new one