Koenkk/zigbee2mqtt #66: Visonic MCT-340 E Door Sensor

oakbrad, 2018-05-26:
I have a couple of these
https://www.amazon.com/Visonic-MCT-340-Wireless-Samsung-SmartThings/dp/B06XDJ3KYC

When trying to pair:

```
6 20:25:02 INFO New device joined the network!
6 20:25:26 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000af05332 (Error: Timed out after 10000 ms)
: true [ { func: [Function: func], ieeeAddr: '0x000d6f000af05332' } ]
in joinqueue
6 20:26:46 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000af05332 (Error: Timed out after 10000 ms)
6 20:27:06 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000af05332 (Error: Timed out after 10000 ms)
: true []
``

Koenkk, 2018-05-27:
Did you factory reset the device?

oakbrad, 2018-05-27:
I've tried resetting it many times, it will start the pairing process over again but gives the same error.

Koenkk, 2018-05-28:
I get this error when I keep pressing the reset button on Xiaomi Aqara devices after holding the reset button for 5 seconds. Can you check different combinations of button hold/press when pairing?

oakbrad, 2018-06-03:
I will keep trying, the documentation I have for it says hold down for 2 seconds to reset but I am having no luck - I have a few of these and can't get any of them to pair.

ciotlosm, 2018-06-03:
@oakbrad you can try https://github.com/Koenkk/zigbee2mqtt/issues/90 details to see if it helps on how things should look when pairing difficult devices.

oakbrad, 2018-06-03:
I think I got one to connect. It appears the magic combo might be fresh zigbee2mqtt start, and holding down the button while inserting the battery.

I am still getting the time out error though
```
 18:34:19 INFO Connecting with device, please wait...
 18:34:19 INFO New device joined the network!
 18:34:19 INFO New device with address 0x000d6f000b02ceca connected!
 18:34:19 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000b02ceca"}'
: false []
: true []
 18:34:43 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000b02ceca (Error: Timed out after 10000 ms)
in joinqueue
: true []
 18:35:23 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000b02ceca (Error: Timed out after 10000 ms)
```

It does end up in my configuration.yaml
```
  '0x000d6f000b02ceca':
    friendly_name: '0x000d6f000b02ceca'
    retain: false
```

And the pairing light on the device stops blinking.

Let me see if I can replicate pairing with another one..

oakbrad, 2018-06-03:
Hold the button for 2 seconds while inserting the battery is definitely the pairing mechanism. Got 3/4 devices paired all with the same result:

` 18:44:36 INFO New device joined the network!
 18:44:36 INFO New device with address 0x000d6f000b3e43d1 connected!
 18:44:36 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x000d6f000b3e43d1"}'
: false []
: true []
 18:45:03 ERROR Cannot get the Node Descriptor of the Device: 0x000d6f000b3e43d1 (Error: Timed out after 10000 ms)`

All of them in the YAML
`  '0x000d6f000b02ceca':
    friendly_name: '0x000d6f000b02ceca'
    retain: false
  '0x000d6f000af05332':
    friendly_name: '0x000d6f000af05332'
    retain: false
  '0x000d6f000b3e43d1':
    friendly_name: '0x000d6f000b3e43d1'
    retain: false
`

ciotlosm, 2018-06-03:
@oakbrad try to document your setup for wiki to help other users with similar devices.

ciotlosm, 2018-06-15:
@oakbrad Are you happy to close this issue? 

To get more information when trying to pair you can always use the [documentation](https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug). 

I had similar problems when trying to pair a cube that was previously paired and the solution was to start over by removing batter, reset it and start over. Always had to press link button every second after `spinlock` to make sure it sends needed information.

oakbrad, 2018-06-18:
@ciotlosm I got the 4 of them paired, they appear in my configuration.yaml as I posted above, but they do not actually do anything or initialize as devices.  All I see in the logs related to them is timeout errors.

Not sure where to go from here?

ciotlosm, 2018-06-19:
@oakbrad can you try and start in DEBUG mode? `DEBUG=* npm start` 
Use hastebin to post results.

ciotlosm, 2018-06-29:
@oakbrad when pairing if you get timeout it's usually because you didn't keep pressing a button or something to keep them sending zigbee packages to make them complete pairing.

Koenkk, 2018-06-29:
Can you post the `database.db` entry of this device?

ajayjohn, 2018-09-09:
@oakbrad @Koenkk @ciotlosm 
Was this issue every resolved? I tried to use the same device and encountered identical errors to what has been reported here.

Is there anything I can provide here, to help troubleshoot the issue? I tried to open the database.db file using an SQLite browser, but it asks me for an SQLCipher encryption password.
Given below are by logs from the Hassio Addon I use.

`[Info] Configuration file found. Will overwrite configurable               fields with values from add-on configuration
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
[2018-09-08T21:59:14.389Z] PM2 log: Launching in no daemon mode
[2018-09-08T21:59:14.459Z] PM2 log: Starting execution sequence in -fork mode- for app name:npm id:0
[2018-09-08T21:59:14.468Z] PM2 log: App name:npm id:0 online
> zigbee2mqtt@0.1.3 start /app
> node index.js
  zigbee2mqtt:info 2018-9-8 21:59:15 Logging to directory: '/share/zigbee2mqtt/log/2018-09-08.21:59:15'
  zigbee2mqtt:info 2018-9-8 21:59:15 Starting zigbee2mqtt version 0.1.3 (commit #c05b70b)
  zigbee2mqtt:info 2018-9-8 21:59:15 Starting zigbee-shepherd
  zigbee2mqtt:info 2018-9-8 21:59:15 zigbee-shepherd started
  zigbee2mqtt:info 2018-9-8 21:59:15 Coordinator firmware version: '20180507'
  zigbee2mqtt:info 2018-9-8 21:59:15 Currently 2 devices are joined:
  zigbee2mqtt:info 2018-9-8 21:59:15 0x8418260000037707 (0x8418260000037707): 73740 - Sylvania LIGHTIFY LED adjustable white BR30 (Router)
  zigbee2mqtt:info 2018-9-8 21:59:15 0x00158d0002136055 (0x00158d0002136055): WXKG11LM - Xiaomi Aqara wireless switch (EndDevice)
  zigbee2mqtt:warn 2018-9-8 21:59:15 `permit_join` set to  `true` in configuration.yaml.
  zigbee2mqtt:warn 2018-9-8 21:59:15 Allowing new devices to join.
  zigbee2mqtt:warn 2018-9-8 21:59:15 Set `permit_join` to `false` once you joined all devices.
  zigbee2mqtt:info 2018-9-8 21:59:15 Zigbee: allowing new devices to join.
  zigbee2mqtt:info 2018-9-8 21:59:15 Connecting to MQTT server at mqtt://homeassistant
  zigbee2mqtt:info 2018-9-8 21:59:15 zigbee-shepherd ready
  zigbee2mqtt:info 2018-9-8 21:59:15 Connected to MQTT server
  zigbee2mqtt:info 2018-9-8 21:59:15 MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  zigbee2mqtt:info 2018-9-8 21:59:15 MQTT publish, topic: 'homeassistant/light/0x8418260000037707/light/config', payload: '{"brightness":true,"color_temp":true,"platform":"mqtt_json","command_topic":"zigbee2mqtt/0x8418260000037707/set","state_topic":"zigbee2mqtt/0x8418260000037707","availability_topic":"zigbee2mqtt/bridge/state","name":"0x8418260000037707","unique_id":"0x8418260000037707_light_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-9-8 21:59:15 MQTT publish, topic: 'homeassistant/sensor/0x00158d0002136055/click/config', payload: '{"icon":"mdi:toggle-switch","value_template":"{{ value_json.click }}","json_attributes":["battery","voltage","action","duration"],"force_update":true,"state_topic":"zigbee2mqtt/0x00158d0002136055","availability_topic":"zigbee2mqtt/bridge/state","name":"0x00158d0002136055","unique_id":"0x00158d0002136055_click_zigbee2mqtt"}'
  zigbee2mqtt:info 2018-9-8 21:59:15 MQTT publish, topic: 'zigbee2mqtt/0x8418260000037707', payload: '{"state":"OFF","brightness":255,"color_temp":370}'
  zigbee2mqtt:info 2018-9-8 21:59:15 MQTT publish, topic: 'zigbee2mqtt/0x00158d0002136055', payload: '{"battery":"100.00","voltage":3045,"linkquality":97}'
  zigbee2mqtt:info 2018-9-8 21:55:17 Device incoming...
  zigbee2mqtt:info 2018-9-8 21:55:17 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 2018-9-8 21:56:01 Message without device!
  zigbee2mqtt:info 2018-9-8 21:56:01 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:01 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:02 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:02 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:03 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:03 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:03 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:04 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:04 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:04 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:05 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:05 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:05 Connecting with device...
  zigbee2mqtt:info 2018-9-8 21:56:05 MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee2mqtt:info 2018-9-8 21:56:05 Device incoming...
[... cut ...]
  zigbee2mqtt:error 2018-9-8 21:57:21 Cannot get the Node Descriptor of the Device: 0x000d6f000b11d7bb (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-9-8 21:57:41 Cannot get the Node Descriptor of the Device: 0x000d6f000b11d7bb (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-9-8 21:58:01 Cannot get the Node Descriptor of the Device: 0x000d6f000b11d7bb (Error: Timed out after 10000 ms)
  zigbee2mqtt:error 2018-9-8 21:58:21 Cannot get the Node Descriptor of the Device: 0x000d6f000b11d7bb (Error: Timed out after 10000 ms)`

Koenkk, 2018-09-10:
The `database.db` can be opened with a text editor.

ajayjohn, 2018-09-10:
@Koenkk Thanks! I checked and it seems this device has not created any db entry yet. All my other successfully paired devices are shown correctly, including the USB coordinator itself.

oakbrad, 2018-09-12:
Sorry all I never got these working and ended up giving them to a friend.

When I was troubleshooting it I never saw it create a database.db entry either like @ajayjohn said.

ajayjohn, 2018-10-08:
@Koenkk Just checking to see if you need anything else to debug this issue.

ajayjohn, 2018-10-12:
**Procedure**
1) I turn on the 'permit_join' option in the official hassio add-on you have provided
2) I initiate the pairing sequence on the sensor (MCT-340 E). The sensor is now in pairing mode.
3) On checking the hassio add-on logs, I find that the device has been detected and the following message keeps repeating every few seconds, in the logs,
'zigbee2mqtt:error 2018-10-12 01:28:40 Cannot get the Node Descriptor of the Device: 0x000d6f000b3b89ae (Error: Timed out after 10000 ms)'
4) Nothing further appears in the log, related to this device. On checking the configuration.yaml as well as the database.db, this device does not have any entries.
5) The sensor continues to be in pairing mode and does not show any signs of being successfully paired (based on its LED blinking pattern).

**Footnotes**
This device is currently supported by native zigbee on HA (bellows). The device is also supported natively by other platforms like SmartThings. Finally, the device and all its details can be viewed in deCONZ too.

Koenkk, 2018-10-15:
When pairing, could you provide your log when running with `DEBUG=* npm start`

henworth, 2018-12-21:
I found this issue after having this very same problem pairing a couple sensors. One sensor paired seemingly without issue (ie. it left pairing mode) but the second has not. They both have been added to `configuration.yaml` correctly and I've created a stub entry in `devices.js` for the Visonic devices. 

I've captured the extent of the debug logs, I hope they are helpful: https://hastebin.com/esayotayoz.apache

henworth, 2018-12-21:
I realized that the logs didn't capture the shepherd debug even though that was enabled as well, maybe a quirk with hassio? Anyway, I wasn't able to capture the full set of missed debug messages due to the limited window size of the hassio addon logs in the gui, but I was able to get these blocks

This gets repeated a few times:

```
2018-12-21 18:20:26 - debug: Received zigbee message of type 'devInterview' with data '"0x000d6f000b3dfe2f"'
2018-12-21 18:20:26 - info: Connecting with device...
2018-12-21 18:20:26 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"connecting with device"}'
2018-12-21 18:20:26 - warn: Message without device!
2018-12-21 18:20:26 - zigbee-shepherd:request REQ --> AF:dataRequest, transId: 16
2018-12-21 18:20:26 - zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2018-12-21 18:20:26 - zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2018-12-21 18:20:26 - zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2018-12-21 18:20:26 - zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2018-12-21 18:20:27 - zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2018-12-21 18:20:27 - zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 16
2018-12-21 18:20:27 - zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2018-12-21 18:20:27 - zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2018-12-21 18:20:27 - zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
2018-12-21 18:20:27 - zigbee-shepherd:request REQ --> AF:dataRequest, transId: 17
2018-12-21 18:20:27 - zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0
2018-12-21 18:20:28 - zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
2018-12-21 18:20:28 - zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 17
2018-12-21 18:20:28 - zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object]
2018-12-21 18:20:28 - zigbee-shepherd:msgHdlr IND <-- AF:incomingMsg, transId: 0
2018-12-21 18:20:28 - zigbee-shepherd:af dispatchIncomingMsg(): type: zclIncomingMsg, msg: [object Object]
```

This also happens towards the end:

```2018-12-21 18:20:32 Received zigbee message of type 'devStatus' with data '"online"' of device 'MCT-340 E' (0x000d6f000b3dfe2f)
2018-12-21 18:21:20 zigbee-shepherd Device: 0x000d6f000b3dfe2f already in network
2018-12-21 18:21:20 zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd
2018-12-21 18:21:20 zigbee-shepherd spinlock: false []
2018-12-21 18:21:20 zigbee-shepherd Device: 0x000d6f000b3dfe2f already in network
2018-12-21 18:21:20 zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd
2018-12-21 18:21:30 zigbee-shepherd Device: 0x000d6f000b3dfe2f already in network
2018-12-21 18:21:30 zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd
2018-12-21 18:21:31 zigbee-shepherd spinlock: false []
2018-12-21 18:21:31 zigbee-shepherd Device: 0x000d6f000b3dfe2f already in network
2018-12-21 18:21:31 zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd
```

Koenkk, 2018-12-24:
If I understand correctly, you are able to pair it now but dont receive any messages from the device, is that right?

henworth, 2018-12-25:
Essentially, yes. The first sensor I paired looked like it worked successfully and it shows as online in the database. The second two seemed to work, but don’t leave the pairing the stage based on the device’s led blink pattern. Those two show up as offline in the database.

However, there are no messages from either set of sensors.

Koenkk, 2018-12-27:
Try with the following in `devices.js`

```js
{
        zigbeeModel: ['MCT-340 E'],
        model: 'MCT-340 E',
        vendor: 'Visonic',
        description: 'Wireless door window temperature sensor',
        supports: 'contact and temperature',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb),
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

henworth, 2019-01-01:
Looks like the only sensor that showed online is now not. I think I captured the relevant section of the logs. If not it shows up on every restart so I can get more context if needed. 

```zigbee2mqtt:debug 2019-1-1 14:52:54 Mounted the cieApp (epId {epId})
Tue, 01 Jan 2019 19:52:54 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Tue, 01 Jan 2019 19:52:54 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1
2019-01-01T19:52:54.902Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.'
  zigbee2mqtt:error 2019-1-1 14:52:54 Failed to configure 0x000d6f000b3e0242 0x000d6f000b3e0242
Tue, 01 Jan 2019 19:52:54 GMT zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object]
Tue, 01 Jan 2019 19:52:54 GMT zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 2
2019-01-01T19:52:55.001Z zigbee-shepherd-converters:devices Configured '(cb) => device.write('ssIasZone', 'iasCieAddr', coordinator.device.getIeeeAddr(), cb)' with result 'Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.'```

Koenkk, 2019-01-01:
Configuring probably fails because the device is offline, could you press a button on the startup, this makes sure the device is awake and ready to be configured.

starbuck93, 2019-01-29:
Related to the Visonic MCT-340 E... Let me know if I need to open a new issue versus tagging along with this one.

I just got zigbee2mqtt up and running on my Raspberry Pi with the CC2531 after a few hours of messing around. I got the SmartThings Motion Sensor (2016) working first after messing with devices.js and fromZigbee.js, then paired the Visonic door sensor. I've seen a lot of strange things with this door sensor.

Firstly, when it pairs, the log outputs around 10 `Message without device! ... Connecting with device...` messages. Now it is paired as it shows up in the log, as well as in the `database.db` and `configuration.yaml`. The weird thing about the pairing is that _two_ devices show up in all 3 locations. 
```
Device with modelID 'MCT-340 E' is not supported.
Device with modelID 'undefined' is not supported.
```

They both have unique `0x` addresses. The database shows the [MCT-340 E (excerpt from database.db](https://hastebin.com/obilotorux.json)  and the [`undefined` device](https://hastebin.com/limoyegela.json). You might notice the undefined device is registered as a Coordinator `..."type":"Coordinator"...` Why does zigbee2mqtt think it's the coordinator? How do I see the actual coordinator's `0x` address?

When I open and close the door (triggering zigbee traffic) the logs show the undefined device is the one sending the messages. Turning on debug mode, the traffic is accurate by sending zoneStatus 32 and 33 for open and closed. But it's not the right device sending that data.

`Received zigbee message of type 'statusChange' with data '{"cid":"ssIasZone","zoneStatus":32}' of device 'undefined' (0x00124b0019366619)`

#801 seems relevant but after reading that a lot of their log files are quite different than what I'm seeing over here. I've also read a lot of #176 but I don't entirely remember how I got there anymore.

Koenkk, 2019-01-30:
@starbuck93 can you try with (please re-pair the device afterwards)

```js
{
        zigbeeModel: ['MCT-340 E'],
        model: 'MCT-340 E',
        vendor: 'Visonic',
        description: 'Wireless door window temperature sensor',
        supports: 'contact and temperature',
        fromZigbee: [],
        toZigbee: [],
        configure: (ieeeAddr, shepherd, coordinator, callback) => {
            const device = shepherd.find(ieeeAddr, 1);
            const actions = [
                (cb) => device.functional('ssIasZone', 'enrollRsp', {enrollrspcode: 0, zoneid: 23}, cb),
                (cb) => device.bind('msTemperatureMeasurement', coordinator, cb),
                (cb) => device.report('msTemperatureMeasurement', 'measuredValue', 30, 600, 1, cb),
            ];
            execute(device, actions, callback);
        },
    },
```

starbuck93, 2019-01-30:
@Koenkk Same thing happened again. Stopped `zigbee2mqtt`, deleted from `db` and `yaml`, changed the `devices.js` and repaired the MCT-340.

https://hastebin.com/baxomijuno.makefile

(you'll see some other zigbee traffic here too)

starbuck93, 2019-01-31:
I can still see https://hastebin.com/oyevulobul.sql. But here it is again https://pastebin.com/K9K9GHNP

Koenkk, 2019-02-03:
@starbuck93 what happens when you trigger the MCT-340 E?

starbuck93, 2019-02-03:
Line 1 is door closed and line 2 is door open
https://hastebin.com/kofeyuzizu.pl

Koenkk, 2019-02-03:
Can you try with 

```js
https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/af.js#L596  till 597 with

```js
if(!coord) return;
if (msg.dstendpoint == 11) {
  msg.dstendpoint = 1;
  type = 'zclIncomingMsg`;
}
targetEp = coord.getEndpoint(msg.dstendpoint);               //  => find local ep
```

starbuck93, 2019-02-03:
Steps taken: 
1. stop zigbee2mqtt
2. edit `af.js`, remove entries from db and config
3. take battery out of MCT-340 E
4. start zigbee2mqtt, wait for setup to finish in logs
5. factory reset MCT-340 E
6. pasted logs starting after zigbee2mqtt setup is finished

https://hastebin.com/goxoxobuke.apache

The last few lines are triggering the MCT-340

Koenkk, 2019-02-06:
@starbuck93 sorry for the delay.

Please try with the following af.js: https://raw.githubusercontent.com/Koenkk/zigbee-shepherd/fix/lib/components/af.js

You don't need to repair the device, just update `af.js` and trigger the contact sensor a few times.

starbuck93, 2019-02-06:
@Koenkk no problem! I appreciate all the help you're providing here.

I had to add `../model/group` to the right directory to get it to run after an error, now I'm getting a new error.

```
/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:608
        console.log("Checkpoint 3", JSON.stringify(targetEp));
                                         ^

TypeError: Converting circular structure to JSON
    at JSON.stringify (<anonymous>)
    at dispatchIncomingMsg (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:608:42)
    at Controller.incomingMsgHandler (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/af.js:754:12)
    at Controller.emit (events.js:197:13)
    at Object.bridge._areqEventBridge (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/event_bridge.js:15:16)
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/components/controller.js:105:16)
    at CcZnp.emit (events.js:197:13)
    at CcZnp._mtIncomingDataHdlr (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:370:14)
    at Immediate.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:336:22)
    at processImmediate (timers.js:632:19)
```

Koenkk, 2019-02-06:
@starbuck93 can you try with: https://raw.githubusercontent.com/Koenkk/zigbee-shepherd/fix/lib/components/af.js (updated it)

starbuck93, 2019-02-06:
@Koenkk Another circular structure error, and here's the output https://hastebin.com/ovitoyevit.coffeescript

Koenkk, 2019-02-06:
Updated again: https://raw.githubusercontent.com/Koenkk/zigbee-shepherd/fix/lib/components/af.js

starbuck93, 2019-02-06:
@Koenkk https://hastebin.com/qezotatuxo.js

everything started up just fine! I let the setup finish outputting a ton of log files and then triggered the MCT-340 and that's the output.

Koenkk, 2019-02-06:
@starbuck93 updated https://raw.githubusercontent.com/Koenkk/zigbee-shepherd/fix/lib/components/af.js again (made a typo). Please do the same, previous log is what I need :)

starbuck93, 2019-02-06:
Here it is! https://hastebin.com/ijehuluwew.js with two triggers of the MCT-340

Koenkk, 2019-02-06:
@starbuck93 can you post your complete `database.db`?

starbuck93, 2019-02-06:
Here is `database.db` https://hastebin.com/janipohici.json
with `configuration.yaml` https://hastebin.com/fomucuceku.js just in case

Koenkk, 2019-02-06:
In a log which you provided earlier (https://github.com/Koenkk/zigbee2mqtt/issues/66#issuecomment-458387806, https://hastebin.com/obilotorux.json) the `MCT-340 E` (`0x000d6f000b02c95d`) had a `modelId` in `database.db`. I'm missing this from your latest `database.db`, probably the pairing did not complete correctly? We first need to have this `modelId` back before we can continue.

starbuck93, 2019-02-06:
It looks like it is in `database.db` as `"id":11` now versus being  `"id":3` before

Koenkk, 2019-02-06:
It's is, but it's missing the `modelId`. When searching the old `database.db` for `MCT-340` you will find it, in the new one not.

starbuck93, 2019-02-06:
@Koenkk I'm getting an error, `Failed to mount the cieApp`, now and I'm unsure why. I finally managed to get the MCT-340 E into pairing mode though so that's good (the mfg's instructions are _almost_ correct, I thought I was doing it the whole time!)

Logs: https://hastebin.com/umaluveked.apache

jjlawren, 2019-02-07:
@starbuck93 what were the actual steps you took to pair the device if the instructions were incomplete?

Koenkk, 2019-02-07:
@starbuck93 I still don't see `MCT-340` in the database.db, which I did in:  (https://github.com/Koenkk/zigbee2mqtt/issues/66#issuecomment-458387806, https://hastebin.com/obilotorux.json) . 

Please
- Stop zigbee2mqtt
- Remove all entries of `0x000d6f000b02c95d` from `database.db`
- Start zigbee2mqtt
- Pair device 
- Trigger contact a few times
- Post log here

starbuck93, 2019-02-07:
@jjlawren The docs say to hold the trigger switch down for 4 seconds while inserting the battery but you have to hold it slightly shorter than that to successfully get it to go into pairing.

@Koenkk I'm working on pairing the device again, it's not working as expected. I'm getting `Cannot get the Node Descriptor of the Device` errors

Koenkk, 2019-02-08:
@starbuck93 is the device in `database.db`?

starbuck93, 2019-02-09:
@Koenkk it is in `database.db` now https://hastebin.com/nukisupozi.json (there are two unique MCT-340 Es in there now because I wanted to test if the pairing issue was a device specific thing or not.)

Also I have not solved the pairing issue. The device(s) are still in pairing mode for some reason. Triggering the device(s) only turns them back on pairing mode like this: https://hastebin.com/sigarefici.js

jjlawren, 2019-02-09:
FYI, a similar pairing issue for this device is being investigated over with Deconz, too: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/978

starbuck93, 2019-02-09:
@jjlawren just finished reading the Deconz issue thread. did you manage to factory reset the MCT-340 E after they appear to be stuck in pairing mode? Each of my two devices blink their red LED 3 times every 5 seconds (to show they are in pairing mode) when I do any one of the following: 
* open the battery cover
* push the trigger button once
* bring the magnet close to close the sensor/door
* take the battery out and put it back in
* close the battery cover.

And zigbee2mqtt will try to pair with the device(s) again every time I do any of those things.

jjlawren, 2019-02-09:
@starbuck93 same for me. It seems like the pairing process half-completes, where the controller thinks it's paired but the MCR-340 does not. Factory reset doesn't change the behavior.

There's a change pending over there that I'll test when available.

wixoff, 2019-02-12:
@jjlawren I agree.  I have a dotted-outline MCT-340 E showing up in my zigbee network map, connected to nothing, just floating out there.  The device has no idea it's paired to anything.

These sensors worked perfectly and very reliably for me in that home-assistant zha component, so they may have a solution somewhere too.

wixoff, 2019-02-21:
Is there anything I can do to help move this forward?  It looks like the Deconz folks have it working as of about a week ago, as noted in the thread @jjlawren linked.

Also, I misspoke in my post above -- the MCT-340 E that I tried to pair does have a link to the controller in the zigbee map, but all else is true: it's a dotted-line box and the device doesn't know it's paired.

wixoff, 2019-03-08:
@Koenkk @ciotlosm I have five of these things so I'd like to contribute if I can.  Please let me know if I can -- just let me know what is needed.

[7 more replies]

closed 2019-05-14 by Koenkk