Koenkk/zigbee2mqtt #67: Error while starting zigbee-shepherd!

jay-jey, 2018-05-27:
Hello, 
after I updated the bridge I could not get it working again, I allways get the message  Error while starting zigbee-shepherd!

I started it with debug settings and get:
`pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=* npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-5-27 12:33:56 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022F9F-if00 is opened. +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +118ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +30ms
  serialport write 8 bytes of data +0ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +11ms
2018-5-27 12:33:59 ERROR Error while starting zigbee-shepherd!
2018-5-27 12:33:59 ERROR Failed to start
`
what I also see is that the green led on the stick is off ater starting the bridge.

Koenkk, 2018-05-27:
Can you press the button on the stick which is close to the usb port and start again?

jay-jey, 2018-05-27:
Pressing the button makes no change

Koenkk, 2018-05-27:
What is your `node --version`?

jay-jey, 2018-05-27:
v8.11.2

Koenkk, 2018-05-27:
Did you try?

```
git pull
rm -rf node_modules
npm install
```

jay-jey, 2018-05-27:
yes and I also tried to delet the complete folder and start again from the beginnig, but now there are some changes again, so I will try this again.

Koenkk, 2018-05-27:
It seems that somehow it cannot connect to your CC2531. Can you try a different port?

jay-jey, 2018-05-27:
Hello, 
so it seems that my pi or the configuration is the problem. I tried it now an other pi, and on this it seems to work, but I will keep on testing.

jay-jey, 2018-05-28:
Hello, this morning I've set the homeassistant to true restarted the pi by sudo reboot. And I've got the same error again. The hole night it runs with one sensor connected. Attached the log file. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2043646/log.txt)

Odianosen25, 2018-05-28:
@jay-jey,

WHen using docker, I think you should add the `--rm=true`. This way docker will remove the container, when you end this session. If you don't add it, it will keep the other containers, whom will hook the sniffer and you will therefore not be able to run.

Try the following
- `docker stop $(docker ps -a -q)`
- `docker rm $(docker ps -a -q)`
- `sudo docker run -it -v $(pwd)/data:/app/data --device=/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0012022F9F-if00 --rm=true koenkk/zigbee2mqtt:arm32v6`

jay-jey, 2018-05-28:
Hello, 
thanks for this hint. I'm completly new to docker. The command docker stop $(docker ps -a -q) gives me this:
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.37/containers/json?all=1: dial unix /var/run/docker.sock: connect: permission denied
"docker stop" requires at least 1 argument.
See 'docker stop --help'.

Usage:  docker stop [OPTIONS] CONTAINER [CONTAINER...] [flags]

Stop one or more running containers

But after some google I looked if a container is runnig via docker ps, and this gives me: 

pi@raspberrypi:~ $ sudo docker ps
CONTAINER ID        IMAGE               COMMAND             CREATED             STATUS              PORTS               NAMES

So I thing no container is runnig and when I run then the docker again I get the same result as mentoined above.

ciotlosm, 2018-05-28:
@jay-jey try to make yourself root before running the docker commands above or make sure every docker has sudo in front:
```
sudo su -
```

Odianosen25, 2018-05-28:
@jay-jey,

Still run `docker rm $(docker ps -a -q)` or `sudo docker rm $(docker ps -a -q)`, then try again

jay-jey, 2018-05-28:
Hello, 
I did the docker commands as root. This was the first thing I tried, when I was reading the permission denied. The command ` $ sudo docker rm $(docker ps -a -q)`
gives me
`
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.37/containers/json?all=1: dial unix /var/run/docker.sock: connect: permission denied
"docker rm" requires at least 1 argument.
See 'docker rm --help'.

Usage:  docker rm [OPTIONS] CONTAINER [CONTAINER...] [flags]

Remove one or more containers
`

jay-jey, 2018-05-31:
Hello, 
 thanks for the support with docker, but I thing this not the main problem. the last days I tried an other fork of the project from AndrewLinden, this works good. Today I did a full cleanup, so I deleted the hole directory and installed all again, and there I saw this errors durring the Install: 

``
pi@berrypi:~ $ sudo git clone https://github.com/Koenkk/zigbee2mqtt.git /opt/zigbee2mqtt
Klone nach '/opt/zigbee2mqtt'...
remote: Counting objects: 1130, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 1130 (delta 0), reused 1 (delta 0), pack-reused 1123
Empfange Objekte: 100% (1130/1130), 2.74 MiB | 2.27 MiB/s, Fertig.
Löse Unterschiede auf: 100% (676/676), Fertig.
Prüfe Konnektivität... Fertig.
pi@raspberrypi:~ $ sudo chown -R pi:pi /opt/zigbee2mqtt
pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm install

> serialport@4.0.7 install /opt/zigbee2mqtt/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/bin/node',
node-pre-gyp verb cli   '/opt/zigbee2mqtt/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@8.11.2 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/opt/zigbee2mqtt/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.11.2 (node-v57 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-linux-arm.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
make: Entering directory '/opt/zigbee2mqtt/node_modules/serialport/build'
  CXX(target) Release/obj.target/serialport/src/serialport.o
../src/serialport.cpp: In function ‘int getIntFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:90:52: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToInt32()->Int32Value();
                                                    ^
../src/serialport.cpp: In function ‘double getDoubleFromObject(v8::Local<v8::Object>, std::string)’:
../src/serialport.cpp:102:53: warning: ‘v8::Local<v8::Number> v8::Value::ToNumber() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9760): Use maybe version [-Wdeprecated-declarations]
   return getValueFromObject(options, key)->ToNumber()->NumberValue();
                                                     ^
../src/serialport.cpp: In function ‘void EIO_AfterOpen(uv_work_t*)’:
../src/serialport.cpp:174:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
     int fd = argv[1]->ToInt32()->Int32Value();
                               ^
../src/serialport.cpp:180:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1618) [-Wdeprecated-declarations]
   data->callback.Call(2, argv);
                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:193:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
   int fd = info[0]->ToInt32()->Int32Value();
                             ^
../src/serialport.cpp:217:116: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
   baton->baudRate = Nan::Get(options, Nan::New<v8::String>("baudRate").ToLocalChecked()).ToLocalChecked()->ToInt32()->Int32Value();
                                                                                                                    ^
../src/serialport.cpp: In function ‘void EIO_AfterUpdate(uv_work_t*)’:
../src/serialport.cpp:240:30: warning: ‘v8::Local<v8::Value> Nan::Callback::Call(int, v8::Local<v8::Value>*) const’ is deprecated (declared at ../../nan/nan.h:1618) [-Wdeprecated-declarations]
   data->callback.Call(1, argv);
                              ^
../src/serialport.cpp: In function ‘Nan::NAN_METHOD_RETURN_TYPE Write(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/serialport.cpp:252:29: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
[... cut ...]
../src/serialport_poller.cpp:86:31: warning: ‘v8::Local<v8::Int32> v8::Value::ToInt32() const’ is deprecated (declared at /home/pi/.node-gyp/8.11.2/include/node/v8.h:9796): Use maybe version [-Wdeprecated-declarations]
   obj->fd_ = info[0]->ToInt32()->Int32Value();
                               ^
  SOLINK_MODULE(target) Release/obj.target/serialport.node
  COPY Release/serialport.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/serialport/build'
node-pre-gyp info ok
added 522 packages in 106.897s
``

danielwelch, 2018-05-31:
These node-pre-gyp errors aren’t a problem. Installing node-pre-gyp is annoyingly noisy, but I’ve gotten these errors on every one of my add-on image builds. And if you read the output from node-pre-gyp, it tells you that everything installed correctly if this very, very long message ends with “ok”, which it does in your case.

CrazyMurdok, 2018-06-01:
Hi All! I have also the problem to get it started. I got the following debug output. Any idea?

DEBUG=* npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-1 18:21:34 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM1 is opened. +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +81ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +20ms
  serialport write 8 bytes of data +0ms
  cc-znp:SRSP <-- SYS:osalNvRead, __timeout__ +3s
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +10ms
2018-6-1 18:21:37 ERROR Error while starting zigbee-shepherd!
2018-6-1 18:21:37 ERROR Failed to start

ciotlosm, 2018-06-01:
@CrazyMurdok did you flash your CC2531 with the correct firmware? Is the stick showing up in your /dev/ list?

CrazyMurdok, 2018-06-01:
@ciotlosm Yes it is. It work fine if I use the repository from Oskar (https://github.com/oskarn97/xiaomi-zb2mqtt) and as I understand the shepard repository find the CC2531. Or what does this line mean in the debug output?

cc-znp The serialport /dev/ttyACM1 is opened. +0ms

Firmware on the CC2531 is this one: CC2531ZNP-Pro-Secure_LinkKeyJoin-maximumDevices.hex

Koenkk, 2018-06-01:
@CrazyMurdok can you try the firmware from https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started#2-flashing-the-cc2531-usb-stick?

CrazyMurdok, 2018-06-02:
@Koenkk Thanks for helping. I think my problem was that I was on Jessie and not on Stretch. Now it is running.

jay-jey, 2018-06-04:
Hello, 
So I tried it now on stretch and it seems to work. 
@Koenkk is it possible that jessie is not working? Could you verify this? If Yes, you should make a hint, that it is only working on stretch.

Koenkk, 2018-06-04:
I've added it to the wiki: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge

Can this issue be closed?

CrazyMurdok, 2018-06-04:
@jay-jey You got also solved your problem with upgrading to stretch? Good to hear that we both get our problem solved. 

@Koenkk from my side this issue can now be closed.

Koenkk, 2018-06-04:
@CrazyMurdok @jay-jey thanks for the help!

closed 2018-06-04 by Koenkk