Koenkk/zigbee2mqtt #68: Adding Hive Lightbulb

scottdarby93, 2018-05-27:
Hi, I'm trying to add a Hive lightbulb and got the following

The following is from the log:
`2018-05-27T14:28:08.251Z - warn: Device with modelID 'FWBulb01' is not supported.
2018-05-27T14:28:08.253Z - warn: Please create an issue on https://github.com/Koenkk/zigbee2mqtt/issues to add support for your device`

This is in my database.db
`{"id":2,"type":"Router","ieeeAddr":"0x00158d00011deaa1","nwkAddr":3795,"manufId":4456,"manufName":"Aurora","powerSource":"Mains (single phase)","modelId":"FWBulb01","epList":[1],"status":"online","joinTime":1527431286,"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8],"outClusterList":[25],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"manufacturerName":"Aurora","modelId":"FWBulb01","dateCode":"20170215","powerSource":1,"swBuildId":"2.3"}},"genIdentify":{"dir":{"value":1},"attrs":{"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"nameSupport":0}},"genScenes":{"dir":{"value":1},"attrs":{"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0,"lastCfgBy":"0xffffffffffffffff"}},"genOnOff":{"dir":{"value":1},"attrs":{"onOff":1}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"currentLevel":254,"onOffTransitionTime":0}},"genOta":{"dir":{"value":2},"attrs":{}}}}},"_id":"EeR9HHFAu97yUU8c"}`

I've tried looking for the file to add myself but I'm using Hass.io and can't seam to find it.

Koenkk, 2018-05-27:
Which version do you have, the one that can only do on/off and brigthness? https://www.hivehome.com/products/hive-active-light-dimmable-warm-white

scottdarby93, 2018-05-27:
Thats the one!

Koenkk, 2018-05-27:
Can you check what model number is written on the bulb?

Koenkk, 2018-05-27:
Supported now, can you check if on/off and dimming works?

scottdarby93, 2018-05-27:
Brilliant thank you for updating this so quickly, How do I update my instance so it supports it? I’m running it on Hass.io <http://hass.io/> with the add-on by danielwelch?

Koenkk, 2018-05-27:
@danielwelch can you update the image?

danielwelch, 2018-05-27:
@scottdarby93 @Koenkk the Docker Hub image updates automatically via a Travis script on successful builds to master branch of the add-on.

Since we are still “pre-version”, the best way to update the add on is to uninstall completely and reinstall. Sometimes you need to remove the repository as well and re-add it via the UI. This won’t be a problem once we hit version 0.1 (once there are versioned releases, you can update via the hassio UI). 
I’m away for the weekend, but I think we’ve actually accomplished everything we wanted to before publishing a 0.1 version, so should be imminent.

scottdarby93, 2018-05-27:
thanks @danielwelch, done all of that but it's still coming up the following error `2018-5-27 17:04:19 INFO New device joined the network!
2018-5-27 17:04:19 WARN Device with modelID 'FWBulb01' is not supported.
2018-5-27 17:04:19 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-5-27 17:04:19 WARN Device with modelID 'FWBulb01' is not supported.
2018-5-27 17:04:19 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices`

Any ideas @Koenkk?

Koenkk, 2018-05-27:
@danielwelch are you sure your docker hassio image is updated after every commit? https://hub.docker.com/r/dwelch2101/zigbee2mqtt-armhf/ = 2 days ago.

danielwelch, 2018-05-27:
I see the confusion. I meant pushes to the add on repository triggered a build (last of those was 2 days ago).

Currently, the Dockerfile just git pulls the master branch of this repository. Builds are triggered automatically only when the add on repository changes, not the underlying library.

Many of the add ons in the community repository rely on versioned releases of whatever underlying thing they are exposing. The add on maintainer will update the version in (usually) the Dockerfile, and bump the add-on version, and a new image is ready to go. But since zigbee2mqtt isn’t using versioned releases (totally reasonable given the pace of development), we don’t have that option.

I wonder if some sort of web hook could be used to trigger Travis builds upon pushes here... I’ve never done that sort of cross-repository stuff before so don’t know what’s possible.

Anyways, for now, I just triggered a manual travis build so should be ready soon.

scottdarby93, 2018-05-28:
Thanks guys, it’s working flawlessly now! Just a quick one though, a lot of thee bulbs look like they support transition time? ‘transtime’ is there plans to implement support for this?

Koenkk, 2018-05-28:
@scottdarby93 I've created a new issue for this: https://github.com/Koenkk/zigbee2mqtt/issues/72

closed 2018-05-28 by Koenkk

scottdarby93, 2018-05-31:
Absolutely brilliant!