Koenkk/zigbee2mqtt #70: Xiaomi RTCGQ11LM motion sensor

Odianosen25, 2018-05-28:
Hello @Koenkk,

Sometimes the Xiaomi RTCGQ11LM sends a occupancy true and false like literally at the same time as shown below. 

```
2018-5-28 06:42:54 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":true}'
2018-5-28 06:43:54 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":false}'
```
Any ideas  why?

Koenkk, 2018-05-28:
Can you provide the log of zigbee2mqtt when this happens?

Odianosen25, 2018-05-28:
Hello @Koenk,

Looking via the logs at the same time this happened, I just noticed that actually, your system sends a `False` first, then a `True`. But for some reason, my system picks up the `True` first before the `False`

This is the same log, at the same time the above happened 
```
2018-05-28T07:41:03.285Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":false,"battery":"100.00"}'
2018-05-28T07:41:03.317Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":true,"battery":"100.00"}'
2018-05-28T07:41:15.526Z - info: MQTT publish, topic: 'zigbee2mqtt/Master Bedroom Wall Switch', payload: '{"click":"single"}'
2018-05-28T07:41:33.319Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":false,"battery":"100.00"}
```

Hmmm not sure how to handle this. But if I may ask, why are the two being sent at the same time?

Koenkk, 2018-05-28:
Can you post the complete log? Because for a `false` it first has to send a `true`.

Odianosen25, 2018-05-28:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2045843/log.txt)

There it is boss

Odianosen25, 2018-05-28:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2045848/log.txt)

There you are boss.

Koenkk, 2018-05-28:
```
2018-05-28T17:20:44.836Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":11,"occupancy":false,"battery":"100.00"}'
2018-05-28T17:39:09.012Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":false,"battery":"100.00"}'
2018-05-28T17:39:09.027Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":true,"battery":"100.00"}'
2018-05-28T17:39:39.034Z - info: MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0,"occupancy":false,"battery":"100.00"}'
```

This explains it. Because of the illuminance change a new message is send (and all other previous attributes are send with it). 

This is required because home assistant requires all attributes to be send.

Odianosen25, 2018-05-28:
Ok thanks and I think I understand the process now. As where it is, when one enters the area, the amount of light that gets to it can be affected. So in the process of updating the illuminance, its also sending data for occupancy. So both data at the same time, separately. 

Well if am right, that's not fun really. But will have to leave with it for now, until I can reason a better way to handle it. Closing for now, thanks

Koenkk, 2018-05-28:
I've added an option to disable this behaviour. Can you try adding to configuration.yaml:

```
advanced:
  cache_state: false
```

Odianosen25, 2018-05-28:
@Koenkk,

Oh thanks. So to use this new system, do I need to reload the docker image or just adding it, it just works?

Regards

Koenkk, 2018-05-28:
See https://github.com/Koenkk/zigbee2mqtt/wiki/Docker#updating

ciotlosm, 2018-05-28:
@Koenkk if we use `homeassistant: true` is it mandatory this setting?:
```
advanced:
  cache_state: false
```

Koenkk, 2018-05-28:
No, this is optional: https://github.com/Koenkk/zigbee2mqtt/wiki/Configuration and true by default.

Odianosen25, 2018-05-30:
Hello @Koenkk,

Apologies I haven't gotten back to you on this, been busy with stuffs so haven't had the time to test. I am still testing the config, and setting it to `false`, it still happened. Now testing with it `true`, but want it tested for a few days before coming to a conclusion.

On another note, I noticed that if the motion sensor activates due to motion, even if someone was to be within the same area, it will still send a false then a true immediately. So the light goes off then on. Is it possible for it to like `reset` its internal timer? So it will not send a `occupancy: false` unless it doesn't notice any motion within the set timer; which in my case is 30 secs.

Thanks

Koenkk, 2018-05-30:
It should already do that, e.g.

```
00:00 motion detected, occupancy: true send
00:10 motion detected, timer reset, occupancy: true send
00:40 occupancy: false send
```

Can you share the zigbee2mqtt log?

Odianosen25, 2018-05-30:
Hello @Koenkk,

Well it doesn't do that, and sometimes it just doesn't send report of motion detected after it has sent a false. Unfortunately as I am using docker, I can't `debug`, so don't know what's being sent.

Will send you the log later today when I am home.

regards

Odianosen25, 2018-05-30:
@Koenkk ,

As requested this is my log within a certain time. Within this time, I moved serveral times within the area, and even the issue of sudden true and false happened.

Hope it helps?

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2055053/log.txt)

ciotlosm, 2018-05-30:
For easier reading I did a grep and filter on repeating text:
```
2018-05-30T15:43:23.689Z - occupancy - false
2018-05-30T15:43:23.713Z - occupancy - true
2018-05-30T15:44:23.714Z - occupancy - false
2018-05-30T16:03:36.858Z - occupancy - false
2018-05-30T16:03:36.938Z - occupancy - true
2018-05-30T16:04:36.938Z - occupancy - false
2018-05-30T16:06:34.320Z - occupancy - false
2018-05-30T16:06:34.661Z - occupancy - true
2018-05-30T16:07:34.661Z - occupancy - false
2018-05-30T16:07:56.698Z - occupancy - false
2018-05-30T16:07:56.720Z - occupancy - true
2018-05-30T16:08:56.723Z - occupancy - false
2018-05-30T16:11:45.341Z - occupancy - false
2018-05-30T16:11:45.361Z - occupancy - true
2018-05-30T16:12:45.363Z - occupancy - false
2018-05-30T16:20:39.600Z - occupancy - false
2018-05-30T16:20:39.630Z - occupancy - true
2018-05-30T16:21:39.635Z - occupancy - false
```

Koenkk, 2018-05-30:
@Odianosen25 this is indeed strange behaviour, do you also have this with:

```
advanced:
  cache_state: false
```
?

Odianosen25, 2018-05-30:
It is now `true`. But I can confirm I had the same issues when it was `false`. I had it as `false` for a while, then since it still happened, changed it to `true`. Will revert to `false` now, then submit some logs say tomorrow if it still happens.

Regards

Odianosen25, 2018-05-31:
Hello @Koenkk,

Log for when it was set to `false` as requested.
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/2059450/log.txt)

Regards

ciotlosm, 2018-05-31:
And my cleaned up version for easier reading:
```
2018-05-30T23:10:06.434Z - occupancy - true
2018-05-30T23:11:06.436Z - occupancy - false
2018-05-30T23:11:13.666Z - occupancy - true
2018-05-30T23:12:13.670Z - occupancy - false
2018-05-30T23:12:14.387Z - occupancy - true
2018-05-30T23:13:14.387Z - occupancy - false
2018-05-30T23:13:24.770Z - occupancy - true
2018-05-30T23:14:24.771Z - occupancy - false
2018-05-31T05:21:18.254Z - occupancy - true
2018-05-31T05:22:18.242Z - occupancy - false
2018-05-31T05:39:44.708Z - occupancy - true
2018-05-31T05:40:44.709Z - occupancy - false
2018-05-31T05:42:14.202Z - occupancy - true
2018-05-31T05:43:14.204Z - occupancy - false
2018-05-31T05:43:18.258Z - occupancy - true
2018-05-31T05:44:18.264Z - occupancy - false
2018-05-31T05:44:27.603Z - occupancy - true
2018-05-31T05:45:27.608Z - occupancy - false
2018-05-31T07:03:23.017Z - occupancy - true
2018-05-31T07:04:23.024Z - occupancy - false
2018-05-31T07:11:00.083Z - occupancy - true
2018-05-31T07:12:00.084Z - occupancy - false
2018-05-31T07:23:44.204Z - occupancy - true
2018-05-31T07:24:44.211Z - occupancy - false
2018-05-31T07:33:24.086Z - occupancy - true
2018-05-31T07:34:24.093Z - occupancy - false
2018-05-31T07:35:00.627Z - occupancy - true
2018-05-31T07:36:00.630Z - occupancy - false
2018-05-31T07:36:12.452Z - occupancy - true
2018-05-31T07:37:12.458Z - occupancy - false
2018-05-31T08:02:42.407Z - occupancy - true
2018-05-31T08:03:42.412Z - occupancy - false
2018-05-31T08:05:25.840Z - occupancy - true
2018-05-31T08:06:25.848Z - occupancy - false
2018-05-31T08:21:41.244Z - occupancy - true
2018-05-31T08:22:41.246Z - occupancy - false
2018-05-31T08:22:50.970Z - occupancy - true
2018-05-31T08:23:50.972Z - occupancy - false
2018-05-31T08:28:24.464Z - occupancy - true
2018-05-31T08:29:24.469Z - occupancy - false
2018-05-31T08:31:59.569Z - occupancy - true
2018-05-31T08:32:59.575Z - occupancy - false
2018-05-31T08:40:05.970Z - occupancy - true
2018-05-31T08:41:05.976Z - occupancy - false
2018-05-31T08:41:17.850Z - occupancy - true
2018-05-31T08:42:17.858Z - occupancy - false
2018-05-31T08:42:21.404Z - occupancy - true
2018-05-31T08:43:21.405Z - occupancy - false
2018-05-31T14:25:03.555Z - occupancy - true
2018-05-31T14:26:03.557Z - occupancy - false
2018-05-31T14:27:01.992Z - occupancy - true
2018-05-31T14:28:01.993Z - occupancy - false
2018-05-31T14:28:05.836Z - occupancy - true
2018-05-31T14:29:05.836Z - occupancy - false
2018-05-31T14:29:11.485Z - occupancy - true
2018-05-31T14:30:11.491Z - occupancy - false
2018-05-31T15:26:20.518Z - occupancy - true
2018-05-31T15:27:20.520Z - occupancy - false
2018-05-31T16:48:38.636Z - occupancy - true
2018-05-31T16:49:38.643Z - occupancy - false
2018-05-31T16:50:43.150Z - occupancy - true
2018-05-31T16:51:43.151Z - occupancy - false
2018-05-31T16:52:04.142Z - occupancy - true
2018-05-31T16:53:04.143Z - occupancy - false
2018-05-31T17:53:51.807Z - occupancy - true
2018-05-31T17:54:51.813Z - occupancy - false
2018-05-31T17:54:52.723Z - occupancy - true
2018-05-31T17:55:52.730Z - occupancy - false
```

Odianosen25, 2018-05-31:
@ciotlosm,

Thanks for the clean up. The issue still happened, but not as frequently as the other. `You will notice there is no more than 60s time interval, which is what is set in the `configuration.yaml` file. This is regardless of the fact several times, I stayed within the zone and it will go off then on immediately. 

Regards

Koenkk, 2018-05-31:
Reading the log everything seems to be correct (e.g. true, false, true, false), could you point me to the problem?

Odianosen25, 2018-06-01:
Hello @Koenkk,

Well yes it does has `true` `false`, but the fact it does, that, I think it kind of inherently shows its an issue. These as the issues I outlined

When I am within the area initially, and it detects me and sends a `true`, which is fine. It runs for the 60 secs its been configured for, which is fine. Now the issue, even if I am still within the area, it doesn't send a `true` again; as I expect it should always send a true every time as long as there is motion detected, and `reset` its internal timer. Instead it will send a false, then a true immediately usually within 5 secs. This can be problematic as though I can easily get around this, others might find it difficult especially if using HA `yaml` for automation. And if one was to connect some light to it, it will give some OFF-ON effect. You see it here, just look at the flow based on the timing;
```
2018-05-31T05:42:14.202Z - occupancy - true
2018-05-31T05:43:14.204Z - occupancy - false
2018-05-31T05:43:18.258Z - occupancy - true
2018-05-31T05:44:18.264Z - occupancy - false
2018-05-31T05:44:27.603Z - occupancy - true
2018-05-31T05:45:27.608Z - occupancy - false
```
and here
```
2018-05-31T08:21:41.244Z - occupancy - true
2018-05-31T08:22:41.246Z - occupancy - false
2018-05-31T08:22:50.970Z - occupancy - true
2018-05-31T08:23:50.972Z - occupancy - false
```
and here
```
2018-05-31T08:41:17.850Z - occupancy - true
2018-05-31T08:42:17.858Z - occupancy - false
2018-05-31T08:42:21.404Z - occupancy - true
2018-05-31T08:43:21.405Z - occupancy - false
```
and here
```
2018-05-31T17:53:51.807Z - occupancy - true
2018-05-31T17:54:51.813Z - occupancy - false
2018-05-31T17:54:52.723Z - occupancy - true
2018-05-31T17:55:52.730Z - occupancy - false
```

I think the solution is thus
- The system should always send a `true`, as long as there is motion detected
- The `false` should only be sent, after the timeout has ran based on the last true (motion detected)

Thanks and regards

Koenkk, 2018-06-01:
Your solution is actually the expected behaviour! 😄 So we need to dive a little bit deeper here. Because for every incoming message from the device it sends a `occupancy: true`: https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L198.

Can you run with `DEBUG=zigbee2mqtt:controller npm start` in this way it logs incoming messages from the sensors.

Odianosen25, 2018-06-01:
Unfortunately I am using docker and you once told me I can't do the debug in this mode :(

Koenkk, 2018-06-01:
Debugging in docker is now supported. https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-debug#docker.

Can your run with `-e DEBUG=zigbee2mqtt:controller`?

Odianosen25, 2018-06-02:
As requested boss @Koenkk,

```
 zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msIlluminanceMeasurement', data: { measuredValue: 0 } } +4m
2018-6-2 06:41:33 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msIlluminanceMeasurement', data: { measuredValue: 0 } } +20ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msOccupancySensing', data: { occupancy: 1 } } +30ms
2018-6-2 06:41:33 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":true}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msOccupancySensing', data: { occupancy: 1 } } +11ms
2018-6-2 06:42:03 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":false}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msIlluminanceMeasurement', data: { measuredValue: 0 } } +1m
2018-6-2 06:42:34 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msIlluminanceMeasurement', data: { measuredValue: 0 } } +4m
2018-6-2 06:46:24 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msOccupancySensing', data: { occupancy: 1 } } +34ms
2018-6-2 06:46:24 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":true}'
2018-6-2 06:46:54 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":false}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msIlluminanceMeasurement', data: { measuredValue: 0 } } +1m
2018-6-2 06:47:50 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"illuminance":0}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msOccupancySensing', data: { occupancy: 1 } } +40ms
2018-6-2 06:47:50 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":true}'
2018-6-2 06:48:20 INFO MQTT publish, topic: 'zigbee2mqtt/Hallway Motion Sensor', payload: '{"occupancy":false}'
```

It kept going in and out of the motion sensor's range constantly, until about 15 secs after the `false` was sent.

Regards

Koenkk, 2018-06-02:
#ab64d079aa5accabf58e25d8ad70f49b2b5484c2 may fix your problem, can you update to the last version?

Odianosen25, 2018-06-02:
@Koenkk, sorry just seeing this for some reasons not sure. Will update and send a report soon.

Regards

Odianosen25, 2018-06-05:
@Koenkk,

Just saw I didn't respond to this, and I am sorry about that as I will like this issue to be closed.

But unfortunately, the same thing still happens and it doesn't send a true each time I enter the area of detection. Actually in the debug, nothing is sent or displayed each time I enter the area. is it possible the issue is from the sensor itself?

Also noticed that sometimes if I enter the area, go out. it sends a false and if i enter again, it doesn't register. I am beginning to think the issue is from how the the sensor is detecting the movement, but not certain. is there Scythian you will want me to try?

Regards

ciotlosm, 2018-06-06:
@Odianosen25 if you have a second sensor try that one.

Koenkk, 2018-06-06:
@Odianosen25 I'm also getting the feeling that your motion sensor isn't working correctly, to verify this.

1. Start with `DEBUG=zigbee2mqtt:controller npm start`
2. Keep waving your hand in front of the motion sensor
3. Every few seconds your log should show:

```
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'msOccupancySensing', data: { occupancy: 1 } } +30ms
```

Odianosen25, 2018-06-06:
@ciotlosm, yeah I have two more ordered and thy are on their way. Not sure why its taking so long as it should have arrived by now. But once in, I will give it a go.

@Koenkk, did as you said and I think I have fixed it now. 

First I followed your instructions and I noticed it wasn't sending any data as it supposed to. Then I held down the pairing button and it entered pairing mode, at this point it spat out some data. From then on, its now working perfectly.

Not sure why this happened, but this is not the first time I have noticed something like this; whereby a device isn't working properly and by putting it in to pairing mode (even if the bridge isn't set to connect to devices), it just works.

Thanks for all the help guys, and will be closing this now.

Regards

closed 2018-06-06 by Odianosen25