Koenkk/zigbee2mqtt #71: Xiaomi MiJia wireless switch long press release event

mihalski, 2018-05-28:
Thank you for a great new way to use out devices!

I'm pretty certain that the Xiaomi MiJia wireless switch has a long press release event to go with the long press. Any chance we could get it added to the bridge?

I came across this when using the Node-RED implementation some time ago. 

Regards,
Michal

ciotlosm, 2018-05-28:
@mihalski do you see anything in logs when doing long press?

mihalski, 2018-05-28:
Sure:
```
018-5-28 21:27:57 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00019dee01', payload: '{"click":"long"}'
```
But nothing when I release.

ciotlosm, 2018-05-28:
I've never encountered something extra besides the long press. Do you have any code reference to such release event? Maybe from a different framework that you were using?

mihalski, 2018-05-28:
Actually it seems the HomeAssistant integration supports it too.
The long click is click_type "long_click_press" and the release from the long click is click_type "hold". Or maybe BOTH release and hold exist because even though the Home Assistant webpages is malformed I can see those entries.

[image]

Maybe the problem is how many different click types you can have? Instead of triple and quadruple it has long_click_release and hold? Although I think that in my automatons hold is acting as long_click_release.. But that's another issue.

https://github.com/home-assistant/home-assistant/blob/master/homeassistant/components/binary_sensor/xiaomi_aqara.py

ciotlosm, 2018-05-28:
@mihalski the "triple" and "quadruple" are for Aqara button not for MiJia. The MiJia only has singural click or long_press. 

Thanks for the links, those should be useful.

mihalski, 2018-05-28:
Also, they always seem to return battery of 100.00... I don't think this is right.

james-fry, 2018-05-28:
> The round button is MUCH better.

Agree. I have both.
Round is better functioning. Square IMO looks better

ciB89, 2018-05-28:
Can't the round button be emulated on the square button? The functionality should only be different firmware, right?

Koenkk, 2018-05-28:
long_release added for WXKG01LM, for the battery it will also return the voltage.

closed 2018-05-28 by Koenkk

mihalski, 2018-05-29:
That was an amazing request to implementation turnaround! Thank you.

I know this is closed but I didn't want to start a new thread in case it was a stupid suggestion.

How about adding a long_press duration (in ms including initial 300ms for long_press) to the mqtt publish data? This could allow some creative usage and I think it would be better to calculate on the bridge than by whatever is consuming the mqtt.

Just my 2c worth.

ciB89, 2018-05-30:
Currently some buttons are on sale on gearbest for approx. 5 euros

mihalski, 2018-05-30:
@ciB89 Thanks. Unfortunately it's the square Aqara ones which lack a lot of functionality (and double clicks mis-register sometimes).

Xiaomi needs a 3rd button with the functionality of the round one but the appearance of the square one.

mihalski, 2018-05-30:
Yes! You're right! Thank you.
No, the others are wall switches. These are currently the only two buttons available.