Koenkk/zigbee2mqtt #73: Allow unique_id through homeassistant mqtt discovery

ciotlosm, 2018-05-28:
Based on https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/sensor/mqtt.py#L49 it should be possible to send `unique_id` and allow friendly names directly from Home Assistant UI. 

I will test locally and see how it behaves. I recommend something like: 0x00157e0001fa4c9d_temperature_zigbee2mqtt

ciotlosm, 2018-05-28:
It seems it's not straight forward. Even if the mqtt sensor has the ability, when going through the discovery the config message is lost and it doesn't work out of the box. I'll try to look in more depth.

ciotlosm, 2018-05-29:
Reopening as it seems it's possible for sensors already, just will need more work for binary sensors and switches.

dgomes, 2018-06-03:
Just adding as reference:

https://github.com/home-assistant/architecture/issues/30

ciotlosm, 2018-06-04:
Partially fixed by https://github.com/Koenkk/zigbee2mqtt/pull/95, but will require Home Assistant code for mqtt switches and lights

closed 2018-06-04 by Koenkk

ciotlosm, 2018-06-10:
Created https://github.com/home-assistant/home-assistant/pull/14912 for binary_sensors with unique id