Koenkk/zigbee2mqtt #74: How to update zigbee2mqtt?

mihalski, 2018-05-28:
Once you make configuration changes git pull does not work:

```
Updating 995649d..9396bde
error: Your local changes to the following files would be overwritten by merge:
	npm-shrinkwrap.json
Please commit your changes or stash them before you merge.
Aborting
```
Is there a way to update? Or do you need to copy the config out and start from scratch?

Koenkk, 2018-05-28:
See https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version

mihalski, 2018-05-28:
This cannot be done.
```
/opt/zigbee2mqtt $ git pull
remote: Counting objects: 27, done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 27 (delta 20), reused 17 (delta 10), pack-reused 0
Unpacking objects: 100% (27/27), done.
From https://github.com/Koenkk/zigbee2mqtt
   995649d..9396bde  master     -> origin/master
Updating 995649d..9396bde
error: Your local changes to the following files would be overwritten by merge:
	npm-shrinkwrap.json
Please commit your changes or stash them before you merge.
Aborting
```

mihalski, 2018-05-28:
```
/opt/zigbee2mqtt $ git diff
diff --git a/data/configuration.yaml b/data/configuration.yaml
index 3ed7c1c..da5b127 100644
--- a/data/configuration.yaml
+++ b/data/configuration.yaml
@@ -1,20 +1,17 @@
-# Home Assistant integration (MQTT discovery)
 homeassistant: false
-
-# allow new devices to join
 permit_join: true
-
-# MQTT settings
 mqtt:
-  # MQTT base topic for zigbee2mqtt MQTT messages
   base_topic: zigbee2mqtt
-  # MQTT server URL
-  server: 'mqtt://localhost'
-  # MQTT server authentication, uncomment if required:
-  # user: my_user
-  # password: my_password
-
-# Serial settings
+  server: 'mqtt://rackpi'
 serial:
-  # Location of CC2531 USB sniffer
   port: /dev/ttyACM0
+devices:
+  '0x00158d000123df04':
+    friendly_name: '0x00158d000123df04'
+    retain: false
+  '0x00158d0001a2b36b':
+    friendly_name: '0x00158d0001a2b36b'
+    retain: false
+  '0x00158d00019dee01':
+    friendly_name: '0x00158d00019dee01'
+    retain: false
diff --git a/npm-shrinkwrap.json b/npm-shrinkwrap.json
index 6360d51..3f1a421 100644
--- a/npm-shrinkwrap.json
+++ b/npm-shrinkwrap.json
@@ -2737,26 +2737,6 @@
         "enum": "2.5.0"
       }
     },
-    "zcl-packet": {
-      "version": "git+https://github.com/Koenkk/zcl-packet.git#360b9d2f02dc1bb40def40d03ac10175b93300e5",
-      "requires": {
-        "concentrate": "0.2.3",
-        "dissolve-chunks": "1.3.0",
-        "enum": "2.5.0",
-        "zcl-id": "0.2.0"
-      },
-      "dependencies": {
-        "zcl-id": {
-          "version": "0.2.0",
-          "resolved": "https://registry.npmjs.org/zcl-id/-/zcl-id-0.2.0.tgz",
-          "integrity": "sha1-E5nV5ba6SbZtDVpQYgCGXTcRfo0=",
-          "requires": {
-            "busyman": "0.3.0",
-            "enum": "2.5.0"
-          }
-        }
-      }
-    },
     "ziee": {
       "version": "0.3.0",
       "resolved": "https://registry.npmjs.org/ziee/-/ziee-0.3.0.tgz",
@@ -2801,6 +2781,26 @@
           "requires": {
             "ms": "2.0.0"
           }
+        },
+        "zcl-packet": {
+          "version": "git+https://github.com/Koenkk/zcl-packet.git#360b9d2f02dc1bb40def40d03ac10175b93300e5",
+          "requires": {
+            "concentrate": "0.2.3",
+            "dissolve-chunks": "1.3.0",
+            "enum": "2.5.0",
+            "zcl-id": "0.2.0"
+          },
+          "dependencies": {
+            "zcl-id": {
+              "version": "0.2.0",
+              "resolved": "https://registry.npmjs.org/zcl-id/-/zcl-id-0.2.0.tgz",
+              "integrity": "sha1-E5nV5ba6SbZtDVpQYgCGXTcRfo0=",
+              "requires": {
+                "busyman": "0.3.0",
+                "enum": "2.5.0"
+              }
+            }
+          }
         }
       }
     },
```

mihalski, 2018-05-28:
When installing from scratch it wasn't enough to just copy the configuration.yaml back. I had to re-add all the devices.
What other files are required to preserve the network for a new install?

ciotlosm, 2018-05-28:
`database.db` in the same `data` folder as `configuration.yaml`

Koenkk, 2018-05-29:
@mihalski you are right, you first need to do a `git checkout HEAD -- npm-shrinkwrap.json`. I've updated the wiki with this: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version

closed 2018-05-31 by mihalski

ilmisu, 2023-09-04:
> @mihalski you are right, you first need to do a `git checkout HEAD -- npm-shrinkwrap.json`. I've updated the wiki with this: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#6-for-later-update-zigbee2mqtt-to-the-latest-version

I do the update using this:
# Fermare il servizio ZigBee2MQTT
sudo systemctl stop zigbee2mqtt
cd /opt/zigbee2mqtt

# Attivare l'ambiente
source /opt/zigbee2mqtt/bin/activate

# Effettuare un backup della configurazione
cp -R data data-backup

# Update
git checkout HEAD -- npm-shrinkwrap.json #here i get the error:  lo specificatore percorso 'npm-shrinkwrap.json' non 
                                                                         corrisponde ad alcun file noto a git

git pull
npm ci

# Ripristino configurazione
cp -R data-backup/* data
rm -rf data-backup

# Disattivare l'ambiente
deactivate

# Avviare il servizio zigbee2mqtt
sudo systemctl start zigbee2mqtt

but I get the error highlighted at line "git checkout HEAD -- npm-shrinkwrap.json"-

any suggestion in how to fix? If I need a fresh install, please don't let me re-associate all the devices, I cant dismount many of them, otherwise it will be a big mess.