Koenkk/zigbee2mqtt #76: Clarification on data poll for xiaomi

subzero79, 2018-05-29:
Hi there, at the moment I have a solution to get the current state (speed) of my ceiling fans and their built in lights (on/off) into homeassistant by using the current power consumption. Speed set and light toggle is handled by pilight. 

At first stage i used a xiaomi plug and then later on moved to a sonoff pow (tasmota fw) due to the fact that i could increase the frequency that the power consumption is pushed to mqtt. Problem with sonoff tasmota pow is that due to low power consumption range (speed 1 is 3.8W and speed 6 max is 50W with light on) the state of lights and fan goes on/off, i push the data to influxdb an POW is just very irregular with power consumption.

The xiaomi plug using homeassistant reports load_power every three minutes when is varying, if is stays constant or little variation it reports maybe every 8 or so minutes . According to people using xiaomi, this is limitation by the gateway. For my purposes it definitely works as the reported load stays more constant, but i also want the state of the lights to be reported faster when people use the physical RF remote for example.

 Is it possible when using the zigbee2mqtt to poll load_power data more frequently from these plugs? I know there is a setting for body motion sensors, but i was curious about the plugs.

I don't have the hw yet, but is on the way.

Thanks for your time

palitu, 2018-05-29:
my Mi Plug is reporting much more often than 1 minute, but seems to change
between different plugs.

Otherwise, it is much more verbose using this and you can see what it is
actually doing.

subzero79, 2018-05-29:
OK, thanks for the info @palitu. I'll wait for my hw to arrive to start testing, seems promising for what i want.

Koenkk, 2018-05-30:
ok, feel free to reopen if you have any more questions.

closed 2018-05-30 by Koenkk

palitu, 2018-05-31:
i have found, since concentrating on it a bit more, that different devices
of the same model can have very different behaviour in their reporting
times.

Really not sure why - could be a firmware thing?  For example, one of mi Mi
Plugs never reports power.

subzero79, 2018-05-31:
I am petty sure they report only on change, from what I see using the gateway. In Standard variation info arrives around every 3 minutes if I up a speed or turn the light. You can test if you have a two speed fan or swapping different bulbs on a lamp.