Koenkk/zigbee2mqtt #77: Merging

oskarn97, 2018-05-29:
Hey!
I'd like to merge with your fork since it's very well maintained and it only makes sense to have a version working together.
The one I built was especially made for use with FHEM Home automation system, so I'd need to add a few endpoints to work with that.

Especially:
- Starting the paring process through MQTT
- Request all paired devices for auto creation (with sensor type)

Would that work for you?

ciotlosm, 2018-05-29:
Pinging also @starkun that had a more advanced fork from AndrewLinden/xiaomi-zb2mqtt

Koenkk, 2018-05-29:
Yes! Note that starting the pairing process through MQTT is already possible: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgeconfigpermit_join

oskarn97, 2018-05-29:
Cool! Could you also merge the ZCL-Packet fork to support Xiaomi Battery Status Messages?

Koenkk, 2018-05-29:
It's already there: https://github.com/Koenkk/zcl-packet

oskarn97, 2018-05-29:
Yes, I did a PR on that to support decoding Aqara Battery Messages. Or did you manage to do that otherwise?

tb-killa, 2018-05-29:
Look : https://github.com/Koenkk/zcl-packet/commit/d5128f0fb2f1adf707e582a7b469af6dfed37426

oskarn97, 2018-05-29:
Ahh, nice - thanks for clarification!

Koenkk, 2018-05-29:
So everything is ok already?

oskarn97, 2018-05-29:
Could you add a endpoint that sends a list of all paired devices with iEEE address and modelId and also a publish if a new device joins the network?
That's all I need to make the FHEM module work with this fork :)

tb-killa, 2018-05-30:
For info about device state (like join) look here : https://github.com/Koenkk/zigbee2mqtt/issues/13

Koenkk, 2018-05-30:
Both features are added: https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgelog

oskarn97, 2018-05-30:
Thanks a lot! I've closed my fork and updated the [https://github.com/oskarn97/fhem-xiaomi-mqtt](FHEM plugin) to work with yours.

Koenkk, 2018-05-30:
@oskarn97 great!

closed 2018-05-30 by Koenkk