Koenkk/zigbee2mqtt #78: Instruction for flashing firmware via Arduino

kirovilya, 2018-05-31:
I have a positive experience of [flashing firmware cc253x through Arduino](https://github.com/kirovilya/ioBroker.zigbee/wiki/%D0%9F%D1%80%D0%BE%D1%88%D0%B8%D0%B2%D0%BA%D0%B0-%D1%87%D0%B5%D1%80%D0%B5%D0%B7-Arduino).
Does it interest anyone? I can add the translation to the wiki, then edit the grammar. Or someone more experienced can make a translation.

ciotlosm, 2018-05-31:
I think would be great to have more flashing options. I just posted on the forum a way to flash using cc-tool (just seconds ago!): https://community.home-assistant.io/t/zigbee2mqtt-getting-rid-of-your-proprietary-zigbee-bridges-xiaomi-hue-tradfri/52108/176?u=ciotlosm 

This is based on yet another great repo around CC2531 gateway emulation in C++ here: https://github.com/Frans-Willem/AqaraHub. That has a good tutorial on how to flash using cc-tool

kirovilya, 2018-05-31:
@ciotlosm It's great. cctool has an even better option, because in my case cclib it is necessary to slightly modify the firmware.
I'll try to flash through cctool and maybe delete my version, and leave them :)

tb-killa, 2018-05-31:
@kirovilya : Which Type of Arduino do you use ? I´m experiment via Nano V3.0 ATmega328P Micro-Board.
Btw: Here: http://magictale.com/2884/flashing-ble112-with-arduino-based-cc-debugger/ you will find an performance boosting Version but there we need to modifiy to add the CC253X Type.

kirovilya, 2018-05-31:
@ciotlosm Oh, no! I read that cctool uses ccdebugger :( so that flashing firmware through arduino will remain

ciotlosm, 2018-05-31:
@kirovilya sorry for that. I was posting offtopic as I was just happy to see more options for flashing.

kirovilya, 2018-05-31:
@tb-killa I uses [Arduino Uno](https://raw.githubusercontent.com/kirovilya/files/master/IMG_20180111_193941.jpg).
I seen this article, but can not check - need compiling with specific port... 

@ciotlosm Don't worry :)

tb-killa, 2018-06-01:
@kirovilya you have to mention the needing of 3.3v on side of the usb Stick, because normaly you put an divider inside the connections. Or you use the internal resistor of the atmega themselve. I know it works without it but if some one broken the usb Stick you could say we mention it as an warning ;-)

definitio, 2018-06-01:
@kirovilya it works with esp8266 (wemos d1 mini) with changed pins. Maybe should be added to wiki?

definitio, 2018-06-01:
@kirovilya yes, successfully flashed 1-2 weeks ago with your cclib. I mean changed pins numbers.

kirovilya, 2018-06-01:
@definitio Good news! Thank you!

Koenkk, 2018-06-01:
Thanks for the instructions!

To keep the getting started as noob friendly as possible I've moved this to [Alternative firmware flashing methods](https://github.com/Koenkk/zigbee2mqtt/wiki/Alternative-firmware-flashing-methods)

kirovilya, 2018-06-01:
Good. I close the issue

closed 2018-06-01 by kirovilya

ciotlosm, 2018-06-01:
I've updated https://github.com/Koenkk/zigbee2mqtt/wiki/Getting-started with linux/macos instructions as well

ciotlosm, 2018-06-02:
@kirovilya in https://github.com/Koenkk/zigbee2mqtt/wiki/Alternative-firmware-flashing-methods you mention modified firmware for flashing via Arduino https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex

Is the above firmware different from the ones @Koenkk prepared? I'm trying to make sense and gather most of the firmwares in a logical place here: https://github.com/Koenkk/zigbee2mqtt-firmware for easier maintenance and linking.

kirovilya, 2018-06-03:
@ciotlosm that firmware not made from Koen firmware. It make from any firmware by remove second line from end (as described here https://github.com/wavesoft/CCLib/issues/19)

ciotlosm, 2018-06-03:
Would you be able to add this firmware to 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

maur8ino, 2018-08-08:
hi @definitio could you explain in detail which pins on the wemos d1 mini you connected to downloader cable?
I've tried different combination, but with no luck:
```
int CC_RST   = D4; // 2
int CC_DC    = D3; // 0
int CC_DD_I  = D2; // 4
int CC_DD_O  = D1; // 5
```

```
int CC_RST   = D7; // 13
int CC_DC    = D6; // 12
int CC_DD_I  = D5; // 14
int CC_DD_O  = D0; // 16
```

they both didn't work.

Thanks!

definitio, 2018-08-10:
@maur8ino just flashed the latest firmware (with removed the second line from the end) with this settings:
```
int CC_RST   = 5;
int CC_DC    = 4;
int CC_DD_I  = 14;
int CC_DD_O  = 12;
```

ericstewart22, 2018-09-27:
Is it possible to flash via a NodeMCU 1.0 (ESP-12E)? It seems to flash okay but I always get an error when trying to run:

`python27.exe cc_info.py -p COM9`

`ERROR: Could not find CCLib_proxy device on port COM9`

HanschDennis, 2018-10-01:
@ericstewart22 
Were you successful with the NodeMCU? I got the same error :(

ericstewart22, 2018-10-01:
@HanschDennis No, unfortunately not. I'll let you know if I get it working.

larsdecker, 2018-10-18:
> I have a Raspberry Pi so I think I will try this fork to flash
> https://github.com/kabbi/CCLib-raspberry

Did it work?

ericstewart22, 2018-10-18:
Not yet. Verification fails and it doesn't go past 25% or 50%. The LED on the 2531 is off when I start flashing and turns on (green) a few seconds later, at the same time verification fails; I think it's a clue as to what is happening but I'm not knowledgeable enough to troubleshoot. I attached a screenshot of what it looks like, in case you're having the same issue, or in case someone else on here can help me troubleshoot.

[image]

ericstewart22, 2018-10-25:
Developer still hasn't responded. Anyone else have any tips?

ericstewart22, 2018-10-27:
Okay, I finally got it working with my NodeMCU v1.0. I gave up on the Raspberry Pi. I listed my setup below and some notes so that others don't have to struggle with this as well.

Windows 10 Pro 64-bit fresh install on old laptop
Arduino 1.8.7 installed, not portable, with default drivers for NodeMCU 1.0
Python 2.7 fresh install, added to PATH
pip install pyserial==3.0.1

CCLIb from here:
https://github.com/kirovilya/CCLib

CCLib_proxy.ino modified to use these pins:
int CC_RST   = D5;
int CC_DC    = D7;
int CC_DD_I  = D6;
int CC_DD_O  = D8;

Pins connected as follows (see attached image)
```
ORANGE     3.3V VCC      3.3
GREEN        GND             GND
YELLOW      RESET           D5
BLUE           DBG DATA    D6
RED            DBG CLOCK  D7
```
I connected pins D6 and D8 on the breadboard (BLACK wire)

No resistors, I think because all NodeMCU's 1.0 pins are 3.3v instead of the 5v on Arduino.

CC2531 was plugged into USB port for power during flashing. Picture shows power bank but I used my laptop's USB ports in the end. 

Flashed this firmware:
https://github.com/kirovilya/files/blob/master/CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex

With this command:
python cc_write_flash.py --in=CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex --erase
Green light on CC2531 should stay off during flashing.

It was very slow, it ran overnight. You can use Window's Presentation Mode to try and make sure your computer doesn't go to sleep or turn off overnight (Ctrl + X, Mobility Centre).

[image]

larsdecker, 2018-10-29:
Thanks a lot for your detailed report. My cc2531 is still in the shipping.
After a received my, i will try your approach with the nodemcu.

derchrome, 2018-10-29:
Just a short question: Is it possible to use an arduino nana V3 for flashing the CC2531?

Thanks in advance.

larsdecker, 2018-10-29:
In my opinion it should possible. But i have no experience with it.
The internal storage of the arduino nano is a little bit low. Maybe that is a problem.

The esp8266 / esp32 are very cheep and powerful. So if the arduino nano is not workling then you can try it with the esp´s.

derchrome, 2018-10-29:
Do you mean something like that:
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.de%2Fulk%2Fitm%2F252718027546

larsdecker, 2018-10-29:
Yes, the esp32 / esp8266 works for a lot of electronics projects.

The esp32 is a new chip and it is more powerful and has support for bluethooth / ble. For flashing the cc2531 both should work.

derchrome, 2018-10-31:
So I shall order the NodeMcu from eBay?

larsdecker, 2018-11-01:
When you need it fast, it will be okay, for that price.
Otherwise you should order it directly from china.

Sometimes the price are around 1,20 € for the esp8266
For example:

https://de.aliexpress.com/item/ESP8266-CH340G-CH340-G-NodeMcu-V3-Lua-Wireless-WIFI-Module-Connector-Development-Board-Based-ESP-12E/32800966224.html?spm=a2g0x.search0104.3.45.38434e9111Xv5W&ws_ab_test=searchweb0_0,searchweb201602_3_10320_10065_10068_5731315_10547_319_317_10548_5728815_10696_10084_10083_10618_10304_10307_10820_5731115_10821_10302_5731415_5731215_5731615_10843_328_10059_10884_5731515_10887_100031_10319_321_322_10103_5731715,searchweb201603_1,ppcSwitch_0&algo_expid=e6c69a84-45ba-46b6-b686-c21df581bd16-9&algo_pvid=e6c69a84-45ba-46b6-b686-c21df581bd16

The esp32 are a little bit more expensive, but has much more power.

HanschDennis, 2018-11-02:
@derchrome I succesfully flashed it with a Arduino Nano. It took about 4 hours and it was tricky to get everything in the right (flash-)mode.

derchrome, 2018-11-02:
Do you have any details about the settings you made?

drndos, 2018-11-10:
I have managed to do it according to @ericstewart22 s instructions. If it doesn't work on first try, try disconnecting and connecting MCU to computer try disconnecting and connecting CC2531 to computer. In the end I don't have 3v3 connected and it works.

derchrome, 2018-11-11:
@drndos
 Did you use the Nano also? My CC2531 arrived yesterday :-)

drndos, 2018-11-11:
@derchrome 
I Tried using Arduino Mega, Wemos R1 mini and Node MCU. I managed to make it working with Node MCU. In all cases I got to a point where it would say that the chip is not responding. Only after few hours and rewiring everything again and again and checking everything I got to point of despair and started plugging it on and off in different order and got it working at last with Node MCU (because I believed it must be working somehow because of that picture and instructions from @ericstewart22 ) . 

I have not tried it with nano, but I would suggest trying:

1. Same mapping as in instructions based on the pinout. 
1. Don't be afraid of using constants D8, D6 ... in your program.
1. Try not connecting 3v3
1. Try connecting arduino and CC2531 to computer in different orders (I saw that my Node MCU didn't light up when connected in wrong order). Try reconnecting.
1. I didn't have proper cable, so I created some ghetto cables from arduino jumper wires by removing the plastic cover on one end and squishing the hole to smaller one and isolating it with isolating tape.
1. I have increased the timeout but I am not sure if it helped
1. Try connecting the DD wire like this D8 -- D6 -- DD and not like D6 -- D8 -- DD (I have no idea if it can make a difference)

RWGaMeR, 2018-11-17:
I have this is error (ERROR: Could not find CCLib_proxy device on port COM5)

larsdecker, 2018-11-20:
My stick is also here. I will try it on the weekend. Hopefully i have luck with that.
Thanks also @drndos for the great instruction.

derchrome, 2018-12-07:
> I have this is error (ERROR: Could not find CCLib_proxy device on port COM5)

I have the same error, any ideas?

drndos, 2018-12-15:
Hello, same thing happened to me. I managed to solder small wire to the base of the gnd pin. Since it takes several hours to complete I would not recommend holding it there.

derchrome, 2018-12-19:
So for me it worked now! After trying several things like reconnecting both, the NodeMcu and the CC2531 in different orders, rechecking the wiring and so on, it finally worked. I disconnected the CC2531 from the PC and connected it to a usb 5v power supply. When I connected it to the power supply the green led on the CC board was off and I could directly flash the board! Before (when connected to the PC) the led has been on all the time and flashing was not possible.

Maybe this helps somebody with similar problems.

roger-, 2018-12-27:
I just spent several hours trying to fix the `ERROR: No chip found. Check your connection and/or wiring!` message I was getting and thought I'd give hints about how I finally got it work. The whole procedure seems to be very temperamental, so it requires patience (if you aren't lucky).

I originally tried with a 3.3 V WeMos D1 mini with `DD_I ` and `DD_O` pins connected together, but that still gave an error. Next I tried a 5 V Arduino Leonardo with the resistive level converter, but that gave the same error. Finally I tried a NodeMCU using the same pinouts as @drndos above, which again gave the same error *but* I was eventually able to get CClib to recognize my CC2531.

I'm not exactly sure what fixed it, but here are a few points that might help:

* Try disconnecting either the `DD_I `  or `DD_O `  pin on your Arduino before powering it up
* Increase the timeout in the Arduino sketch to 1000 ms (the `delay()` statement)
* Plug the CC2531 into a USB power supply instead of the host PC
* Switch the order in which you power on the Arduino and the CC2531
* Try a different PC + OS (Windows didn't work, Ubunutu did)
* CClib seems to work with Python 3.6, but I did use the old version of pyserial that was recommended
* I noticed that some of the data pins from the CC2531 had 3.3 V on them after plugging them it, might be useful to check for that after you wire everything up

janisHD, 2019-01-19:
To all with similar problems:
I could resolve the issue with setting the baudrate to 1200 kpbs, any baudrate higher than that leads to a known issue:
https://github.com/arduino/ArduinoCore-avr/issues/53

After making the changes in the .ino and ccproxy.py the flashing works flawlessly and within a few minutes!

Regards

larsdecker, 2019-01-19:
Not that easy, to flash that stick. I now try it with the nodeMCU with no luck. When i try to check the values the nodeMCU seems to be go off (the blue led goes off). The green LED of usb stick is on. Have anyone some tricks how it work with the nodeMCU Amica?

creativtransfer, 2019-01-20:
Have a look at https://hackaday.io/project/163487-zigbee-cc2531-smart-home-usb-adapter
There's a nice tutorial covering a complete install using an ESP8266 / Wemos D1 mini as a programmer.

druxx, 2019-01-28:
I'm close to giving up: after fighting with my CC2531 sticks for weeks, I'm finally able to flash more than 85%, but then I'm getting reproducibly an error:

Flashing:
 -- Chip erase...
 -- Flashing 4 memory blocks...
 -> 0x0000 : 8176 bytes
    Progress 100%... OK
 -> 0x1ff6 : 10 bytes
    Progress 100%... OK
 -> 0x3fff0 : 1 bytes
    Progress 100%... OK
 -> 0x2000 : 239616 bytes
    Progress 91%...  ERROR: Could not read from the serial port!

I'm using an Arduino nano and I'm getting the same result on a MacBook and an Ubuntu desktop. I've played with the baud rates, but no success with 1200,4800,19200. 

Any ideas why the flashing works properly for hours and then suddenly fails just shortly before finishing?

druxx, 2019-01-29:
hooray, I was able to flash my first CC2531. 

Had to increase the read timeout from 3secs to 10secs. I've tried this before, but at at different baud rates. The whole thing is really a little bit tricky

creativtransfer, 2019-01-29:
Good to hear you've made it. 
Where exactly did you increase the timeout? Please post the relevant lines of code...

And which firmware have you installed now?

CC2531ZNP-Prod.hex 
or 
CC2531ZNP-Pro-Secure_LinkKeyJoin_mod.hex

way2many, 2019-02-07:
I've followed exactly what @drndos did with a Wemos D1 mini (followed his pin configuration and his cable solution also) and @BotoX recommendation for a delay of 5000ms and was able to flash it just under 2 hours. Wemos was connected on my laptop with Windows 10 and the usb-stick was on a 5V 2 amp USB power supply. If errors on chip detection, check cables. Led lamp on CC2531 has to be off.

Hope this bit of information will help someone as it wasn't easy to follow all the different possible configurations.

I would also like to thanks @drndos for his pictures. It helped tremendously!

timdonovanuk, 2019-02-14:
Hi - none of the links for getting started, alternate flashing methods etc work anymore :(

creativtransfer, 2019-02-14:
Have you checked https://hackaday.io/project/163487-zigbee-cc2531-smart-home-usb-adapter ?
This is kind of a tutorial on how you do it with an ESP8266/nodeMCU/wemos d1 mini but this is basically the same for any arduino compatible device.

creativtransfer, 2019-02-14:
Have you checked https://hackaday.io/project/163487-zigbee-cc2531-smart-home-usb-adapter ?
This is kind of a tutorial on how you do it with an ESP8266/nodeMCU/wemos d1 mini but it's basically the same for any other arduino compatible device. You just have to adjust your programming pins accordingly...

AS137430, 2019-02-16:
After weeks of trying, finally got the flash working with @vogler 's recommendation to use test hook cables. In earlier failed attempts: I improvised with Bus Pirate probe cable, nodeMCU v3, Arduino Mega...

I used a nodeMCU v2 with the following pinouts:
`int CC_RST   = D5;
int CC_DC    = D7;
int CC_DD_I  = D6;
int CC_DD_O  = D8;
`
The Arduino sketch used these values
`
// Initialize serial port
  Serial.begin(9600);
  // Wait for chip to initialize
  delay(5000);
`
and I also incorporated @druxx 's idea to increase timeout in cclib/ccproxy.py:
`
   self.ser = serial.Serial(port, baudrate=9600, timeout=20.0, write_timeout=3.0)
`

timdonovanuk, 2019-02-27:
Could someone please help clarify the below i.e. which pins connect to where? It so *nearly* makes sense but then doesn't actually at all :)

[image]

creativtransfer, 2019-02-27:
What do you want to use for flashing? Arduino uno, nano, nodemcu, wemos d1 mini?
You have to look in your code, there you'll find something like this:
...
// Pinout configuration (Configured for wemos D1 mini)
int CC_RST   = D5;
int CC_DC    = D7;
int CC_DD_I  = D6;
int CC_DD_O  = D4; 
...
In this case, it's for a wemos d1 mini, so just hook it up to the corresponding pins named D4, etc.

timdonovanuk, 2019-02-27:
I figured it out, but the problem with ambiguous instructions...the following diagram looks a lot like the debug cable, except its not, it represents the direct pins on the actual board:

[image]

If you're using the debugging cable (mine anyway) its actually (with the red cable strip at the bottom and the clip of the cable to your left, denoted by the indent)....:

[image]

fluffymadness, 2019-03-07:
tried flashing a cc2530 via nodemcu with baud 9600 delay in arduino sketch, and readtimeout:20 write_timeout:3 for serial in ccproxy.py
Sadly i only get the following:

Flashing:
 - Chip erase...
 - Flashing 4 memory blocks...
 -> 0x0000 : 8176 bytes
    Progress 100%... OK
 -> 0x1ff6 : 10 bytes
    Progress 0%...  ERROR: Flash verification error on offset 0x1ff6

AS137430, 2019-03-08:
@fluffymadness : Most of the other attempts seem to use separate power supply. You can try powering from PC/laptop USB and disconnect 3.3V of nodemcu. 

Or you might want to try CCLoader [https://github.com/Koenkk/zigbee2mqtt/issues/1057#issuecomment-468920302](url)

fluffymadness, 2019-03-08:
Thx.
I got it running by flashing it via teensy 3.1 and external power supply.
Whats interesting...flashing via teensy only takes about 2minutes not 2 hours like listed.

[10 more replies]