Koenkk/zigbee2mqtt #79: Timeout when starting directly after boot

oskarn97, 2018-05-31:
Hey!
The FHEM users and I are experiencing the same issue with your fork that I did in the past with mine.
If you start the fork directly after boot on a raspberry pi 3 (with FHEM installed, I do not know if that is related to that). The solution was to retry connecting every few seconds until the port was available.
It was also important to explicitly close the port, otherwise reconnecting was not possible until reboot.
See [https://github.com/oskarn97/xiaomi-zb2mqtt/commit/9f700ee6c1f96d7852534a96d57a466ec640c6f1](url)

`2018-5-31 11:23:48 INFO Starting zigbee-shepherd
2018-5-31 11:23:51 ERROR Error while starting zigbee-shepherd!
2018-5-31 11:23:51 ERROR Failed to start
2018-5-31 11:24:30 ERROR 
	{"message":"Port is already open","stack":"Error: Port is already open\n    at SerialPort.open (/opt/xiaomi/node_modules/serialport/lib/serialport.js:156:24)\n    at CcZnp.init (/opt/xiaomi/node_modules/cc-znp/lib/ccznp.js:140:8)\n    at Promise.post (/opt/xiaomi/node_modules/q/q.js:1181:36)\n    at Promise.promise.promiseDispatch (/opt/xiaomi/node_modules/q/q.js:808:41)\n    at /opt/xiaomi/node_modules/q/q.js:1411:14\n    at runSingle (/opt/xiaomi/node_modules/q/q.js:137:13)\n    at flush (/opt/xiaomi/node_modules/q/q.js:125:13)\n    at _combinedTickCallback (internal/process/next_tick.js:131:7)\n    at process._tickCallback (internal/process/next_tick.js:180:9)"}`

ciotlosm, 2018-05-31:
It depends on your mqtt setup as well. For example with home assistant addon, the supervisor makes sure stuff starts up in correct order.

mihalski, 2018-05-31:
In my case the port was not open. Is there anything else that could cause this?

oskarn97, 2018-06-01:
I am also unsure what is causing this. However, I am able to reproduce it with the same code on another raspberry pi 3.
Can be fixed by explicitly calling `shepherd.controller._znp.close(function(err){});` on failure and trying some seconds later again.

Koenkk, 2018-06-01:
I've added the recovery mechanism, does this solve the problem?

oskarn97, 2018-06-03:
Thanks, that is a start - the recovery timeout should either be increased to 45 seconds or applied multiple times until it starts. One time after 5 seconds does not solve the problem unfortunately.

Koenkk, 2018-06-04:
Increased the timeout, can you check again?

oskarn97, 2018-06-04:
perfect, works. Thanks!

closed 2018-06-04 by oskarn97