Koenkk/zigbee2mqtt #81: Configure logging

Koenkk, 2018-05-31:
Allow to configure different log_levels (e.g. only WARN/ERROR).

ciotlosm, 2018-05-31:
@Koenkk would be nice to be able to change this through mqtt while running.

tb-killa, 2018-06-02:
We should also ADD this type of "enable extended Zigbee-Package debugging" like mentioned here: https://github.com/Koenkk/zigbee2mqtt/issues/36#issuecomment-394095615

Koenkk, 2018-06-02:
@tb-killa this can be easily done by using the debug module.

https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L8
https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L180

Would you mind doing a PR?

ciotlosm, 2018-06-08:
It seems that changing log levels is possible also during runtime. See https://github.com/winstonjs/winston#using-logging-levels: `You may also dynamically change the log level of a transport:`

ciotlosm, 2018-06-11:
@tb-killa if you can point me in the right direction based on comments form @Koenkk happy to have a look.

ciotlosm, 2018-06-13:
@Koenkk added in commit https://github.com/Koenkk/zigbee2mqtt/pull/110/commits/c65bd2b60b34e0372a52aefb88d09495260195f7

Koenkk, 2018-06-14:
I think this issue is done, what do you think @ciotlosm ?

ciotlosm, 2018-06-14:
I think so too

closed 2018-06-14 by ciotlosm