Koenkk/zigbee2mqtt #82: Mijia wireless Button was triggered(single click event) every hour

x01673, 2018-06-01:
Hello,
 My room light was turned on evey hour tolight ,so I checked the HA logbook, and found out that my automation was triggered due to my Mija button was clicked.but i am sure I did not click it, the first trigger was at 23:48
 I was  expecting  the light to turn on at 3:48, and it did happen as expected.
please check the log below,
Is there anything I can do to troubleshooting this? 
Thanks

--------------------
2018-6-1 23:48:54 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00022bcc7f', payload: '{"battery":"100.00","voltage":3092,"click":"single"}'
2018-6-2 00:48:25 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00022bcc7f', payload: '{"battery":"100.00","voltage":3092,"click":"single"}',
2018-6-2 01:48:37 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00022bcc7f', payload: '{"battery":"100.00","voltage":3082,"click":"single"}',
2018-6-2 02:48:48 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00022bcc7f', payload: '{"battery":"100.00","voltage":3082,"click":"single"}',
2018-6-2 03:48:21 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d00022bcc7f', payload: '{"battery":"100.00","voltage":3082,"click":"single"}'

ciotlosm, 2018-06-01:
What kind of automation are you using?

x01673, 2018-06-01:
@Koenkk , sorry I forgot to mention that I am using @danielwelch hassio-zigbee2mqtt and i will check if there is way to do that in hassio addon tommorw.

x01673, 2018-06-02:
@Koenkk I have restarted the addon with debug set to true, and  the problem is solved somehow.

Koenkk, 2018-06-02:
Does it also work without debug now?

x01673, 2018-06-02:
Hi @Koenkk,
I was able to reproduce this issue once I click the switch (WXKG01LM),
please see the attachment , in the file out.log, it shows that the switch was clicked every hour from **2018-6-2 12:47:13 to 2018-6-2 18:48:15** (UTC+8)
[out.log](https://github.com/Koenkk/zigbee2mqtt/files/2064943/out.log)
[err.log](https://github.com/Koenkk/zigbee2mqtt/files/2064946/err.log)

,

motaz9k, 2018-06-02:
@Koenkk yesterday, the button sent a single click every hour, today for a change is double 
this is a really strange bug
`
Jun 02 13:10:22 orangepizero node[4169]: 2018-6-2 13:10:22 INFO MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Switch1', payload: '{"battery":"100.00","voltage":3032,"click":"double"}'
Jun 02 14:10:34 orangepizero node[4169]: 2018-6-2 14:10:34 INFO MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Switch1', payload: '{"battery":"100.00","voltage":3032,"click":"double"}'
Jun 02 15:10:46 orangepizero node[4169]: 2018-6-2 15:10:46 INFO MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Switch1', payload: '{"battery":"100.00","voltage":3032,"click":"double"}'
Jun 02 16:10:54 orangepizero node[4169]: 2018-6-2 16:10:54 INFO MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Switch1', payload: '{"battery":"100.00","voltage":3032,"click":"double"}'
Jun 02 17:11:06 orangepizero node[4169]: 2018-6-2 17:11:06 INFO MQTT publish, topic: 'zigbee2mqtt/Xiaomi_Switch1', payload: '{"battery":"100.00","voltage":3032,"click":"double"}'
`

Koenkk, 2018-06-02:
I think i've found the bug, can you update zigbee2mqtt and check if this has been fixed?

motaz9k, 2018-06-02:
Thanks @Koenkk I updated and I will let you know soon

x01673, 2018-06-03:
@Koenkk,
I think its fix now, I did not find aditional click event except the one I clicked this morning.
Thanks

closed 2018-06-03 by x01673

Koenkk, 2018-06-03:
Great!

motaz9k, 2018-06-03:
I can confirm that the bug has been fixed, thanks @Koenkk you are the best

classiphil, 2018-10-27:
I have the same Problem with a Miija Door sensor MCCGQ01LM. Every hour it sends the contact info too. So my home assistant starts the automations.

`payload: '{"contact":true,"linkquality":126,"battery":"27.00","voltage":2955}'`

Koenkk, 2018-10-29:
@classiphil you need to trigger you automation based on a state change instead of a single MQTT message.