Koenkk/zigbee2mqtt #83: Remove firmware confusion

ciotlosm, 2018-06-01:
There are too many places for the firmware:
https://github.com/Koenkk/zigbee2mqtt/tree/master/firmware
https://github.com/Koenkk/zigbee2mqtt-firmware/tree/master/coordinator/Z-Stack%20Home%201.2.2a.44539/Projects/zstack/ZNP/CC253x/dev

Documentation points to the second and there have been several revisions linked in some of the issues. 

Recommendation:
- [ ] Remove firmware from https://github.com/Koenkk/zigbee2mqt
- [ ] Remove firmware from https://github.com/Koenkk/zigbee2mqtt-firmware/tree/master/coordinator/Z-Stack%20Home%201.2.2a.44539/Projects/zstack/ZNP/CC253x/dev 
- [ ] Create a folder in root of https://github.com/Koenkk/zigbee2mqtt-firmware with a way to identify latest "stable"
- [ ] Create a folder in root of https://github.com/Koenkk/zigbee2mqtt-firmware for testing firmwares with maybe a readme in the folder describing the function

Koenkk, 2018-06-01:
I think everything is done here.

closed 2018-06-01 by ciotlosm

keesschollaart81, 2018-06-21:
What is the difference between 
https://github.com/Koenkk/Z-Stack-firmware
and
https://github.com/Koenkk/zigbee2mqtt-firmware

Koenkk, 2018-06-21:
zigbee2mqtt-firmware will be removed soon (contains TI Z-stack code). 

Z-stack is the new one.

keesschollaart81, 2018-06-21:
If I flashed with the 'old' version (3 weeks ago), might there be a reason to upgrade?

Koenkk, 2018-06-21:
Only when you are having stability problems. (but currently this only happens with larger networks, 15+ nodes).