Koenkk/zigbee2mqtt #85: Send status over MQTT

Odianosen25, 2018-06-02:
It will be nice to be able to receive all status updates like `INFO` over MQTT. Like when the system starts up, all those data like the pair devices should be sent over MQTT.

Also during the pairing process, all data sent over MQTT. This will enable not only to start the process over mqtt (which is possible now), but also to monitor it. This way, one can control the process over GUI, or over voice and create the same process like the Xiaomi hub used to do; but not in Chinese this time :)

Regards

ciotlosm, 2018-06-02:
@Odianosen25 we already have status sent through MQTT when a device is connected. An you can look at logs of the addon through UI (if you're running home assistant addon). 

I have to agree that a bit more debug details through MQTT would be helpful to create voice assist sensors or even a custom "console" component on the UI.

Odianosen25, 2018-06-02:
Well I don't use Hassio. Though I can get all my logs via the `log.txt`, but just need everything sent over mqtt. Like the entire pairing process from `..._lock[]`
```
2018-05-26T04:33:43.194Z - info: Zigbee: allowing new devices to join.
2018-05-26T04:34:21.275Z - info: Connecting with device, please wait...
2018-05-26T04:34:24.500Z - info: Connecting with device, please wait...
2018-05-26T04:34:27.734Z - info: Connecting with device, please wait...
2018-05-26T04:34:30.964Z - info: Connecting with device, please wait...
2018-05-26T04:34:34.204Z - info: Connecting with device, please wait...
2018-05-26T04:34:37.436Z - info: Connecting with device, please wait...
2018-05-26T04:34:40.670Z - info: Connecting with device, please wait...
2018-05-26T04:34:43.904Z - info: Connecting with device, please wait...
2018-05-26T04:34:47.139Z - info: Connecting with device, please wait...
2018-05-26T04:34:50.579Z - info: Connecting with device, please wait...
2018-05-26T04:34:53.808Z - info: Connecting with device, please wait...
2018-05-26T04:34:57.047Z - info: Connecting with device, please wait...
2018-05-26T04:35:00.280Z - info: Connecting with device, please wait...
2018-05-26T04:35:03.504Z - info: Connecting with device, please wait...
2018-05-26T04:35:10.465Z - info: Connecting with device, please wait...
2018-05-26T04:35:10.480Z - info: New device joined the network!
2018-05-26T04:35:10.484Z - info: New device with address 0x00158dxxxxxx connected!
```

Koenkk, 2018-06-02:
Added MQTT 'pairing' log type. https://github.com/Koenkk/zigbee2mqtt/wiki/MQTT-topics-and-message-structure#zigbee2mqttbridgelog

Odianosen25, 2018-06-02:
Thanks for this, and tested the update and its great. Haven't tested them all yet, but the `zigbee2mqtt/bridge/config/devices` is cool.

Closing thread

closed 2018-06-02 by Odianosen25