Koenkk/zigbee2mqtt #88: flashing and wiring CC2530

1981nick1982, 2018-06-03:
I intend to buy a [cc2530](Development-Board-CC2530F256-Serial-Port-Wireless-Module-24MHz/32767470382.html?spm=2114.search0604.3.8.39866ec0fRGKA0&ws_ab_test=) . However, it is not clear in the wiki how it should be flashed and how it should be connected to, for example, an RPI.

I would have liked to know which parts I have to buy to flash the CC2530. I have read [this](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/
) page but this does not create an overall clarity in my opinion.

 If I understand correctly, should I use the pinout of this diagram to connect the CC2530 to the CC debugger / CC2531 cable? http://ptvo.info/wp-content/uploads/2018/05/smartrf04eb-pinout.jpg

And use  [USB to TTL - Aliexpress](https://www.aliexpress.com/item/CH340G-5V-USB-to-TTL-UART-Serial-Converter/32524218818.html?spm=2114.search0604.3.1.618d7b96XEHlzs&ws_ab_test=)
for the link to the pi?

I connect this stick as follows:

TX = P0_3
RX = P0_2
RST = P0_5
GND =?
3V3 =?
5V =?

kirovilya, 2018-06-03:
@1981nick1982 For speedup you can buy dev-board like this https://www.aliexpress.com/item/ZigBee-CC2530-Sensor-Node-Baseboard-Functional-Module-Expansion-Board-USB-Port-24MHz-256KB/32767498002.html put CC2530 and connect CCDebugger to this board.
Without board, you need connect CCDebugger to CC2530 as described http://ptvo.info/wp-content/uploads/2018/05/smartrf04eb-pinout.jpg

After flashing, for using you need USB-TTL cable and connection:
TX = P0_3
RX = P0_2
GND = GND
3v3 = VCC
and on CC2530 P20 to GND

kirovilya, 2018-06-03:
@1981nick1982 With expansion board you not need USB-TTL. You will have to try to hide this board - this is the only problem :) Hide the cc2530 with a TTL cable without a board much easier.

1981nick1982, 2018-06-03:
Hiding is not a problem. Next to the distribution box, I would place a [cabinet](https://nl.aliexpress.com/item/-/32845317832.html?spm=a2g0s.13010208.99999999.268.6d553c00E9UigB) in which I place the pi with the Zigbee2MQTT, Z-wave stick and an RF Link connected to it. The antennae of the RF Link and CC2530, I extend with a small piece of cable so that these come on the outside of the box.

ptvoinfo, 2018-06-03:
@1981nick1982 
If you'll connect CC2530 to RPi you don't need a USB-TTL converter. You may connect the P0_2, P0_3, GND directly to the corresponding pins on the RPi header.

1981nick1982, 2018-06-03:
So i can use pin 6 / 8 and 10 on my pi to connect?
https://www.raspberrypi-spy.co.uk/wp-content/uploads/2012/06/Raspberry-Pi-GPIO-Layout-Model-B-Plus-rotated-2700x900.png

should I not give the cc2530 power?

ptvoinfo, 2018-06-04:
@1981nick1982 

**Pins:** yes, you may use 6, 8 and 10
**Power:** you may connect 3.3V or 5V from the board to the corresponding pin on CC2530. If your CC2530 boards does not have 5V pin you should connect 3.3V only.

1981nick1982, 2018-06-04:
@ptvoinfo Perfect, i will give it a try. Maybe someone can make a wiki page of this information?

ciotlosm, 2018-06-04:
@1981nick1982 if you could create something basic based on your understanding, I'm happy to pimp it out with formatting and stuff

ptvoinfo, 2018-06-05:
I've prepared a tutorial.
http://ptvo.info/how-to-select-and-flash-cc2530-144/
Comments are welcome.

kirovilya, 2018-06-05:
@ptvoinfo great work!

ciotlosm, 2018-06-05:
@ptvoinfo Looks great!
Only things I can think of are: a wiring picture with a RPI and one of the boards

james-fry, 2018-06-06:
>I've prepared a tutorial.
>http://ptvo.info/how-to-select-and-flash-cc2530-144/
>Comments are welcome.

Great job! 
My USB CC2530 has arrived from China, but I am away with work this week.
Looking forward to programming and testing this based on @kirovilya wiring advice.

BTW @ptvoinfo what is the use case for the CC2530 board with  additional 8051?

ptvoinfo, 2018-06-06:
@james-fry The additional chip implements the USB interface and the CC debugger. This board can be programmed without the debugger. Therefore it holds small space on your work table :). It can be useful for developers. If you need only one CC2530 or CC2531 for a coordinator role you may save money.

james-fry, 2018-06-06:
Thanks.
I will update the repo wiki with link to your guide.

ryanbeaton, 2018-06-08:
So much good information developing, thanks.
Received my CC2531 with download cable yesterday (they actually sent two of each - score!), my CC2530 "RF TO USB (CC2530 CC2591)" today but still waiting on the CC Debugger. Still super keen though.

Would have probably started out with the "dev-board" but I ordered exactly as per the Getting Started wiki page because I'm literal when reading instructions :) That was back in the 'old days' when the wiki was only 8 pages and there was no "Supported sniffer devices" page.

Another HA user in Perth (WA) has offered to lend me his CC Debugger so maybe get something going this weekend, fingers crossed, and report in.

ciotlosm, 2018-06-15:
@1981nick1982 considering the documentation now exists (thanks @ptvoinfo ) can we close this issue?

1981nick1982, 2018-06-15:
Yes, it can be closed. Thank you guys.

closed 2018-06-15 by ciotlosm