Koenkk/zigbee2mqtt #90: Cube joining problem

kirovilya, 2018-06-03:
Today I can not join Cube to Shepherd (cc2531)... 

[image]

Interview process do not start, only "ZDO:tcDeviceInd" event received.

I've seen similar problems here:
https://github.com/Koenkk/zigbee2mqtt/issues/25#issuecomment-388045812 
https://github.com/zigbeer/zigbee-shepherd/issues/25

ciotlosm, 2018-06-03:
This is by far the most annoying to pair. I will try again tonight, maybe I can document a pattern

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ciotlosm, 2018-06-03:
I was able to pair it. I will try to make a clip and upload to youtube.

kirovilya, 2018-06-03:
@ciotlosm Thank you. I successfully connect a cube many times before. But now...

ciotlosm, 2018-06-03:
@kirovilya video will show up here: https://youtu.be/uhMrcIAdGxg . Quality will be low until processing is finished.

Koenkk, 2018-06-04:
@ciotlosm I've added your video to the wiki: https://github.com/Koenkk/zigbee2mqtt/wiki/Pairing-devices

Can this issue be closed?

kirovilya, 2018-06-04:
Yes. Thanks to everyone

closed 2018-06-04 by kirovilya