Koenkk/zigbee2mqtt #92: Xiaomi Aqara wireless switch (WXKG11LM) - more than one broadcast for one action

CrazyMurdok, 2018-06-04:
Hi all,

I bought an Xioami Aqara switch and paired this one with success. But I'm wondering why it broadcast an action more than ones. Sometimes up to five times in a row.

See below:

2018-6-4 16:16:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 16:16:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 16:16:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 16:16:23 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 16:16:24 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 16:16:25 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'

This was only one click, but in this example it appears five times. Could it be a range problem between the sniffer and the switch?

ciotlosm, 2018-06-04:
If this was just on the initial pairing, try a few more tests after a few minutes now that it's paired. I assumed the multiple messages at start ar due to the fact that we keep pressing buttons to make it pair.

CrazyMurdok, 2018-06-04:
@ciotlosm No, that isn't a pairing issue. The pairing took place two days ago.

Koenkk, 2018-06-04:
Can you update zigbee2mqtt to the last version? Probably same issue as: https://github.com/Koenkk/zigbee2mqtt/issues/82

CrazyMurdok, 2018-06-04:
@Koenkk Sorry, I did not realized that this issue came up before.

I have now updated to the last version (how can I check which version is running) and after some checks I will let you know if this issue is fixed for me. Many thanks for this project and for your help and work. ;-)

CrazyMurdok, 2018-06-04:
Most of the time is seems to work now, but I still see more than one broadcast sometimes. See below the single click at 19:57:43

2018-6-4 19:57:43 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:57:43 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:57:43 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:57:44 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:58:07 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:58:10 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"double"}'
2018-6-4 19:58:14 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'
2018-6-4 19:58:19 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"triple"}'
2018-6-4 19:58:21 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3025,"click":"single"}'

Koenkk, 2018-06-04:
Can you provide the log when running with `DEBUG=zigbee2mqtt:controller npm start`?

CrazyMurdok, 2018-06-05:
@Koenkk Yes, tomorrow I will run it in the debug mode, so that I can provide you the output.

But one think I could say is that I got the same issue which is discribed in Issue [https://github.com/Koenkk/zigbee2mqtt/issues/82](url)

I wasn't at home today, so no one could press the switch. But I got round about every our a broadcast from the switch. see below

2018-6-5 09:47:06 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 10:37:11 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 11:27:17 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 12:17:22 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 13:07:29 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 13:57:34 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 14:47:41 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 15:37:47 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 16:27:52 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-5 17:18:00 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'

CrazyMurdok, 2018-06-05:
@Koenkk I tried to start the the service in the debug mode, but I'm not able to stop it before. How can I stop it?

If I use "npm stop" it says:

npm ERR! missing script: stop

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2018-06-05T20_17_46_909Z-debug.log

And if I try to start the debug without stopping the service I get this:

/opt/zigbee2mqtt $ DEBUG=zigbee2mqtt:controller npm start ?

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js "?"

2018-6-5 22:22:50 INFO Starting zigbee-shepherd
2018-6-5 22:22:51 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-5 22:23:51 INFO Starting zigbee-shepherd
2018-6-5 22:23:51 ERROR Error while starting zigbee-shepherd!
2018-6-5 22:23:51 ERROR Failed to start

CrazyMurdok, 2018-06-05:
The service is still running. Is there no chance to restart it via terminal? The only way is to reset the sniffer? OK, that is really a task for tomorrow. I have to go to bed now.

Koenkk, 2018-06-05:
Normally it should stop when exiting with ctrl+c, can you kill it with `kill` or reboot your device?

CrazyMurdok, 2018-06-06:
It is running as a service, so I couldn't stop it with ctrl+c. So I understand that the only way to stop it is to kill the process? No way to stop it via "npm stop"?

And which process I have to kill? I didn't find anything in the Wiki.

ciotlosm, 2018-06-06:
@CrazyMurdok Did you use https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#5-optional-running-as-a-daemon-with-supervisord or how did you set it up? It really depends on your setup to understand how to stop it.

CrazyMurdok, 2018-06-06:
@ciotlosm Sorry, I'm not very familiar with Linux yet. And yes, I use supervisord.

CrazyMurdok, 2018-06-06:
@ciotlosm Thanks. That worked. So now I was able to start zigbee2mqtt in debug mode.

CrazyMurdok, 2018-06-06:
I have no idea why the switch is doing about every hour a broadcast. I never touched it.

  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { onOff: 1 } } +13s
2018-6-6 09:50:33 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data:
   { '65281': { '1': 3035, '3': 35, '4': 5032, '5': 105, '6': [Array], '10': 0 } } } +50m
2018-6-6 10:40:42 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data: { '65281': [ <3 empty items>, 35, <2 empty items>, [Array] ] } } +7ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data:
   { '65281': { '1': 3035, '3': 35, '4': 5032, '5': 105, '6': [Array], '10': 0 } } } +50m
2018-6-6 11:30:48 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data:
   { '65281':
      [ <1 empty item>,
        3035,
        <2 empty items>,
        5032,
        105,
        [Array],
        <3 empty items>,
        0 ] } } +7ms
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data:
   { '65281': { '1': 3035, '3': 35, '4': 5032, '5': 105, '6': [Array], '10': 0 } } } +50m
2018-6-6 12:20:52 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035}'
  zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genBasic',
  data: { '65281': [ <3 empty items>, 35, <2 empty items>, [Array] ] } } +8ms

Odianosen25, 2018-06-06:
@CrazyMurdok, its normal for the devices to send data occasionally. The data it usually sends, are things like battery and voltage (whatever that means) states. It will not send things like click, contact or occupancy states unless its activated. 

Regards

Koenkk, 2018-06-06:
@CrazyMurdok it is indeed normal that device send its battery status +- every hour. I don't see any strange things in your log, is this problem solved?

ciotlosm, 2018-06-06:
@Koenkk Wouldn't such a message trigger a click event in HA?

```
2018-6-5 11:27:17 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
```

Koenkk, 2018-06-06:
Yes, but above it is:

```
zigbee2mqtt:controller Recieved zigbee message with data { cid: 'genOnOff', data: { onOff: 1 } } +13s
```

--> the switch sent a click event.

ciotlosm, 2018-06-06:
@CrazyMurdok Lovely. Can you please close this issue for now? I'll open a new one to provide version on startup.
Currently version is a tricky thing, but I guess we can use git commit number until versioning (0.1) gets out of the door.

CrazyMurdok, 2018-06-06:
I can close it. But this switch is still toggle my lights on and off without pressing the button more than ones. And it is now running as a service and started via supervisor. And I can again see that one click on the button generate multiple payloads at once. But I will close this issue and no longer use this button to turn on and off the light. Maybe some of the other Xioami devices which will arrive in the next days will better work. :-)

Just for your information. this below was only one click. And I got three payloads. This switch make a lot more strange things. But it is to difficult for me to explain this in english. 

2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'

ciotlosm, 2018-06-06:
@CrazyMurdok the 3 events at once could have been a bug, however I have the exact same device and it works. Can you try pairing it again?

Where are you from? Maybe we can find another language as common ground :-)

Koenkk, 2018-06-07:
@CrazyMurdok for 

```
2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
2018-6-6 20:52:49 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0002133db7', payload: '{"battery":"100.00","voltage":3035,"click":"single"}'
``` 

It would be interesting to have the output when running with: `DEBUG=zigbee2mqtt:controller npm start`. 
Could you put this log on https://pastebin.com/ (to avoid any clutter in this thread)?

CrazyMurdok, 2018-06-07:
@ciotlosm I will try to pair it again and hopefully my bugs are all gone

I'm from Germany.

CrazyMurdok, 2018-06-07:
@Koenkk First I will try to do a new pairing. And if this issue will still appear I will run it again in debug mode and put the result on pastebin.

CrazyMurdok, 2018-06-07:
Great! Pairing it again seems to be fixed the issue. So many thanks for helping and for your great work from all of you to make all these components cloud free. ;-)

ciotlosm, 2018-06-07:
@CrazyMurdok glad to hear it worked!. Can you also please close the issue?

closed 2018-06-07 by CrazyMurdok

HydrelioxGitHub, 2019-02-16:
I re-open this issue as I have a similar behavior with my xiaomi device. When I click ont time on the button there is multiple payload on mqtt (sometime 2, sometimes more). 
When you said you should pair again the device, should I remove it first or not ?

Koenkk, 2019-02-16:
@HydrelioxGitHub can you post the log when this happens?

HydrelioxGitHub, 2019-02-25:
It's seem to be coming from home assistant. When I update to the last version (0.88), the problem stop to occur.

Thank you.