Koenkk/zigbee2mqtt #96: Stable release (version 0.1 ?)

ciotlosm, 2018-06-05:
Writing below the items I think need to be solved before 0.1? version is released:

- [ ] [Unhandled Error - Cannot get the Node Descriptor](https://github.com/Koenkk/zigbee2mqtt/issues/41)
- [ ] [Implement remove device from database](https://github.com/Koenkk/zigbee2mqtt/issues/35)
- [ ] [Per device state](https://github.com/Koenkk/zigbee2mqtt/issues/13)
- [ ] [Xiaomi Aqara wireless switch (WXKG11LM) - more than one broadcast for one action](https://github.com/Koenkk/zigbee2mqtt/issues/92)

Happy to see feedback on this.

ptvoinfo, 2018-06-05:
@ciotlosm 
#1 - I've added a handler for the "error" event and @Koenkk merged this update. "Cannot get the Node Descriptor" this is not an error because it may appear with offline devices.

ciotlosm, 2018-06-05:
@ptvoinfo should we close the issue then?

Koenkk, 2018-06-05:
@ptvoinfo don't you mean this issue? https://github.com/Koenkk/zigbee2mqtt/issues/42

ptvoinfo, 2018-06-06:
@Koenkk I mean #41

ciotlosm, 2018-06-06:
@ptvoinfo @Koenkk I did still see the message at #41 still, but now it doesn't crash zigbee2mqtt and usually it's a timeout for not pressing a link button often enough during pairing. I think we can close that one off. 

@ptvoinfo Are you happy with the list for 0.1?

ptvoinfo, 2018-06-06:
@ciotlosm I'm happy with v0.0 already :)

Koenkk, 2018-06-06:
I've created a milestone: https://github.com/Koenkk/zigbee2mqtt/milestone/1

ciotlosm, 2018-06-06:
Ok, I'll close this one and we'll use the milestone

closed 2018-06-06 by ciotlosm