Koenkk/zigbee2mqtt #100: WARN Error while starting zigbee-shepherd, attemping to fix

kiucheng, 2018-06-06:
I got it running right after I install it but runs into this error after I have my Pi restarted. 
`2018-6-6 19:51:05 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +33ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s
  zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22)
    at Object.onceWrapper (events.js:315:30)
    at emitOne (events.js:116:13)
    at CcZnp.emit (events.js:211:7)
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18)
    at ontimeout (timers.js:498:11)
    at tryOnTimeout (timers.js:323:5)
    at Timer.listOnTimeout (timers.js:290:5) +5ms
2018-6-6 19:51:08 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)`

ciotlosm, 2018-06-06:
Please use proper code block next time:

```
2018-6-6 19:51:05 INFO Starting 
zigbee-shepherd 
zigbee-shepherd:init 
zigbee-shepherd booting... +0ms 
zigbee-shepherd:request REQ --> SYS:osalNvRead +33ms 
zigbee-shepherd:request RSP <-- SYS:osalNvRead +3s 
zigbee-shepherd:init Coordinator initialize had an error: Error: request timeout 
    at CcZnp.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:254:22) 
    at Object.onceWrapper (events.js:315:30) 
    at emitOne (events.js:116:13) at CcZnp.emit (events.js:211:7) 
    at Timeout.<anonymous> (/opt/zigbee2mqtt/node_modules/cc-znp/lib/ccznp.js:233:18) 
    at ontimeout (timers.js:498:11) 
    at tryOnTimeout (timers.js:323:5) 
    at Timer.listOnTimeout (timers.js:290:5) +5ms 
2018-6-6 19:51:08 WARN Error while starting 
zigbee-shepherd, attemping to fix... (takes 60 seconds)
```

Make sure you're running latest master version. Make sure you have correct /dev/ device for your CC2531 stick path.

kiucheng, 2018-06-06:
Thanks.
Yes I have double the the version and /dev/device path. Both are correct. Just tried removing /opt/zigbee2mqtt and start again and still the same.

When it start successfully the first time, I was even able to pair my Aqara motion sensor

ciotlosm, 2018-06-06:
I assume you tried a reboot in case a port got stuck open.

Koenkk, 2018-06-07:
This is expected behaviour, implemented because of: https://github.com/Koenkk/zigbee2mqtt/issues/79

ciotlosm, 2018-06-07:
@kiucheng if you wait the 60 seconds, does it start?

kiucheng, 2018-06-07:
No, it doesn't
```2018-6-7 15:38:47 INFO Starting zigbee-shepherd
2018-6-7 15:38:50 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-7 15:39:50 INFO Starting zigbee-shepherd
2018-6-7 15:39:53 ERROR Error while starting zigbee-shepherd!
2018-6-7 15:39:53 ERROR Failed to start```

kiucheng, 2018-06-07:
I have re-flash the CC2531 and reinstall again. It starts successfully the first time. Then I went configurating supervisor. Run `sudo supervisorctl stop zigbee2mqtt` and then ran into the same error again

jarrah31, 2018-06-07:
I too have the same problem...

I just installed zigbee2mqtt yesterday and added my first Xiaomi Aqara Door Sensor this evening. The software is on it's own dedicated Pi running the latest master build. 

My /opt/zigbee2mqtt/data/configuration.yaml looks like this:
```
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.1.50'
  user: homeassistant
  password: removed
serial:
  port: /dev/ttyACM0
devices:
  '0x00158d0001b149eb':
    friendly_name: '0x00158d0001b149eb'
    retain: false
```
I changed the friendly_name, saved the file, restarted with `sudo supervisorctl start zigbee2mqtt`. Upon restarting the errors below started occurring, and now it won't work anymore.

I'm using a CC2531 USB sniffer and only used device MCCGQ11LM so far.

```
> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-7 18:06:56 INFO Starting zigbee-shepherd
2018-6-7 18:06:58 INFO zigbee-shepherd started
2018-6-7 18:06:58 INFO Currently 0 devices are joined:
2018-6-7 18:06:58 INFO Connecting to MQTT server at mqtt://192.168.1.50
2018-6-7 18:06:58 INFO zigbee-shepherd ready
2018-6-7 18:07:02 INFO Connected to MQTT server
2018-6-7 18:07:02 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
2018-6-7 18:07:02 WARN `permit_join` set to  `true` in configuration.yaml.
2018-6-7 18:07:02 WARN Allowing new devices to join.
2018-6-7 18:07:02 WARN Set `permit_join` to `false` once you joined all devices.
2018-6-7 18:07:02 INFO Zigbee: allowing new devices to join.
spinlock: false []
2018-6-7 18:25:58 INFO Connecting with device...
2018-6-7 18:25:58 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-7 18:26:01 INFO Connecting with device...
2018-6-7 18:26:01 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-7 18:26:04 INFO Connecting with device...
2018-6-7 18:26:04 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-7 18:26:08 INFO Connecting with device...
2018-6-7 18:26:08 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-7 18:26:11 INFO Connecting with device...
2018-6-7 18:26:11 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-6-7 18:26:11 INFO Device incoming...
2018-6-7 18:26:11 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-7 18:26:11 INFO New device with address 0x00158d0001b149eb connected!
2018-6-7 18:26:11 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x00158d0001b149eb"}'
2018-6-7 18:27:31 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001b149eb', payload: '{"contact":true}'
2018-6-7 18:27:33 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001b149eb', payload: '{"contact":false}'
2018-6-7 18:27:34 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001b149eb', payload: '{"contact":true}'
2018-6-7 18:27:35 INFO MQTT publish, topic: 'zigbee2mqtt/0x00158d0001b149eb', payload: '{"contact":false}'

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-7 18:40:37 INFO Starting zigbee-shepherd
2018-6-7 18:40:37 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-7 18:41:37 INFO Starting zigbee-shepherd
2018-6-7 18:41:37 ERROR Error while starting zigbee-shepherd!
2018-6-7 18:41:37 ERROR Failed to start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-7 18:41:41 INFO Starting zigbee-shepherd
2018-6-7 18:41:41 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-7 18:42:41 INFO Starting zigbee-shepherd
2018-6-7 18:42:41 ERROR Error while starting zigbee-shepherd!
2018-6-7 18:42:41 ERROR Failed to start
```

jarrah31, 2018-06-07:
Enabling debug has the same error:
```
pi@zigbeepi:/opt/zigbee2mqtt $ DEBUG=zigbee-shepherd* npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-7 19:11:03 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
2018-6-7 19:11:03 WARN Error while starting zigbee-shepherd, attemping to fix... (takes 60 seconds)
2018-6-7 19:12:03 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +1m
2018-6-7 19:12:03 ERROR Error while starting zigbee-shepherd!
2018-6-7 19:12:03 ERROR Failed to start
```
Happy to try any other debugging options?

jarrah31, 2018-06-07:
More troubleshooting steps...

- Stopped the service
- Left it for 20 mins
- Started again - same `Error while starting zigbee-shepherd`
- Rebooted - same error straight away (monitored log file as program auto-starts)
- Stopped the service
- Manually ran in debug mode straight after and it worked!
```
pi@zigbeepi:~ $ cd /opt/zigbee2mqtt/
pi@zigbeepi:/opt/zigbee2mqtt $ DEBUG=zigbee-shepherd* npm start

> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-7 19:51:44 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +36ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +50ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +20ms
```

- Exited using ctrl+c
- Manually started debug mode again and it worked ok
- Exited
- manually just ran `npm start` - all ok
- Started the service `sudo supervisorctl start zigbee2mqtt` - all ok
- Stop service `sudo supervisorctl stop zigbee2mqtt`
- Start service again - no longer working. :(
- Try manual npm start a few times with no luck
- Tried starting with debug a couple of times, no luck.
- Rebooted again - start automatically with the error
- `sudo supervisorctl stop zigbee2mqtt`
- Manual npm start - it works!
- ctrl+c - and manual start - it works.

It seems something is wrong with the service stop process....?

jarrah31, 2018-06-07:
Right, found a solution!! Basically, I ditched supervisor (as per the wiki instructions) and used systemctl:

```
sudo apt-get remove supervisor
sudo reboot
```
Add the following text to `/etc/systemd/system/zigbee2mqtt.service`
```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```
Test it works by starting and then check the status
```
pi@zigbeepi:/opt/zigbee2mqtt $ sudo systemctl start zigbee2mqtt
pi@zigbeepi:/opt/zigbee2mqtt $ systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-06-07 20:27:22 BST; 3s ago
 Main PID: 665 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─665 npm
           ├─678 sh -c node index.js
           └─679 node index.js

Jun 07 20:27:22 zigbeepi systemd[1]: Started zigbee2mqtt.
Jun 07 20:27:23 zigbeepi npm[665]: > zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
Jun 07 20:27:23 zigbeepi npm[665]: > node index.js
Jun 07 20:27:24 zigbeepi npm[665]: 2018-6-7 20:27:24 INFO Starting zigbee-shepherd
Jun 07 20:27:25 zigbeepi npm[665]: 2018-6-7 20:27:25 INFO zigbee-shepherd started
Jun 07 20:27:25 zigbeepi npm[665]: 2018-6-7 20:27:25 INFO Currently 1 devices are joined:
```
You can monitor the logs similar to tail by running:
`journalctl -u zigbee2mqtt.service -f`

Test stopping and restarting (all fine for me with multiple stop/starts)
sudo systemctl restart zigbee2mqtt
sudo systemctl stop zigbee2mqtt
sudo systemctl start zigbee2mqtt

Enable the service to start on boot:
`sudo systemctl enable zigbee2mqtt.service`

Hopefully this works for others too!

kiucheng, 2018-06-07:
Thanks! Tested start and stop service a couple of times. It works!

ciotlosm, 2018-06-08:
@jarrah31 great job man! Maybe best to update wiki as well.

Koenkk, 2018-06-08:
@jarrah31 thanks, I've made this the default configuration: https://github.com/Koenkk/zigbee2mqtt/wiki/Running-the-bridge#5-optional-running-as-a-daemon-with-systemctl

closed 2018-06-08 by Koenkk