Koenkk/zigbee2mqtt #97: Port speed for cc2530 models

kirovilya, 2018-06-06:
cc2530 models connected as //dev/ttyUSB0 by default has "bad" baudrate (and rtscts too)... Set it on connection in shepherdSettings:

```
const shepherdSettings = {
    net: {
        panId: advancedSettings && advancedSettings.pan_id ? advancedSettings.pan_id : 0x1a62,
        channelList: [advancedSettings && advancedSettings.channel ? advancedSettings.channel : 11],
    },
    dbPath: data.joinPath('database.db'),
    sp: { baudrate: 115200, rtscts: false },
};
```

ciotlosm, 2018-06-06:
Is this configurable in zigbee2mqtt? If it is, we should include documentation around configuration to specify this, and also update CC2530 model page with details on correct values.

kirovilya, 2018-06-06:
@ciotlosm It be static. Not need configurable. I think so.

ciotlosm, 2018-06-14:
@kirovilya I've added support in configuration for both `baudrate` and `rtscts`. See https://github.com/Koenkk/zigbee2mqtt/pull/123. Let me know if that is good enough.

PR #123 (2018-06-15): Port speed for cc2530 models
Fixes #97 

- [x] Allow port `baudrate` and `rtscts` from `configuration.yaml`
- [x] Set defaults to those in zigbee-shepherd
- [x] Update documentation
commit 01effab2 (2018-06-15): Fixes #97