Koenkk/zigbee2mqtt #102: Ikea Trafri Button

ciB89, 2018-06-07:
Ikea offers a controller that goes along with its TRADFRI series [link](https://www.ikea.com/de/de/catalog/products/30338849/?cid=ps%7Cde%7Cno_cc_topic%7C4MLRrWwPr0gHjIOUyfOOktEtkrhBQM)

It's not supported on the wiki, but I was wondering if it would work with zigbee2mqtt?

ciotlosm, 2018-06-07:
@ciB89 if you have the controller, you can just try it, but it's more likely that will work with some tweaking.

tb-killa, 2018-06-07:
@ciB89 @ciotlosm I wait for my Tradfri Controller but i´m at the moment I am busy with my work.
I know that @Koenkk  has also bought a controller and do some tests in the past.
Maybe he could also help.

Do you already own the controller?

Maybe we could do some hacky like we do with the "Hue dimmer switch" https://github.com/Koenkk/zigbee2mqtt/issues/36 ?

From what I know we could do the same bind for "genOnOff".

ciB89, 2018-06-07:
I've got one now. It seems that the switch wants to be paired directly to the device that you want to use it with, e.g. lamp or an LED power supply. There's a pairing button on it that you are supposed to press >10 seconds until a red light turns on that indicates that the remote is paired with that device (it can be linked to ten devices). The bridge is needed for this, although it is never paired with the bridge directly (as far as I can tell from the instructions). I tried pairing it with the zigbee stick on my HA, but the red light won't turn on and debug doesn't show anything interesting either.

tb-killa, 2018-06-07:
Please provide your conent of the database.db with your "paired" ikea device.
It would also be great if you start your instance via DEBUG=* npm start like mentioned via wiki.

About this LED i could say after doing some manual pairing with the Hue Dimmer Switch the device light is turning on action. 

We could test the same code sample if you like ?

Koenkk, 2018-06-07:
I already spend quite some time on this:

- The button can be paired but doesnt send anything to the gateway, it looks that it can only be used when paired directly with a bulb
- The same goes for the motion sensor
- The dimmer, after some hacking, works. I will add support for it soon.

tb-killa, 2018-06-07:
@Koenkk Do you test via manual mapping like we do with the Hue Dimmer Switch ?
I found some "SmartThings" groovy Code like we show for the Hue Device, there are also the two mentioned mappings.
Sadly i doesn´t have such type of device now so maybe you could check if it could work ?

Koenkk, 2018-06-07:
Do you have a link to the SmartThings code?

tb-killa, 2018-06-07:
https://github.com/RichMercer/SmartThings/blob/master/ikea-tradfri-remote.groovy

For other one ( the second binding) has to wait until tomorrow so i could send from my PC because i save the URL.

Koenkk, 2018-06-07:
@tb-killa that looks interesting, it binds the genOnOff cluster on endpoint 1 till 4 to the coordinator. Haven't tried that before, will run some tests tomorrow.

kirovilya, 2018-06-08:
I have this remote control.
It is successfully connected to coordinator. But coordinator not receive it's messages.
I found why! http://faire-ca-soi-meme.fr/domotique/2018/03/14/une-telecommande-zigbee-pour-les-controler-tous/
It's send message to "group". Coordinator not endDevice in group and do not recived it's message.

I do not known what to do... May be put coordinator to group? But how?

ciB89, 2018-06-08:
Got the remote paired with HA after I paired the driver with HA first. Receive no messages though.
Here is the database.db for the remote:

```
{"id":5,"type":"EndDevice","ieeeAddr":"0x90fd9ffffed97ed6","nwkAddr":24438,"manufId":4476,"epList":[1],"status":"online","joinTime":1528475316,"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{}}},"_id":"v7bdc5jXYmRkWPrr"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x90fd9ffffed97ed6","nwkAddr":24438,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"status":"online","joinTime":1528475316,"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{}}},"_id":"v7bdc5jXYmRkWPrr"}
{"id":5,"type":"EndDevice","ieeeAddr":"0x90fd9ffffed97ed6","nwkAddr":24438,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"status":"online","joinTime":1528475316,"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{}},"genPowerCfg":{"dir":{"value":1},"attrs":{"65533":1,"batteryVoltage":29,"batteryPercentageRemaining":74}},"genIdentify":{"dir":{"value":3},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{"65533":1}},"genOta":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"v7bdc5jXYmRkWPrr"}
```

Koenkk, 2018-06-08:
I don't know if the coordinator can be put in a group, that article shows how to use the TRADFRI remote to control a Xiaomi plug, but still it doesn't send events to the coordinator.

tb-killa, 2018-06-08:
@ciB89 : please add this to your devices.js
```
{
        zigbeeModel: ['TRADFRI remote control'],
        model: 'TODO',
        vendor: 'IKEA',
        description: 'TODO',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [],
    },
````

and this to  ```lib/zigbee.js ``` into the ```handleReady function```

```
        // TEMP: Bind Ikea Remote Control to coordinator.
        let ikearemote = '0x90fd9ffffed97ed6';
        ikearemote = this.shepherd.find(ikearemote, 1);
        let coordinator = this.shepherd.list().find((d) => d.type === 'Coordinator');
        coordinator = this.shepherd.find(coordinator.ieeeAddr, 1);
        ikearemote.bind('genOnOff', coordinator, (err) => {
            if (err) {
                console.log('FAILED TO BIND genOnOff :(');
            } else {
                console.log('GOOD TO BIND genOnOff !');
            }
        });
        //
        ikearemote.bind('genLevelCtrl', coordinator, (err) => {
            if (err) {
                console.log('FAILED TO BIND genLevelCtrl :(');
            } else {
                console.log('GOOD TO BIND genLevelCtrl !');
            }
        });
```
After that please start you zigbee2mqtt instance via ```DEBUG=* npm start ``` and let the Remote Control in pairing mode while the instance start up.
Please provide log after you wait some time and push some of the buttons of the remote control.

ciB89, 2018-06-11:
Restarting the docker in Hassio resets the custom changes to the files... I'll setup a second SD-card with Hassbian and I'll test again.

ciB89, 2018-06-11:
```
2018-6-11 15:37:37 INFO zigbee-shepherd started
2018-6-11 15:37:37 INFO Currently 0 devices are joined:
2018-6-11 15:37:37 INFO Connecting to MQTT server at mqtt://192.168.0.41
2018-6-11 15:37:38 INFO zigbee-shepherd ready
/opt/zigbee2mqtt/lib/zigbee.js:92
        ikearemote.bind('genOnOff', coordinator, (err) => {
                   ^

TypeError: Cannot read property 'bind' of undefined
    at Zigbee.handleReady (/opt/zigbee2mqtt/lib/zigbee.js:92:20)
    at emitNone (events.js:106:13)
    at ZShepherd.emit (events.js:208:7)
    at Immediate._onImmediate (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/lib/shepherd.js:89:18)
    at runCallback (timers.js:810:20)
    at tryOnImmediate (timers.js:768:5)
    at processImmediate [as _immediateCallback] (timers.js:745:5)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@0.1.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@0.1.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

tb-killa, 2018-06-11:
@ciB89 This failure seems to be come forward if we use this type of debugging Code without first has got the Enddevice in the database (pairing).
So lets do some practise first:
Comment out the Code Section for binding (zigbee.js).
Restart zigbee2mqtt and pair the device like a normal device so we get the needing information at startup inside the zigbee database.
After that stop zigbee2mqtt and bring the code section in work, and restart them into debug=*

If we use the new code from @Koenkk we could do the binding from the converter themselve but we would test it first :)

ciB89, 2018-06-11:
The binding did work ("GOOD TO BIND genOnOff !"), but the debug does not show me anything when pressing buttons..

ciB89, 2018-06-11:
After restarting the shephard it fails the binding although I reproduce every step I had taken sofar..

tb-killa, 2018-06-11:
Bring the Device in pairing mode again, maybe it need this twice because of standby or another situation.
It could need some sort of multiple restarts of zigbee2mqtt instance or maybe multiple usage of on / off button pressing to bring the needing informations on top.

fails could be the situations mentioned on top because i get this on the "Hue dimmer switch" too.

tb-killa, 2018-06-11:
We should also add some debugging console.log to check the raw data:

try put console.log after zigbee-shepherd\lib\components\af.js:693
and after zigbee-shepherd\lib\components\af.js:572

sample like this:
```
// Original
var coord = af.controller.getCoord();
// Added
console.log(type, msg)
```

and
```
//Original
zcl.parse(msg.data, function (err, zclData) {
                if (!err)
                    zclIncomingParsedMsgEmitter(msg, zclData);
                    // Added
                    console.log(type, msg)
            });
```

Now we could check deeper with ```DEBUG=* npm start```

ciB89, 2018-06-11:
So, I took out the batteries and reinserted them and that worked really well - it seems that just pressing the pairing button only works for the IKEA stuff, not so much for the shephard (the red light indicator for pairing only turns on if e.g. the driver is plugged in).

Button presses on the remote still have no effect. I spotted this though:
`2018-06-11T17:51:14.477Z - error: Cannot get the Node Descriptor of the Device: 0x90fd9ffffed97ed6 (Error: Timed out after 10000 ms)
`
I will add the additional debugging stuff next.

ciB89, 2018-06-11:
This is the full log. BTW, my log.txt doesn't actually show the entire log. How can I change that?

```
> zigbee2mqtt@0.1.0 start /opt/zigbee2mqtt
> node index.js

2018-6-11 18:22:32 INFO Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp The serialport /dev/ttyACM0 is opened. +0ms
  zigbee-shepherd:request REQ --> SYS:osalNvRead +38ms
  cc-znp:SREQ --> SYS:osalNvRead, { id: 3840, offset: 0, len: 1, value: [ 85 ] } +9ms
  serialport write 8 bytes of data +0ms
  cc-znp:SRSP <-- SYS:osalNvRead, { status: 0, len: 1, value: <Buffer 55> } +75ms
  zigbee-shepherd:request RSP <-- SYS:osalNvRead, status: 0 +82ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +17ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 131, len: 2, value: [ 98, 26 ] } +20ms
  serialport write 6 bytes of data +78ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 131, len: 2, value: <Buffer 62 1a> } +26ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +28ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 132, len: 4, value: [ 0, 8, 0, 0 ] } +14ms
  serialport write 6 bytes of data +40ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 132, len: 4, value: <Buffer 00 08 00 00> } +14ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +16ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +11ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 98, len: 16, value: [ 1, 3, 5, 7, 9, 11, 13, 15, 0, 2, 4, 6, 8, 10, 12, 13 ] } +13ms
  serialport write 6 bytes of data +28ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 98, len: 16, value: <Buffer 01 03 05 07 09 0b 0d 0f 00 02 04 06 08 0a 0c 0d> } +34ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +36ms
  zigbee-shepherd:request REQ --> SAPI:readConfiguration +12ms
  cc-znp:SREQ --> SAPI:readConfiguration, { configid: 99, len: 1, value: [ 0 ] } +14ms
  serialport write 6 bytes of data +47ms
  cc-znp:SRSP <-- SAPI:readConfiguration, { status: 0, configid: 99, len: 1, value: <Buffer 00> } +15ms
  zigbee-shepherd:request RSP <-- SAPI:readConfiguration, status: 0 +17ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +13ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +15ms
  serialport write 6 bytes of data +30ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 09 00 ff fe 04 66 04> } +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:init Now the ZNP is a coordinator. +2ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 0 } +4ms
  serialport write 6 bytes of data +12ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 0, value: <Buffer 09 09 00 ff fe 04 66 04> } +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 1 } +3ms
  serialport write 6 bytes of data +11ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 1, value: <Buffer 64 31 02 12 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 2 } +3ms
  serialport write 6 bytes of data +10ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 2, value: <Buffer 00 00 02 12 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 5 } +3ms
  serialport write 6 bytes of data +10ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 5, value: <Buffer 0b 00 02 12 00 4b 12 00> } +7ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +1ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 6 } +3ms
  serialport write 6 bytes of data +10ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 6, value: <Buffer 62 1a 02 12 00 4b 12 00> } +8ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +9ms
  zigbee-shepherd:request REQ --> SAPI:getDeviceInfo +2ms
  cc-znp:SREQ --> SAPI:getDeviceInfo, { param: 7 } +2ms
  serialport write 6 bytes of data +10ms
  cc-znp:SRSP <-- SAPI:getDeviceInfo, { param: 7, value: <Buffer dd dd dd dd dd dd dd dd> } +9ms
  zigbee-shepherd:request RSP <-- SAPI:getDeviceInfo +10ms
[... 1663 lines cut ...]
  data:
   { '0': 24,
     '1': 22,
     '2': 1,
     '3': 253,
     '4': 255,
     '5': 0,
     '6': 33,
     '7': 1,
     '8': 0 },
  zclMsg:
   { frameCntl: { frameType: 0, manufSpec: 0, direction: 1, disDefaultRsp: 1 },
     manufCode: 0,
     seqNum: 22,
     cmdId: 'readRsp',
     payload: [ [Object] ] } }
  zigbee2mqtt:controller Recieved zigbee message with data 0x90fd9ffffed97ed6 +4s
2018-6-11 18:23:52 INFO Connecting with device...
2018-6-11 18:23:52 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd Device: 0x90fd9ffffed97ed6 join the network. +37ms
  zigbee2mqtt:controller Recieved zigbee message with data 0x90fd9ffffed97ed6 +20ms
2018-6-11 18:23:52 INFO Device incoming...
2018-6-11 18:23:52 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:controller Recieved zigbee message with data online +6ms
```

ciotlosm, 2018-06-11:
Use hastebin next time to reduce message length. This is bloating the topic too much

james-fry, 2018-06-12:
IIRC this device and the tradfri dimmer are working with zigate and deconz.
I think the coordinator creates a dummy group (with zero members) and then the controller joins that group. I saw some info about it on the HA community forum. Will see if I can find it again...

james-fry, 2018-06-12:
See here:
https://community.home-assistant.io/t/use-ikea-tradfri-dimmer-in-home-assistant/33211/35

ciB89, 2018-06-18:
Is there any progress on this yet?

Koenkk, 2018-06-18:
Not from my side, I don't know how to create a fake group using zigbee-shepherd.

tb-killa, 2018-06-19:
@Koenkk @kirovilya  : I use the latest Version from git ```INFO Starting zigbee2mqtt version 0.0.0 (commit #4c4fcd7)```

If i bring my remote control in pairing i got this:

```
zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +9s
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +75ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +7ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +984ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +5ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +1s
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +1s
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +85ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +20ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +949ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
```

Koenkk, 2018-06-19:
What if you do:

```bash
rm -rf node_modules
npm install
```

tb-killa, 2018-06-19:
@Koenkk Its a complete fresh installation, do this alright before posting.
@kirovilya this looks interesting:
After restart in Debug:
```
6/19/2018, 10:23:10 PM INFO Currently 1 devices are joined:
6/19/2018, 10:23:10 PM INFO unknown (0x90fd9ffffed17c15): unkown - undefined unknown (EndDevice)
```
It seems to be that zigbee2mqtt doesnt collect the right data from the remote.

kirovilya, 2018-06-19:
@tb-killa It is normal for me :) Just put block in devices.js of zigbee-shepherd-converters:
```
    {
        zigbeeModel: ['TRADFRI remote control'],
        model: 'E1524',
        vendor: 'IKEA',
        description: 'TRADFRI remote control',
        supports: '',
        fromZigbee: [],
        toZigbee: [],
    },
```

kirovilya, 2018-06-19:
@tb-killa or try remove and rejoin remote

tb-killa, 2018-06-19:
@kirovilya converter entry was present. I remove and add the device multiple times.

I insert some log output inside the foundation.js file 
```
ru.clause('variable', function (name, dataTypeParam) {
    if (!dataTypeParam) dataTypeParam = 'dataType';

    this.tap(function () {
        var dataType = getDataType(this.vars[dataTypeParam]);
        console.log(dataType)
        console.log(name)
        ru[dataType](name)(this);
    });
});
```

and got this log 
```
zigbee-shepherd:msgHdlr IND <-- ZDO:tcDeviceInd +3s
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +88ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +5ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +997ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +2ms
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +103ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +917ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +3ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +1s
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +37ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +17ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +977ms
strPreLenUint8
attrData
undefined
attrData
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:685
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:685:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
```

kirovilya, 2018-06-19:
@tb-killa need log dataTypeParam too.
unknown datatype - strange...

tb-killa, 2018-06-20:
I extend the needing output
```
6/20/2018, 6:58:20 PM INFO MQTT publish, topic: 'zigbee2mqtt/bridge/state', payload: 'online'
  cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 52708, status: 0, nwkaddr: 52708, activeepcount: 1, activeeplist: <Buffer 01> } +72ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +5ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +6ms
  cc-znp:SRSP <-- ZDO:mgmtPermitJoinReq, { status: 0 } +18ms
  cc-znp:AREQ <-- ZDO:permitJoinInd, { duration: 255 } +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:permitJoinInd +1ms
  cc-znp:AREQ <-- ZDO:mgmtPermitJoinRsp, { srcaddr: 0, status: 0 } +1ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:mgmtPermitJoinRsp +1ms
  cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 52708, nwkaddrofinterest: 52708, endpoint: 1 } +1ms
  serialport write 10 bytes of data +3ms
  cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +15ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 52708, status: 0, nwkaddr: 52708, len: 34, endpoint: 1, profileid: 49246, deviceid: 2096, deviceversion: 2, numinclusters: 6, inclusterlist: <Buffer 00 00 01 00 03 00 09 00 05 0b 00 10>, numoutclusters: 7, outclusterlist: <Buffer 03 00 04 00 05 00 06 00 08 00 19 00 00 10> } +911ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +2ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +36ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 52708, destendpoint: 1, srcendpoint: 1, clusterid: 0, transid: 1, options: 48, radius: 30, len: 9, data: <Buffer 00 01 00 04 00 05 00 07 00> } +3ms
  serialport write 24 bytes of data +4ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +13ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +2ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 1 } +972ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +9ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 52708, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 0, securityuse: 0, timestamp: 6222012, transseqnumber: 0, len: 54, data: <Buffer 18 01 01 04 00 00 42 0e 49 4b 45 41 20 6f 66 20 53 77 65 64 65 6e 05 00 00 42 16 54 52 41 44 46 5249 20 72 65 6d 6f 74 65 20 63 6f 6e 74 72 6f 6c 07 ... > } +11ms
undefined
strPreLenUint8
attrData
undefined
undefined
attrData
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:686
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:686:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
```

tb-killa, 2018-07-01:
@kirovilya 
I started with fresh flashed stick and new zigbee2mqtt instance:

Sadly i got the error again:

```

  cc-znp:AREQ <-- ZDO:endDeviceAnnceInd, { srcaddr: 20585, nwkaddr: 20585, ieeeaddr: '0x90fd9ffffed17c15', capabilities: 128 } +2m
spinlock: false []
  zigbee-shepherd:request REQ --> ZDO:nodeDescReq +6ms
  cc-znp:SREQ --> ZDO:nodeDescReq, { dstaddr: 20585, nwkaddrofinterest: 20585 } +3ms
  serialport write 9 bytes of data +4ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:endDeviceAnnceInd +3ms
  cc-znp:SRSP <-- ZDO:nodeDescReq, { status: 0 } +51ms
  cc-znp:AREQ <-- ZDO:nodeDescRsp, { srcaddr: 20585, status: 0, nwkaddr: 20585, logicaltype_cmplxdescavai_userdescavai: 2, apsflags_freqband: 64, maccapflags: 128, manufacturercode: 4476, maxbuffersize: 82, maxintransfersize: 82, servermask: 0, maxouttransfersize: 82, descriptorcap: 0 } +2s
  zigbee-shepherd:msgHdlr IND <-- ZDO:nodeDescRsp +8ms
  zigbee-shepherd:request REQ --> ZDO:activeEpReq +12ms
  cc-znp:SREQ --> ZDO:activeEpReq, { dstaddr: 20585, nwkaddrofinterest: 20585 } +5ms
  serialport write 9 bytes of data +3ms
  cc-znp:SRSP <-- ZDO:activeEpReq, { status: 0 } +22ms
  cc-znp:AREQ <-- ZDO:activeEpRsp, { srcaddr: 20585, status: 0, nwkaddr: 20585, activeepcount: 1, activeeplist: <Buffer 01> } +990ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:activeEpRsp +2ms
  zigbee-shepherd:request REQ --> ZDO:simpleDescReq +9ms
  cc-znp:SREQ --> ZDO:simpleDescReq, { dstaddr: 20585, nwkaddrofinterest: 20585, endpoint: 1 } +2ms
  serialport write 10 bytes of data +2ms
  cc-znp:SRSP <-- ZDO:simpleDescReq, { status: 0 } +12ms
  cc-znp:AREQ <-- ZDO:simpleDescRsp, { srcaddr: 20585, status: 0, nwkaddr: 20585, len: 34, endpoint: 1, profileid: 49246, deviceid: 2096, deviceversion: 2, numinclusters: 6, inclusterlist: <Buffer 00 00 01 00 03 00 09 00 05 0b 00 10>, numoutclusters: 7, outclusterlist: <Buffer 03 00 04 00 05 00 06 00 08 00 19 00 00 10> } +998ms
  zigbee-shepherd:msgHdlr IND <-- ZDO:simpleDescRsp +10ms
  zigbee-shepherd:request REQ --> AF:dataRequest, transId: 1 +33ms
  cc-znp:SREQ --> AF:dataRequest, { dstaddr: 20585, destendpoint: 1, srcendpoint: 1, clusterid: 0, transid: 1, options: 48, radius: 30, len: 9, data: <Buffer 00 01 00 04 00 05 00 07 00> } +3ms
  serialport write 24 bytes of data +3ms
  cc-znp:SRSP <-- AF:dataRequest, { status: 0 } +12ms
  zigbee-shepherd:request RSP <-- AF:dataRequest, status: 0 +1ms
  cc-znp:AREQ <-- AF:dataConfirm, { status: 0, endpoint: 1, transid: 1 } +972ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: dataConfirm, msg: [object Object] +3ms
  zigbee-shepherd:msgHdlr IND <-- AF:dataConfirm, transId: 1 +4ms
  cc-znp:AREQ <-- AF:incomingMsg, { groupid: 0, clusterid: 0, srcaddr: 20585, srcendpoint: 1, dstendpoint: 1, wasbroadcast: 0, linkquality: 18, securityuse: 0,timestamp: 16490997, transseqnumber: 0, len: 54, data: <Buffer 18 01 01 04 00 00 42 0e 49 4b 45 41 20 6f 66 20 53 77 65 64 65 6e 05 00 00 42 16 54 52 41 44 46 52 49 20 72 65 6d 6f 74 65 20 63 6f 6e 74 72 6f 6c 07 ... > } +9ms
  zigbee-shepherd:af dispatchIncomingMsg(): type: incomingMsg, msg: [object Object] +3ms
/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
    at Dissolve.<anonymous> (/opt/zigbee2mqtt/node_modules/zcl-packet/lib/foundation.js:683:21)
    at Dissolve._transform (/opt/zigbee2mqtt/node_modules/dissolve/index.js:80:16)
    at Dissolve.Transform._read (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:184:10)
    at Dissolve.Transform._write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_transform.js:172:12)
    at doWrite (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:279:12)
    at writeOrBuffer (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:266:5)
    at Dissolve.Writable.write (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:211:11)
    at Dissolve.Writable.end (/opt/zigbee2mqtt/node_modules/dissolve/node_modules/readable-stream/lib/_stream_writable.js:426:10)
    at FoundPayload._getObj (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:242:12)
    at getObjCB (/opt/zigbee2mqtt/node_modules/zigbee-shepherd/node_modules/zcl-packet/lib/foundation.js:51:22)
```

I often reset the remote device via 4 pushes of the button on the back and get the device leaving message complete.
But after the device seems to device again, it crashed ...

kirovilya, 2018-07-02:
@tb-killa It is the same error like https://github.com/Koenkk/zigbee2mqtt/issues/129 ... Something in your zigbee-network want to connect but can't :)

tb-killa, 2018-07-02:
@kirovilya Yes it seems to be some of this error. 
I will flash the original Firmware from TI Source themselve back to my stick because i think there is some sort of break in our diy firmware.
Or maybe also use some sort of older snapshot of zigbee2mqtt from github to check if we didn´t break something in the last time.

I only know that with an older release the remote device work one time.
after that i updated and after that its broke and i got on different devices in pairing mode this crazy "crash".

tb-killa, 2018-07-04:
I got it now:
```
spinlock: false []
spinlock: true []
2018-7-4 21:54:44 [32mINFO[39m Connecting with device...
2018-7-4 21:54:44 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:54:49 [32mINFO[39m Connecting with device...
2018-7-4 21:54:49 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:54:59 [32mINFO[39m Connecting with device...
2018-7-4 21:54:59 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:06 [32mINFO[39m Connecting with device...
2018-7-4 21:55:06 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:13 [32mINFO[39m Connecting with device...
2018-7-4 21:55:13 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:20 [32mINFO[39m Connecting with device...
2018-7-4 21:55:20 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:27 [32mINFO[39m Connecting with device...
2018-7-4 21:55:27 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:34 [32mINFO[39m Connecting with device...
2018-7-4 21:55:34 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:41 [32mINFO[39m Connecting with device...
2018-7-4 21:55:41 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:48 [32mINFO[39m Connecting with device...
2018-7-4 21:55:48 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:55 [32mINFO[39m Connecting with device...
2018-7-4 21:55:55 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
2018-7-4 21:55:55 [32mINFO[39m Device incoming...
2018-7-4 21:55:55 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-7-4 21:55:55 [32mINFO[39m New device with address 0x90fd9ffffed17c15 connected!
2018-7-4 21:55:55 [32mINFO[39m MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x90fd9ffffed17c15"}'
2018-7-4 21:55:55 [33mWARN[39m Device with modelID 'TRADFRI remote control' is not supported.
2018-7-4 21:55:55 [33mWARN[39m Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-7-4 21:55:55 [33mWARN[39m Device with modelID 'TRADFRI remote control' is not supported.
2018-7-4 21:55:55 [33mWARN[39m Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

this is my database.db entry:
```
{"id":4,"type":"EndDevice","ieeeAddr":"0x90fd9ffffed17c15","nwkAddr":30452,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Battery","modelId":"TRADFRI remote control","epList":[1],"status":"online","joinTime":1530734067,"endpoints":{"1":{"profId":49246,"epId":1,"devId":2096,"inClusterList":[0,1,3,9,2821,4096],"outClusterList":[3,4,5,6,8,25,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI remote control","dateCode":"20170302","powerSource":3,"swBuildId":"1.2.214"}},"genPowerCfg":{"dir":{"value":1},"attrs":{"65533":1,"batteryVoltage":23,"batteryPercentageRemaining":5}},"genIdentify":{"dir":{"value":3},"attrs":{}},"genGroups":{"dir":{"value":2},"attrs":{}},"genScenes":{"dir":{"value":2},"attrs":{}},"genOnOff":{"dir":{"value":2},"attrs":{}},"genLevelCtrl":{"dir":{"value":2},"attrs":{}},"genAlarms":{"dir":{"value":1},"attrs":{}},"genOta":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{}},"lightLink":{"dir":{"value":3},"attrs":{}}}}},"_id":"iZKH6uOfcdRWxh97"}
```

kirovilya, 2018-07-04:
I get this error too:
```
/node_modules/zcl-packet/lib/foundation.js:683
        ru[dataType](name)(this);
                    ^

TypeError: ru[dataType] is not a function
```
Will fix like here https://github.com/Koenkk/zigbee2mqtt/issues/150#issuecomment-402549850 after success

tb-killa, 2018-07-06:
@kirovilya Maybe you could check my log (DEBUG=*) for Data.
Maybe we see some infos about how we gould bind the device ??
https://pastebin.com/E7x33r4t

kirovilya, 2018-07-07:
@tb-killa it is only "pairing" log. Here no new info for me.
How I undestanging, Remote Control operate with group of EndDevice and send them messages.
But in our coordinator we can not see this messages cause we not EndDevice and not in group with Remote Control.
If we create in coordinator any endpoint like EndDevice and bind it to group of Remote - we see messages. But I do not known how to do it...

tb-killa, 2018-08-05:
@Koenkk @kirovilya 
Does somebody test the extension frameworks ?
https://github.com/zigbeer/zive/wiki
https://github.com/zigbeer/ziee/wiki
https://github.com/zigbeer/zapp-cie/wiki

Maybe we could build simple zive or ziee application who build sample endpoint with needing acl and attributes and simple bind our endpoint 1 to this ?

I´v read the zapp-cie sample app, it provide iaszone samples too, maybe with this we could also simple integrate our motion / windows - sensors ??

kirovilya, 2018-08-06:
no, I did not try.

Koenkk, 2018-08-07:
I also didn't, but it calls the same code as we did.

tb-killa, 2018-08-07:
I thought that we could use Zive or Ziee for building endpoint fake Group so Devices like ikea remote or motion sensor could be Bind to it?
But if its also same as we used actually.. We have to Show..

worf0r, 2018-08-17:
I closed the othe Issue. I have some of this Buttons and sadly can't use them with this soultion. How did you pair the button? 10 Seconds button pressing and then?

Koenkk, 2018-08-17:
@worf0r press the reset button quickly 4 times (red light flashes) while keeping it close to the coordinator.

worf0r, 2018-08-20:
Ok. Pairing works. When I pair Button directly with a light, the light seems to only speak with the button. Not with iobroker adapter :(

Edzilla2000, 2018-08-31:
I would be quite interested in a working solution for this as well, as the tradfri remotes are quite nice, with the ability to change scenes and brightness.

I managed a workaround, but it is quite a bit hacky:
I paired a single bulb out of five with tradfri and the remote, and that bulb was imported into HA.
Next, I paired the remaining 4 bulbs with zigbee2mqtt and added them in HA.
Then I made a light group in HA containing all five bulbs:

```
light:
  - platform: group
    name: salon
    entities:
      - light.salon1_2
      - light.salon2
      - light.salon3
      - light.salon4
      - light.salon5_2
```
Then I made two automations: the first one matches the brightness and color temp of the tradfi paired bulb to the other 4
```
- id: '1525140123412'
  alias: salon off -> on
  trigger:
    - platform: state
      entity_id:
        - light.salon1_2
  action:
    service: light.turn_on
    entity_id:
        - light.salon2
        - light.salon3
        - light.salon4
        - light.salon5_2
    data_template:
      brightness: "{{ state_attr('light.salon1_2', 'brightness') }}"
      color_temp: "{{ state_attr('light.salon1_2', 'color_temp') }}"

```
the other one matches the "off" state of the tradfri bulb to the other 4:
```
- id: '1525140123413'
  alias: salon on -> off
  trigger:
    - platform: state
      entity_id:
        - light.salon1_2
      from: 'on'
      to: 'off'
  action:
    service: light.turn_off
    entity_id:
        - light.salon2
        - light.salon3
        - light.salon4
        - light.salon5_2

```

Obviously this is not a great solution as it requires that I keep the tradfri bridge working, but I'm using the ikea bulbs as router for the zigbee network and I need the range the bulbs give me...

tb-killa, 2018-09-02:
@Koenkk @kirovilya 
From what we actually know .. and this seems to be the same as "deCONZ" be do...
Device is joining to coordinator .. after that we have to create dummy group ...
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/23

and join the device to this group-id .. after that the remote controll sends cluster with broadcast to group-id ... we have to join now coordinator or fake end-device to this group too.

for this we have to use some sort if framework code -> https://github.com/Koenkk/zigbee2mqtt/issues/102?_pjax=%23js-repo-pjax-container#issuecomment-410548441

to build ep with known clusters like genonoff ... 

Do you think this could be a solution ? 
I need help to check this ...

BTW.. we should check if could open some sort of developer chat for this big project .. could be help ;)

kirovilya, 2018-09-03:
@tb-killa I've been thinking about this before.
But I still have not found a way to create the fake-endpoint with group on the coordinator.
I plan to sort out the zigbee-groups first. To be able to operate by including, excluding devices from zigbee-groups.
Then, learn how to bind endpoints of devices so that you can manage a group of devices.
And then think about how to create a fake device on the coordinator (I'm afraid it will not work).

but I'm still busy and have no time :(

RonnyAL, 2018-09-25:
Anyone still working on this? I've got this remote but currently no sockets compatible with my IKEA bulbs, so it's just laying around. Would be awesome if I could utilize this remote again! :)

markusn1, 2018-10-10:
As it does not seem likely to get the remote working with zigbee2mqtt, is there at least a way I can use zigbee2mqtt AND the remote to controll the light? I was not able to pair the remote with the lamp, after already pairing the lamp with zigbee2mqtt. I guess it's called pairing for a reason, but maybe there is a way?

wimpie007, 2018-10-14:
@markusn1 : I was able to pair a light with zigbee2mqtt AND a remote.
First, i paired the lamp (turning it of 6 times...), and then reset the remote (4 clicks?) and then long-press the remote. Now both zigbee2mqtt and the remote  can control the lamp, but the status of the lamp isnt reported back to HA when changed with the remote, so that's an annoyance...

kiall, 2018-10-17:
https://github.com/Koenkk/zigbee2mqtt/issues/15 looks related. The switches control a group, and groups are as yet not supported - am I missing something?

thoj, 2018-10-27:
IKEA is pushing these hard. Especially for their new colored bulbs. Anything i can do to help? I'm not sure where to start.

Koenkk, 2018-11-07:
@tynor88 it has the same issue as the button.

apiel, 2018-11-08:
To create a dummy device, it might be possible with Zive:

```
var ziee = new Ziee();
    ziee.init('genGroups', 'dir', { value: 1 });
    ziee.init('genLevelCtrl', 'dir', { value: 1 });
    // here it will be necessary to init all attributes...

    var localEp = new Zive({
        profId: 260,  // 'HA'
        devId: 257,   // 'dimmableLight'
        discCmds: []
    }, ziee);

    // in zigbee2mqtt should be something like this.zigbee.shepherd.mount...
    shepherd.mount(localEp, function (err, epId) {
        if (!err) {
            console.log('local endpoint mounted', epId);  // 11
            // now we have an endpoint in the coordinator
            // in my case it is endpoint 11

           // then we have to say the endpoint to join a group
           var ep = shepherd.find(getCoordinatorAddr(), epId);
           var group = 1;
           ep.bind('genLevelCtrl', group, (err) => { console.log('bind coor genLevelCtrl done', err); });
        }
    });
```
I still didn't managed to get action from my ikea remote... But hopefully this is the way to go, let's see, I will keep trying.

picsou83, 2018-11-25:
Hi

Very cool, 
I managed to create a group with the functionnals commands and it works, My TRADFRI on/off switch control my outlet.

to optimize all that I let you do it :-)

I added the tradfri control outlet on group 2 and 
i added the trafri on off switch  on group 2

and magic no bind to do

the code I used
````
const telecommande = '0x000d6ffffea4efc6';
           const ep2 = this.zigbee.shepherd.find(telecommande, 1);
                           
 ep2.functional('genGroups', 'add', {  groupid: 0x0002 , groupname: 'group2' } , function (err, data) {
  logger.info(err);    
        logger.info(data);    
});

````

and 

````
const prise =  '0x000d6ffffe2f0a14';        
    const ep1 = this.zigbee.shepherd.find(prise, 1);
    
                           
 ep1.functional('genGroups', 'add', {  groupid: 0x0002 , groupname: 'group2' } , function (err, data) {
  logger.info(err);    
        logger.info(data);    
});

````

also with dummy device it's possible to have a return of status of a remote control with the group

[119 more replies]

closed 2019-02-18 by Koenkk