Koenkk/zigbee2mqtt #104: Ikea Tradfri driver

ciB89, 2018-06-08:
I successfully paired the [driver](https://www.ikea.com/us/en/catalog/products/10356189/) for LED-strips with HA. I have no idea though how to create the converter since there are no buttons on the driver to trigger anything.

```
2018-6-8 18:23:52 INFO Device incoming...
2018-6-8 18:23:52 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
2018-6-8 18:23:52 INFO New device with address 0x90fd9ffffe73e86a connected!
2018-6-8 18:23:52 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"device_connected","message":"0x90fd9ffffe73e86a"}'
2018-6-8 18:23:52 WARN Device with modelID 'TRADFRI transformer 30W' is not supported.
2018-6-8 18:23:52 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
2018-06-08T16:23:52.327Z zigbee2mqtt:controller Recieved zigbee message with data online
2018-6-8 18:23:52 WARN Device with modelID 'TRADFRI transformer 30W' is not supported.
2018-6-8 18:23:52 WARN Please see: https://github.com/Koenkk/zigbee2mqtt/wiki/How-to-support-new-devices
```

Here is the database.db entry for the device:

```
{"id":4,"type":"Router","ieeeAddr":"0x90fd9ffffe73e86a","nwkAddr":5512,"manufId":4476,"epList":[1],"status":"online","joinTime":1528475028,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{}}},"_id":"eKKWJ5wwo2kDYzW1"}
{"id":4,"type":"Router","ieeeAddr":"0x90fd9ffffe73e86a","nwkAddr":5512,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI transformer 30W","epList":[1],"status":"online","joinTime":1528475028,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{}}},"_id":"eKKWJ5wwo2kDYzW1"}
{"id":4,"type":"Router","ieeeAddr":"0x90fd9ffffe73e86a","nwkAddr":5512,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI transformer 30W","epList":[1],"status":"online","joinTime":1528475028,"endpoints":{"1":{"profId":49246,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"dir":{"value":1},"attrs":{"10":{},"65533":1,"zclVersion":1,"appVersion":17,"stackVersion":87,"hwVersion":1,"manufacturerName":"IKEA of Sweden","modelId":"TRADFRI transformer 30W","dateCode":"20170529","powerSource":1,"swBuildId":"1.2.245"}},"genIdentify":{"dir":{"value":1},"attrs":{"65533":1,"identifyTime":0}},"genGroups":{"dir":{"value":1},"attrs":{"65533":1,"nameSupport":0}},"genScenes":{"dir":{"value":3},"attrs":{"65533":1,"count":0,"currentScene":0,"currentGroup":0,"sceneValid":0,"nameSupport":0}},"genOnOff":{"dir":{"value":1},"attrs":{"16387":1,"65533":1,"onOff":1,"globalSceneCtrl":1,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"dir":{"value":1},"attrs":{"15":0,"16384":254,"65533":1,"currentLevel":254,"remainingTime":0,"onOffTransitionTime":5,"onLevel":255}},"genOta":{"dir":{"value":2},"attrs":{}},"genPollCtrl":{"dir":{"value":2},"attrs":{}},"haDiagnostic":{"dir":{"value":1},"attrs":{"65533":1,"numberOfResets":0,"macRxBcast":0,"macTxBcast":0,"aPSRxBcast":0,"aPSTxBcast":0,"nwkFcFailure":0,"packetBufferAllocateFailures":0,"averageMacRetryPerApsMessageSent":0,"lastMessageLqi":228,"lastMessageRssi":-43}},"lightLink":{"dir":{"value":3},"attrs":{"65533":1}}}}},"_id":"eKKWJ5wwo2kDYzW1"}
```

Koenkk, 2018-06-08:
Lets first try to control it, add to your devices.js

```
    {
        zigbeeModel: [TRADFRI transformer 30W'],
        model: 'TODO',
        vendor: 'IKEA',
        description: 'TODO',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [tz.onoff],
    },
```

ciB89, 2018-06-08:
I can't access devices.js on Hass.io, or?

ciotlosm, 2018-06-08:
@ciB89 you can if you have access to host, via `docker exec`

ciB89, 2018-06-08:
Thanks, will get back to it hopefully on sunday - have to copy the RSA-Key first to the resin-os.
For anyone who is interested and does not know how to do it:

1. Shut down your pi and put the SD Card in your computer.
2. Create a file in the resinos-boot partition called authorized_keys and put your ssh public key in it.
3. Put the SD Card back in your pi and start it up.
4. Run this SSH command:
5.` ssh -t root@hassio -p 22222 “docker exec -it \$(docker ps -f name=homeassistant -q) bash”`

If you’re on windows first putty to root@hassio port 22222 then run
`docker exec -it $(docker ps -f name=homeassistant -q) bash`

ciB89, 2018-06-12:
I added the device to to devices.js - what's up next?

Koenkk, 2018-06-12:
What did you add?

ciB89, 2018-06-12:
What you posted above:
```
    {
        zigbeeModel: [TRADFRI transformer 30W'],
        model: 'TODO',
        vendor: 'IKEA',
        description: 'TODO',
        supports: 'TODO',
        fromZigbee: [],
        toZigbee: [tz.onoff],
    },
```

Koenkk, 2018-06-12:
Ok, is it possible to turn the transfomer on/off?

topic: `zigbee2mqtt/0x90fd9ffffe73e86a/set`
payload: `{"state": "ON"}`

ciB89, 2018-06-12:
For some reason it says it cannot find the device although it for sure joined the network:

```
  zigbee2mqtt:controller Recieved zigbee message with data 0x90fd9ffffe73e86a +246ms
2018-6-12 13:06:38 INFO Connecting with device...
2018-6-12 13:06:38 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"connecting with device"}'
  zigbee-shepherd Device: 0x90fd9ffffe73e86a join the network. +14ms
  zigbee2mqtt:controller Recieved zigbee message with data 0x90fd9ffffe73e86a +13ms
2018-6-12 13:06:38 INFO Device incoming...
2018-6-12 13:06:38 INFO MQTT publish, topic: 'zigbee2mqtt/bridge/log', payload: '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:controller Recieved zigbee message with data online +4ms
2018-6-12 13:06:51 ERROR Cannot handle 'zigbee2mqtt/0x90fd9ffffe73e86a/set' because deviceID of '0x90fd9ffffe73e86a' cannot be found
```

ciB89, 2018-06-12:
My bad, I had to use the friendly name which I forgot that I had given to it. It works turning it on/off :)

Koenkk, 2018-06-12:
Ok, does it also support color? The site says: https://www.ikea.com/gb/en/products/lighting/integrated-lighting/tr%C3%A5dfri-driver-for-wireless-control-grey-art-60342656/, *Use the wireless dimmer to dim, turn on/off up to 10 light sources - all will behave in the same way.* 

For brightness:
```
    {
        zigbeeModel: ['TRADFRI transformer 30W'],
        model: '603.426.56',
        vendor: 'IKEA',
        description: 'TRADFRI driver for wireless control (30 watt)',
        supports: 'on/off, brightness',
        fromZigbee: [fz.light_brightness, fz.ignore_onoff_change],
        toZigbee: [tz.onoff, tz.light_brightness, tz.ignore_transition],
    },
```

ciB89, 2018-06-12:
Hm... well I was thinking that you could also use the driver to power the colored LED-light strips. But I might be wrong since they also come with a separate power supply with a button on it to change color/mode.

Koenkk, 2018-06-12:
Ok, can you test if the brightness also works? If so, I can add support for it.

topic: `zigbee2mqtt/0x90fd9ffffe73e86a/set`
payload: `{"brightness": "100"}`

brightness = 0 - 255 range

ciB89, 2018-06-12:
already tested, works perfectly! also the range (0-255) is correct (I first assumed 0-100)

Koenkk, 2018-06-12:
Supported now!

closed 2018-06-12 by Koenkk

rarosalion, 2019-04-05:
Hey, I was having this same issue, and I thought I'd mention, the "ICPSHC24-30EU-IL-1" is also known as "TRADFRI Driver 30W".  I added that description to the zigbeeModel list on my setup, and the light works perfectly now!

Koenkk, 2019-04-05:
@rarosalion could you make a pr of that? https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js