Koenkk/zigbee2mqtt #106: Battery Attribute not Visible within HA (MCCGQ11LM - Aqara Door/Window Sensor)

jarrah31, 2018-06-09:
I can't see that this has been reported before, either in the forum on on here, so it's either an issue or my lack of understanding how this works.

Basically, the battery attribute isn't appearing in HA as seen in this screenshot:
[image]

I have auto discover turned on within HA and Z2M and I can see the battery messages are being sent to HA:
```
Jun 09 12:37:26 zigbeepi npm[2855]: 2018-6-9 12:37:26 INFO MQTT publish, topic: 'zigbee2mqtt/window_ensuite_velux', payload: '{"battery":"100.00","voltage":3065}'
Jun 09 13:08:37 zigbeepi npm[2855]: 2018-6-9 13:08:37 INFO MQTT publish, topic: 'zigbee2mqtt/window_ensuite_velux', payload: '{"battery":"100.00","voltage":3065,"contact":false}'
Jun 09 13:08:38 zigbeepi npm[2855]: 2018-6-9 13:08:38 INFO MQTT publish, topic: 'zigbee2mqtt/window_ensuite_velux', payload: '{"battery":"100.00","voltage":3065,"contact":true}'
```
This is the message sent to set up the sensor:
```
Jun 09 09:54:00 zigbeepi npm[2855]: 2018-6-9 09:54:00 INFO MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'
```
Should battery information appear on HA within the States page? When the sensors are connect via Xiaomi gateway they appear like this:
[image]

On a separate note, is there a way to implement the "open since" attribute?

Koenkk, 2018-06-09:
Open since can be added.

The battery should be visible here: https://imgur.com/a/iaEn5WG

jarrah31, 2018-06-09:
Unfortunately not - when I click my sensor it only gives me the following:
[image]

Having the battery appear as an attribute for the binary_sensor is very useful for creating all sorts of sensors or alerts, e.g - I use the following for my window sensors that connect via Xiaomi GW and would like to do the same with zigbee2mqtt.
```
study_lrg_win_batt:
        value_template: '{{ states.binary_sensor.door_window_sensor_158d0001d68293.attributes.battery_level | default(0) | int }}'
        unit_of_measurement: '%'
        friendly_name: "Study Large Window Sensor Battery"
        icon_template: >
          {% set battery_level = states.binary_sensor.door_window_sensor_158d0001d68293.attributes.battery_level | default(0) | int %}
          {% set battery_round = (battery_level / 10) | int * 10 %}
          {% if battery_round >= 100 %}
            mdi:battery
          {% elif battery_round > 0 %}
            mdi:battery-{{ battery_round }}
          {% else %}
            mdi:battery-alert
          {% endif %}
```

Koenkk, 2018-06-09:
The MQTT binary sensor doesnt support `json_attributes` so that explains why it is not shown (the MQTT sensor does). Perhaps we need to raise an issue at home assistant.

jarrah31, 2018-06-10:
I will be happy to raise this with Home Assistant. 

In order to provide a working example, do you have to hand the manual command line version of publishing sensor details please?

e.g. how would I publish this from the command line instead?
`MQTT publish, topic: 'homeassistant/binary_sensor/0x00158d0001b149eb/contact/config', payload: '{"payload_on":false,"payload_off":true,"value_template":"{{ value_json.contact }}","device_class":"door","json_attributes":["battery","voltage"],"state_topic":"zigbee2mqtt/window_ensuite_velux","availability_topic":"zigbee2mqtt/bridge/state","name":"window_ensuite_velux","unique_id":"0x00158d0001b149eb_contact_zigbee2mqtt"}'`
Plus the payload?
`MQTT publish, topic: 'zigbee2mqtt/window_ensuite_velux', payload: '{"battery":"100.00","voltage":3065,"contact":false}'`

Thanks! 
(btw, love what you're doing here, can't wait to ditch the Xiaomi gateways that talk to cloud servers)

ciotlosm, 2018-06-10:
I'll look into making a PR. Seems easy enough to just replicate https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/sensor/mqtt.py#L128 into https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/binary_sensor/mqtt.py#L95

jarrah31, 2018-06-10:
Superb, thank you very much!

ciotlosm, 2018-06-10:
Preparing here: https://github.com/ciotlosm/home-assistant/tree/mqtt_attributes unfortunately I have 3 failing tests. The future PR will include unique_id for renaming entities on front-end to remove need to use friendly name in configuration.yaml if you don't want to edit yaml.

ciotlosm, 2018-06-10:
@Koenkk how do you send data for binary sensors? Seems there is a conceptual problem. If you want to sent json_attributes, you must send value_template as well to mention which is the 'on'/'off' field, otherwise it would be an invalid type of data .

jarrah31, 2018-06-11:
> Check home-assistant/home-assistant#14912 for the PR
I really appreciate your help, thanks! It's very interesting to see how the Github code review process works on a large project like this. :)

Koenkk, 2018-06-11:
@ciotlosm this is the configuration for a binary sensor.

```
            payload_on: false,
            payload_off: true,
            value_template: '{{ value_json.contact }}',
            device_class: 'door',
            json_attributes: ['battery', 'voltage'],
```

ciotlosm, 2018-06-11:
@Koenkk you're ahead of the code it seems 👍 
@jarrah31 not easy to get stuff in, especially if you're rushing things :)

jarrah31, 2018-06-12:
@ciotlosm did you have to create another PR in the end or was the original one ok?

ciotlosm, 2018-06-12:
Another PR

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

ciotlosm, 2018-06-13:
Current PR for json_attributes is https://github.com/home-assistant/home-assistant/pull/14957. The second one after the split is already merged.

jarrah31, 2018-06-21:
I'm not sure I follow what the end result will be from the PR - will an upcoming version of HA show the battery attributes for each entity? How can I find out when this will be fixed? TIA

ciotlosm, 2018-06-22:
@jarrah31 there is a bit of controversy around how it should be implemented. However you can already get battery status, by creating a separate sensor based on the correct mqtt topic and use templates to get the battery key if you don't want to wait.

jarrah31, 2018-06-25:
Thanks for the tip, but for some reason the yaml code below isn't picking up the battery details. Am I doing something wrong because it looks ok to me?

```
sensor:
  - platform: mqtt
    name: “battery_window_ensuite_velux”
    state_topic: “zigbee2mqtt/window_ensuite_velux”
    unit_of_measurement: ‘%’
    value_template: “{{ value_json.battery }}”
```
This is what zigbee2mqtt sent to HA:
```
Jun 25 20:30:33 zigbeepi npm[24551]: 2018-6-25 20:30:33 INFO MQTT publish, topic: 'zigbee2mqtt/window_ensuite_velux', payload: '{"battery":"49.00","voltage":3045}'
```

ciotlosm, 2018-06-25:
Looks ok to me too

jarrah31, 2018-06-26:
I think there must have been a hidden character somewhere because after researching many forum posts and other people's examples, I couldn't see anything wrong, so I manually retyped the section and it worked! 

Happy now :)

h4nc, 2018-11-13:
It's really confusing that there are binary_sensor examples here:

https://github.com/Koenkk/zigbee2mqtt/wiki/Integrating-with-Home-Assistant

that the json_attributes feature, although it does not work. 

Like mentioned above I also had to use "sensor" to listen to the batterystatus changes. 

It would be better to delete those json_attribute lines in examples (because they are useless) and show how to get the battery value using the code above.

ryanbeaton, 2018-11-19:
@Koenkk I completely agree with @ciotlosm 

> @jarrah31 I'll close this one as support for battery exists in zigbee2mqtt. Please follow the PR on Home Assistant repo for a fix.

however I think @h4nc has a point that the json_attributes in the binary_sensor examples should be removed as they are not valid home assistant configuration blocks and are causing confusion.

DevRGT, 2019-06-11:
> Unfortunately not - when I click my sensor it only gives me the following:
> [image]
> 
> Having the battery appear as an attribute for the binary_sensor is very useful for creating all sorts of sensors or alerts, e.g - I use the following for my window sensors that connect via Xiaomi GW and would like to do the same with zigbee2mqtt.
> 
> ```
> study_lrg_win_batt:
>         value_template: '{{ states.binary_sensor.door_window_sensor_158d0001d68293.attributes.battery_level | default(0) | int }}'
>         unit_of_measurement: '%'
>         friendly_name: "Study Large Window Sensor Battery"
>         icon_template: >
>           {% set battery_level = states.binary_sensor.door_window_sensor_158d0001d68293.attributes.battery_level | default(0) | int %}
>           {% set battery_round = (battery_level / 10) | int * 10 %}
>           {% if battery_round >= 100 %}
>             mdi:battery
>           {% elif battery_round > 0 %}
>             mdi:battery-{{ battery_round }}
>           {% else %}
>             mdi:battery-alert
>           {% endif %}
> ```
> 
> Thanks for also considering "open since" as this is useful for alerting when doors have been left open too long.

Hello @jarrah31,

I moved also from Xiaomi GW to Zigbee2MQTT.
How did you solve to be able to display different icon and theme depending of the battery level ?